Koenkk/zigbee2mqtt #12017: all problems after the last update

salopette, 2022-04-01:
### What happened?

After the update, the whole Zigbee network reacts very slowly and strangely.
The lqi is always 0-47, although it was well over 100 before.
Timeouts always occur.

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20220219

### Adapter

LAUNCHXL-CC1352P-2

### Debug log

Publish 'set' 'state' to 'Baum-Licht-ST' failed: 'Error: Command 0x84fd27fffece98e9/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 6493 - 1 - 194 - 6 - 11 after 10000ms)'

salopette, 2022-04-01:
[image]

xekil, 2022-04-01:
same problem, i also updated my stick zzh! but without success...

1800Zeta, 2022-04-01:
seeing similar, have rolled back ZZH firmware and the addon version but it's still the same

1800Zeta, 2022-04-01:
OK now fully updated, had to force a couple of devices to 'check in' with a state change but things seem to be working again with ZZH 20220219 and Z2M 1.25

stoinov, 2022-04-03:
Same here. I've only noticed my warnings after the zzh! flash, but they were possibly there before that.
Managed to fix some of the lights by resetting them (4-5 times turning them on/off), but some are not joining after that manipulation.

closed 2022-04-09 by salopette