Koenkk/zigbee2mqtt #12018: Aqara water leak sensor reports false leakage after update to latest version

groenmarsmannetje, 2022-04-01:
### What happened?

I have 3 of these `[Aqare water leak sensors](https://www.zigbee2mqtt.io/devices/SJCGQ11LM.html#xiaomi-sjcgq11lm)` and after updating to version 1.25.0 they all have false leakage reports because it is still totally dry.

### What did you expect to happen?

The state of these water leak sensors should be unaffected after update.

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20211217

### Adapter

zzh

xekil, 2022-04-01:
Hello,

I had exactly the same problem and after several hours looking for a solution I found a very simple solution to solve this problem, just soak the detector in water so that it sends a leak alert, then just dry it and the status becomes dry in home assistant.

Hoping to have helped you ;)

groenmarsmannetje, 2022-04-01:
@xekil thanks for sharing. I will try this out. Probably connecting the two metal pins with a paperclip will already work then.

xekil, 2022-04-01:
@groenmarsmannetje  with pleasure

andiukas, 2022-04-01:
for me Availability is offline then  when i started again pair

groenmarsmannetje, 2022-04-02:
Very annoying … after a couple of restarts the sensors are giving false reports again.

Koenkk, 2022-04-02:
@groenmarsmannetje could you provide the debug log when the incorrect reports are being send?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

groenmarsmannetje, 2022-04-02:
At the moment the sensors are acting normally again. But will do this next time when it occurs again.

njordan77, 2022-04-02:
Having the same problem immediately after updating to 1.25
the workarround does fix it, but the issue comes back again after a while.
battery removal does not help, but this was mentioned before as the issue already came up in 2019.

@Koenkk any idea why this issue now is present @1.25? I'm using z2m for years and having the waterleak sensors at least 2 years but never had this behaviour. Cheers and thanks, Norbert

Koenkk, 2022-04-03:
@mcgivergim I expect this is is caused by https://github.com/Koenkk/zigbee-herdsman-converters/blob/2351faa29151a1d4ef34a1012bd5159f57454537/lib/xiaomi.js#L215 , before https://github.com/Koenkk/zigbee-herdsman-converters/pull/3994 we didn't parse any message from the basic Xiaomi cluster: https://github.com/Koenkk/zigbee-herdsman-converters/blob/6f422d10d1d10e41cdc5dae22f584c5a4a9d2345/devices/xiaomi.js#L904

McGiverGim, 2022-04-03:
I will look into it. I've one Aqara water leak so I can try to reproduce.

McGiverGim, 2022-04-03:
Some of you, with the problem, can put the logs in debug mode and look for something like:
```
debug 2022-04-02 20:49:35: Received Zigbee message from 'fuga_agua_ac', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3125,"10":41058,"100":0,"3":24,"4":13224,"5":43,"6":[0,1],"8":516}}' from endpoint 1 with groupID 0
```
with the name of the Aqara device?
If it does not appear (this messages is sent by the device in a periodic manner), push the "button" on the device and it will send one of them.

McGiverGim, 2022-04-04:
Nobody can publish some log with the problem? In my case, I can't reproduce the problem.

This is a log without water:
```
debug 2022-04-03 09:36:28: Received Zigbee message from 'fuga_agua_ac', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3125,"10":41058,"100":0,"3":24,"4":13224,"5":43,"6":[0,0],"8":516},"modelId":"lumi.sensor_wleak.aq1"}' from endpoint 1 with groupID 0
```

This is a log with water:
```
debug 2022-04-03 09:39:56: Received Zigbee message from 'fuga_agua_ac', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3125,"10":41058,"100":1,"3":25,"4":13224,"5":43,"6":[0,1],"8":516},"modelId":"lumi.sensor_wleak.aq1"}' from endpoint 1 with groupID 0
```
As you can see, the attribute `100` shows `0` and `1` that is what the code expects. We suppose that you are getting something different here and only want to be sure before doing a blind revert.

njordan77, 2022-04-04:
Hi, sorry for the delay. I could but after trying the "hint" about sending a "false alert via shorting the two sense contacts" i now fixed it and since last night i did not have any alarms at all. tried today if normal/real alert would work - and it does.

Please let me know if i can help any further. As i do not understand the background of this issue, i only hope that this quick fix is somehow the solution and the problem will not come back :-) Cheers Norbert

groenmarsmannetje, 2022-04-04:
Maybe I have found a way to reproduce it. When I only restart the zigbee2mqtt addon the problem doesn’t occur, but when I reboot the host (I use Hassos) then after the everything has been started the sensors are reporting leakage, but automatically turn off after a minute. The first time after installing the new version the state remained on until I did the manual workaround. Unfortunately I didn’t have debug logging on.

McGiverGim, 2022-04-05:
> When I only restart the zigbee2mqtt addon the problem doesn’t occur, but when I reboot the host (I use Hassos) then after everything has been started the sensors are reporting leakage, but they automatically turn off after approx. a minute.

@groenmarsmannetje If the problem is the code of the genBasic that was added, then it has nothing to do with a reboot of Z2M or HAOS. The code only processes messages sent by the device and the device does not send anything after a reboot because it is not aware of that.
If this is reproducible, then is something about the internal state that HA or Z2M store, one of them show true until it get the "real" state. If it changes one minute later I suspect that HAOS is publishing the true until it gets the real state from Z2M one minute later, that is false.

> Unfortunately I didn’t have debug logging on

Is better the debug on, but with info you can see too if there is some `water_leak: true` at some moment and `water_leak: false` one minute later or not. In this way we can know if the true has been Z2M or HAOS. Do you have the log of your reboot?

> The second device seemed to be low on battery, and after replacing the battery it first sent `{"water_leak":true}` after each button press

Maybe it sends something after a battery replacement, but I tried to remove, wait, and add again the battery without luck. What I suspect is that internally, maybe the device sends a "true" until the first real "true", then it starts working right. I would like to have this in logs, maybe there is a difference that we can use, only to be sure before reverting. 

Someone at this thread have the problem or it is working as intended now? Some of you changed the battery or maybe repaired it before the problem?

groenmarsmannetje, 2022-04-05:
I am not sure how it works under the hood. But in case of reboot then the coordinator is also power cycled and I do not think this happen when I restart the addon. Just a thought…

Tho85, 2022-04-05:
The problem has not resurfaced so far.

I tried a different angle: I have a daily backup of all my files and looked through the last 20 backups of the state database in `database.db`. In these backups the attribute `100` was always set to `1`, even though there hasn't been a water leak:
```
{"id":48,"type":"EndDevice","ieeeAddr":"0x00158d000242c579","nwkAddr":16833,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_wleak.aq1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65281":{"1":3015,"3":21,"4":17320,"5":15,"6":[0,1],"8":516,"10":10013,"100":1},"manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","modelId":"lumi.sensor_wleak.aq1","appVersion":4}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":4,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1648427932876,"defaultSendRequestWhen":"immediate"}
{"id":48,"type":"EndDevice","ieeeAddr":"0x00158d000242c579","nwkAddr":16833,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_wleak.aq1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65281":{"1":3015,"3":21,"4":17320,"5":15,"6":[0,1],"8":516,"10":10013,"100":1},"manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","modelId":"lumi.sensor_wleak.aq1","appVersion":4}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":4,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1648087358398,"defaultSendRequestWhen":"immediate"}
{"id":48,"type":"EndDevice","ieeeAddr":"0x00158d000242c579","nwkAddr":16833,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_wleak.aq1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65281":{"1":3015,"3":20,"4":17320,"5":15,"6":[0,1],"8":516,"10":10013,"100":1},"manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","modelId":"lumi.sensor_wleak.aq1","appVersion":4}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":4,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1647912634184,"defaultSendRequestWhen":"immediate"}
{"id":48,"type":"EndDevice","ieeeAddr":"0x00158d000242c579","nwkAddr":16833,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_wleak.aq1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65281":{"1":3015,"3":19,"4":17320,"5":15,"6":[0,1],"8":516,"10":10013,"100":1},"manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","modelId":"lumi.sensor_wleak.aq1","appVersion":4}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":4,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1647740936697,"defaultSendRequestWhen":"immediate"}
{"id":48,"type":"EndDevice","ieeeAddr":"0x00158d000242c579","nwkAddr":16833,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_wleak.aq1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65281":{"1":3015,"3":21,"4":17320,"5":15,"6":[0,1],"8":516,"10":10013,"100":1},"manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","modelId":"lumi.sensor_wleak.aq1","appVersion":4}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":4,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1648506302376,"defaultSendRequestWhen":"immediate"}
{"id":48,"type":"EndDevice","ieeeAddr":"0x00158d000242c579","nwkAddr":16833,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_wleak.aq1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65281":{"1":3015,"3":19,"4":17320,"5":15,"6":[0,1],"8":516,"10":10013,"100":1},"manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","modelId":"lumi.sensor_wleak.aq1","appVersion":4}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":4,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1647569242593,"defaultSendRequestWhen":"immediate"}
[... cut ...] ta":{}}},"appVersion":4,"stackVersion":2,"hwVersion":2,"dateCode":"20170721","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1647828291909,"defaultSendRequestWhen":"immediate"}
```
Note that the lines are not in chronological order. If I go further back, I can even see this value changing between 0 and 1, even though I didn't have a water leak back then either.

For me it looks like attribute 100 does in fact not designate a water leak, but some other value.

McGiverGim, 2022-04-05:
Strange. In my database the 100 is 0 (as expected). What seems wrong is this value to 1 and this explains the behavior seen here. I think that when z2m is restarted, it uses the value of the database until it receives a real message of the device. For this reason it starts to work when you put the device in water (or you press the button of the device).
@Tho85 your actual (not the historic) database contains a 0 or a 1?

Tho85, 2022-04-05:
@McGiverGim It contains a `0` now.

McGiverGim, 2022-04-05:
Then I think we need to see if once this value is right (contains 0), the device fails or not at some moment.
I think it will work ok now.

malaakso, 2022-04-06:
I'm still running 1.24 and my sensor has showed "water_leak" as null for quite some time (maybe already in 1.23. I did test once that putting the sensor in water changed water leak to true and drying the sensor turned it back to false. But restarting z2m seems to have made the value disappear again.

groenmarsmannetje, 2022-04-06:
> I am not sure how it works under the hood. But in case of reboot then the coordinator is also power cycled and I do not think this happens when I restart the addon. The memory/cache of stick, mqtt and HA will then also be flushed and reinitialized with wrong (on) state. While that should be state unknown or unavailable (or last known state). Just a thought…
> 
> I will try to reproduce it tomorrow with debug on.

This is a debug log with the records for one of the sensors when leak is on (see second line) during startup after reboot and automatically turns of 16 seconds later (see last line):

```
info  2022-04-06 14:46:22: kelder_watersensor (0x00158d0002704a23): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
debug 2022-04-06 14:46:25: Received MQTT message on 'zigbee2mqtt/kelder_watersensor' with data '{"battery":99,"battery_low":null,"last_seen":"2022-04-01T12:41:18+02:00","linkquality":98,"temperature":22,"voltage":2975,"water_leak":true}'
debug 2022-04-06 14:46:25: Received MQTT message on 'zigbee2mqtt/kelder_watersensor/availability' with data 'online'
debug 2022-04-06 14:46:27: Passive device 'kelder_watersensor' was last seen '0.89' hours ago.
info  2022-04-06 14:46:27: MQTT publish: topic 'zigbee2mqtt/kelder_watersensor/availability', payload 'online'
info  2022-04-06 14:46:32: MQTT publish: topic 'homeassistant/sensor/0x00158d0002704a23/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/kelder_watersensor/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d0002704a23"],"manufacturer":"Xiaomi","model":"Aqara water leak sensor (SJCGQ11LM)","name":"kelder_watersensor","sw_version":"3000-0001"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/kelder_watersensor","name":"kelder_watersensor_battery","state_class":"measurement","state_topic":"zigbee2mqtt/kelder_watersensor","unique_id":"0x00158d0002704a23_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info  2022-04-06 14:46:32: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0002704a23/water_leak/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/kelder_watersensor/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d0002704a23"],"manufacturer":"Xiaomi","model":"Aqara water leak sensor (SJCGQ11LM)","name":"kelder_watersensor","sw_version":"3000-0001"},"device_class":"moisture","json_attributes_topic":"zigbee2mqtt/kelder_watersensor","name":"kelder_watersensor_water_leak","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/kelder_watersensor","unique_id":"0x00158d0002704a23_water_leak_zigbee2mqtt","value_template":"{{ value_json.water_leak }}"}'
info  2022-04-06 14:46:32: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0002704a23/battery_low/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/kelder_watersensor/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d0002704a23"],"manufacturer":"Xiaomi","model":"Aqara water leak sensor (SJCGQ11LM)","name":"kelder_watersensor","sw_version":"3000-0001"},"device_class":"battery","entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/kelder_watersensor","name":"kelder_watersensor_battery_low","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/kelder_watersensor","unique_id":"0x00158d0002704a23_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low }}"}'
info  2022-04-06 14:46:32: MQTT publish: topic 'homeassistant/sensor/0x00158d0002704a23/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/kelder_watersensor/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d0002704a23"],"manufacturer":"Xiaomi","model":"Aqara water leak sensor (SJCGQ11LM)","name":"kelder_watersensor","sw_version":"3000-0001"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/kelder_watersensor","name":"kelder_watersensor_voltage","state_class":"measurement","state_topic":"zigbee2mqtt/kelder_watersensor","unique_id":"0x00158d0002704a23_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}'
info  2022-04-06 14:46:32: MQTT publish: topic 'homeassistant/sensor/0x00158d0002704a23/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/kelder_watersensor/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00158d0002704a23"],"manufacturer":"Xiaomi","model":"Aqara water leak sensor (SJCGQ11LM)","name":"kelder_watersensor","sw_version":"3000-0001"},"device_class":"temperature","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/kelder_watersensor","name":"kelder_watersensor_temperature","state_class":"measurement","state_topic":"zigbee2mqtt/kelder_watersensor","unique_id":"0x00158d0002704a23_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
info  2022-04-06 14:46:32: MQTT publish: topic 'homeassistant/sensor/0x00158d0002704a23/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/kelder_watersensor/availability"}],"availability_m [... cut ...] ","state_topic":"zigbee2mqtt/kelder_watersensor","unique_id":"0x00158d0002704a23_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
debug 2022-04-06 14:46:41: Passive device 'kelder_watersensor' was last seen '0.89' hours ago.
info  2022-04-06 14:46:41: MQTT publish: topic 'zigbee2mqtt/kelder_watersensor', payload '{"battery":99,"battery_low":false,"last_seen":"2022-04-06T13:53:15+02:00","linkquality":54,"tamper":false,"temperature":23,"voltage":2975,"water_leak":false}'
````

Tho85, 2022-04-06:
@McGiverGim The problem has not re-occurred so far. I continue to have debug logging enabled and will provide a log file should this issue appear again.

ggottwald, 2022-04-07:
I also had this issue at one of my two sensors. After doing the workaround, the sensor no longer sent a false "water_leak: true" messages. Even after rebooting my pi or restarting z2m, no false leak warning appeared.

njordan77, 2022-04-07:
I've the same situation as @ggottwald . after workarround the issue is gone.....and did only happen on one of my two sensors. you can imagine how fast i drove home from work when i realized the alarm in the office. later i realized that the remote update i triggered was linked to this.....

bbrendon, 2022-04-09:
This happened once before https://github.com/Koenkk/zigbee2mqtt/issues/1967

poisondima, 2022-04-09:
After upgrading yesterday to version 1.25, got a trigger on one sensor, and then at night on the other, good to be home! One of the sensors still hangs in the status of the alarm.
[image]

McGiverGim, 2022-04-09:
Can you get some debug logs of this sensor? Press the button on top to force a genBasic message.

poisondima, 2022-04-09:
@McGiverGim Unfortunately, when you click on that sensor there is nothing in the log.
But there is another sensor, which itself sent a message without pressing:
```
debug 2022-04-09 10:21:12: Received Zigbee message from 'water_stiralka', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3025,"10":0,"100":0,"3":26,"4":5032,"5":7,"6":[1,16777218],"8":516}}' from endpoint 1 with groupID 0
debug 2022-04-09 10:21:12: lumi.sensor_wleak.aq1: unknown key 6 with value 1,16777218
debug 2022-04-09 10:21:12: lumi.sensor_wleak.aq1: unknown key 8 with value 516
debug 2022-04-09 10:21:12: lumi.sensor_wleak.aq1: Processed data into payload {"voltage":3025,"battery":100,"temperature":26,"water_leak":false}
debug 2022-04-09 10:21:12: lumi.sensor_wleak.aq1: Processed data into payload {"voltage":3025,"battery":100,"temperature":26,"water_leak":false}
info  2022-04-09 10:21:12: MQTT publish: topic 'zigbee2mqtt/water_stiralka', payload '{"battery":100,"battery_low":false,"last_seen":"2022-04-09T10:21:12+03:00","linkquality":75,"tamper":false,"temperature":26,"voltage":3025,"water_leak":false}'
```

McGiverGim, 2022-04-09:
Yes, but I wanted one of the sensor with the problem. 
If the sensor does not send nothing when pressing the button, then I suspect, seeing other logs, z2m is taking the value of the database when restarting, that it's 1 for some reason.
When the device does not send nothing when pressing in my case changing battery fix the issue.

McGiverGim, 2022-04-09:
@poisondima seein my logs, the device sends itself a `genBasic` message each 50 minutes.
Can you left your device with the problem alone to see if it sends this message each 50 minutes or not and what contains?

poisondima, 2022-04-09:
@McGiverGim that's exactly what I'm waiting for when the device sends me a request
[image]

poisondima, 2022-04-09:
@McGiverGim I got the message:
```
debug 2022-04-09 11:28:09: Received Zigbee message from 'water_kuxnia', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3155,"10":62550,"100":1,"3":23,"4":17320,"5":27,"6":[0,1],"8":516}}' from endpoint 1 with groupID 0
debug 2022-04-09 11:28:09: lumi.sensor_wleak.aq1: unknown key 6 with value 0,1
debug 2022-04-09 11:28:09: lumi.sensor_wleak.aq1: unknown key 8 with value 516
debug 2022-04-09 11:28:09: lumi.sensor_wleak.aq1: Processed data into payload {"voltage":3155,"battery":100,"temperature":23,"water_leak":true}
debug 2022-04-09 11:28:09: lumi.sensor_wleak.aq1: Processed data into payload {"voltage":3155,"battery":100,"temperature":23,"water_leak":true}
info  2022-04-09 11:28:09: MQTT publish: topic 'zigbee2mqtt/water_kuxnia', payload '{"battery":100,"battery_low":false,"last_seen":"2022-04-09T11:28:09+03:00","linkquality":141,"tamper":false,"temperature":23,"voltage":3155,"water_leak":true}'
```

McGiverGim, 2022-04-09:
I suppose you have not rebooted Z2M, right?

Ok, this message demonstrates that something is wrong with the devices at some moment, because it sends the message with a "1" instead of a "0".
I will modify the code to ignore the `genBasic`, is a pitty, because if the real leak happens whe you are rebooting or something, this was a way to get it some minutes later.
Only as curiosity, if you short the contacts of the device, it will start to send the correct message?

poisondima, 2022-04-09:
I did a Z2M reboot on 2022-04-09 10:12:32
Put it in water:
```
debug 2022-04-09 11:42:42: Received Zigbee message from 'water_kuxnia', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
info  2022-04-09 11:42:42: MQTT publish: topic 'zigbee2mqtt/water_kuxnia', payload '{"battery":100,"battery_low":false,"last_seen":"2022-04-09T11:42:42+03:00","linkquality":141,"tamper":false,"temperature":23,"voltage":3155,"water_leak":true}'
```
Took it out of the water:
```
debug 2022-04-09 11:45:11: Received Zigbee message from 'water_kuxnia', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID 0
info  2022-04-09 11:45:11: MQTT publish: topic 'zigbee2mqtt/water_kuxnia', payload '{"battery":100,"battery_low":false,"last_seen":"2022-04-09T11:45:11+03:00","linkquality":141,"tamper":false,"temperature":23,"voltage":3155,"water_leak":false}'
```

McGiverGim, 2022-04-09:
PR pushed to ignore the field. If nobody has some better idea, is the only way at this moment to not receive this kind of false positives 😢

closed 2022-04-09 by Koenkk