Koenkk/zigbee2mqtt #12019: TuYa TS0601_smart_air_house_keeper converter error

vmishurov, 2022-04-01:
### What happened?

I get error messages "Unrecognized DP #20":

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20211217

### Adapter

SONOFF ZB Dongle

### Debug log

Debug Received Zigbee message from 'air_sensor1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":20}],"seq":256}' from endpoint 1 with groupID 0

Warning zigbee-herdsman-converters:TuyaSmartAirBox: Unrecognized DP #20 with data {"dp":20,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}

el-bakkali, 2022-04-01:
I get the same issue since recently updating to 1.25.0-1
and the readings for pm25 are acting erractic too due to recent update 

adapter im using is sonoff ZB dongle plus

Koenkk, 2022-04-02:
To figure out what datapoint 20 (dp) means you can follow: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats . If we know what it means we can expose it to MQTT.

Burki24, 2022-04-03:
Have also problems with the Air Housekeeper after updating tu 1.25.0. The DP 20 is allready known in lib-Tuya, as i adapted the converter to fit. It has different Datapoints to Smart Air Box. https://github.com/Koenkk/zigbee-herdsman-converters/pull/3997

   ```js
 // tuya smart air box
    tuyaSabCO2: 2,
    tuyaSabTemp: 18,
    tuyaSabHumidity: 19,
    tuyaSabVOC: 21,
    tuyaSabFormaldehyd: 22,
    // tuya Smart Air House Keeper, Multifunctionale air quality detector.
    // CO2, Temp, Humidity, VOC and Formaldehyd same as Smart Air Box
    tuyaSahkMP25: 2,
    tuyaSahkCO2: 22,
    tuyaSahkFormaldehyd: 20,
```
So, I'm working on this issue allready.

Koenkk, 2022-04-03:
@Burki24 thanks!

Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/4074, 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-04-03 by Koenkk