Koenkk/zigbee2mqtt #12021: Aqara E1 MCCGQ14LM not working with EZSP backend

agners, 2022-04-01:
### What happened?

Pairing Aqara E1 (MCCGQ14LM) seems to succeed (see logs), however, I don't get any events state updates after the device joined. It *feels* as if the device immediately drops from the network just after joining (there are no indication in the logs though). No log entries when "opening/closing" the magnet sensor. Also, pressing the reset button short once causes a `Received Zigbee message`, so it seems the device is still on the network.

The Battery value has a question mark and the State tab is empty (empty dict `{}`).

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

7.0.2.0 build 406

### Adapter

CP2102N/MGM210P based device

### Debug log

```
Zigbee2MQTT:info  2022-04-01 15:23:39: Device '0x54ef4410002930eb' joined                                                                                                                                                                                                                                                    
Zigbee2MQTT:info  2022-04-01 15:23:39: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410002930eb","ieee_address":"0x54ef4410002930eb"},"type":"device_joined"}'                                                                                                                  
Zigbee2MQTT:info  2022-04-01 15:23:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x54ef4410002930eb"},"type":"device_connected"}'                                                                                                                                                  
Zigbee2MQTT:info  2022-04-01 15:23:39: Starting interview of '0x54ef4410002930eb'                                                                                                                                                                                                                                            
Zigbee2MQTT:info  2022-04-01 15:23:39: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410002930eb","ieee_address":"0x54ef4410002930eb","status":"started"},"type":"device_interview"}'                                                                                            
Zigbee2MQTT:info  2022-04-01 15:23:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x54ef4410002930eb"},"type":"pairing"}'                                                                                                                                
Zigbee2MQTT:debug 2022-04-01 15:23:41: Received Zigbee message from '0x54ef4410002930eb', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0                                                                                                                                    
Zigbee2MQTT:debug 2022-04-01 15:23:41: Received Zigbee message from '0x54ef4410002930eb', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0                                                                                                                                  
Zigbee2MQTT:debug 2022-04-01 15:23:42: Received Zigbee message from '0x54ef4410002930eb', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0}' from endpoint 1 with groupID 0                                                                                                                                     
Zigbee2MQTT:debug 2022-04-01 15:23:42: Received Zigbee message from '0x54ef4410002930eb', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0                                                                                                                                                  
Zigbee2MQTT:debug 2022-04-01 15:23:42: Received Zigbee message from '0x54ef4410002930eb', type 'attributeReport', cluster 'genBasic', data '{"appVersion":4,"modelId":"lumi.magnet.acn001"}' from endpoint 1 with groupID 0                                                                                                  
Zigbee2MQTT:debug 2022-04-01 15:23:42: Received Zigbee message from '0x54ef4410002930eb', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0                                                                                                                                                  
Zigbee2MQTT:debug 2022-04-01 15:23:43: Received Zigbee message from '0x54ef4410002930eb', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x0000000000000000","zoneState":0}' from endpoint 1 with groupID 0                                                                                                  
Zigbee2MQTT:debug 2022-04-01 15:23:43: No converter available for 'MCCGQ14LM' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x0000000000000000","zoneState":0}'                                                                                                                                  
Zigbee2MQTT:debug 2022-04-01 15:23:46: Received Zigbee message from '0x54ef4410002930eb', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x0000000000000000","zoneState":0}' from endpoint 1 with groupID 0                                                                                                  
Zigbee2MQTT:debug 2022-04-01 15:23:46: No converter available for 'MCCGQ14LM' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x0000000000000000","zoneState":0}'                                                                                                                                  
Zigbee2MQTT:debug 2022-04-01 15:23:47: Received Zigbee message from '0x54ef4410002930eb', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x0000000000000000","zoneState":0}' from endpoint 1 with groupID 0                                                                                                  
Zigbee2MQTT:debug 2022-04-01 15:23:47: No converter available for 'MCCGQ14LM' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x0000000000000000","zoneState":0}'                                                                                                                                  
Zigbee2MQTT:debug 2022-04-01 15:23:48: Received Zigbee message from '0x54ef4410002930eb', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x0000000000000000","zoneState":0}' from endpoint 1 with groupID 0                                                                                                  
Zigbee2MQTT:debug 2022-04-01 15:23:48: No converter available for 'MCCGQ14LM' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x0000000000000000","zoneState":0}'                                                                                                                                  
Zigbee2MQTT:debug 2022-04-01 15:23:49: Received Zigbee message from '0x54ef4410002930eb', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x0000000000000000","zoneState":0}' from endpoint 1 with groupID 0                                                                                                  
Zigbee2MQTT:debug 2022-04-01 15:23:49: No converter available for 'MCCGQ14LM' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x0000000000000000","zoneState":0}'                                                                                                                                  
Zigbee2MQTT:debug 2022-04-01 15:23:50: Received Zigbee message from '0x54ef4410002930eb', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x0000000000000000","zoneState":0}' from endpoint 1 with groupID 0                                                                                                  
Zigbee2MQTT:debug 2022-04-01 15:23:50: No converter available for 'MCCGQ14LM' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x0000000000000000","zoneState":0}'                                                                                                                                  
[... cut ...] e,"vendor":"Xiaomi"},"type":"pairing"}'                
Zigbee2MQTT:debug 2022-04-01 15:24:06: Received Zigbee message from '0x54ef4410002930eb', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.magnet.acn001"}' from endpoint 1 with groupID 0

<pressing reset button short>
Zigbee2MQTT:debug 2022-04-01 16:21:02: Received Zigbee message from '0x54ef4410002930eb', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.magnet.acn001"}' from endpoint 1 with groupID 0
```

Koenkk, 2022-04-01:
This happens because ezsp is not responding with the correct manufacturer code.

ZHA fixed it in https://github.com/zigpy/bellows/commit/b6d5db70373f516f034c6f7b9915aed9ccee6b39 , @kirovilya is this something we could add to the zigbee-herdsman adapter?

agners, 2022-04-01:
FWIW, related work around in bellows: https://github.com/zigpy/bellows/commit/b6d5db70373f516f034c6f7b9915aed9ccee6b39

kirovilya, 2022-04-01:
Earlier, I tried to work with a similar device through EZSP adapter - I set the manufacturer's code at the start of the adapter. https://github.com/Koenkk/zigbee2mqtt/issues/9274#issuecomment-957966019

I looked at how it was done in ZHA - we need to think about how to do the same in our adapter.
But now I don't have similar Aqara device to check - sold it :(

agners, 2022-04-02:
@kirovilya from what I can tell its just using the IEEE prefix and sets the manufacturing code. For the implementation, test wise, you probably can use any device's IEEE to see if your implementation works. I can do tests with the real device here, but I am really not fluent in JS/TS myself :sweat_smile: 

I have three devices here which cause similar problems: Aqara Button WXKG11LM, Aqara Motion Sensor RTCGQ11LM, and the mentioned Magnet/Door sensor MCCGQ14LM.

kirovilya, 2022-04-06:
@agners Hi. I tried to implement the substitution of ManufacturerCode when specific ieee addresses are found.
Make a backup of your z2m installation.
Then, find the path ./node_modules/zigbee-herdsman/dist/adapter/ezsp/driver/ and put the files from this archive there.
[driver.zip](https://github.com/Koenkk/zigbee2mqtt/files/8429497/driver.zip)
Restart Z2M and try to repair your device.
If not work, then restore the same files from your backup.

ministryofsillywalks, 2022-04-12:
@kirovilya not OP but I just stumbled across this issue as I have the same problem pairing the E1 sensor with my sonoff zigbee bridge based on ezsp.
I tried your driver.zip fix but sadly it didn't work for me. The sensor is still not correctly working.
All I did was cp the files from the zip into the folder you mentioned inside my docker install of z2m. Should I have done something apart from that? (like set any persmission etc?)
Happy to debug this some more if you need me to. Let me know what I can do to help!

kirovilya, 2022-04-12:
@ministryofsillywalks in addition to copying files and restarting z2m, you need to re-pair the device

ministryofsillywalks, 2022-04-12:
My z2m instance crashes when interviewing with that driver.
`Zigbee2MQTT:info 2022-04-12 10:00:18: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef44100025d447","ieee_address":"0x54ef44100025d447"},"type":"device_leave"}'
Zigbee2MQTT:info 2022-04-12 10:00:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x54ef44100025d447"},"type":"device_removed"}'
Zigbee2MQTT:info 2022-04-12 10:00:21: Device '0x54ef44100025d447' joined
Zigbee2MQTT:info 2022-04-12 10:00:21: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef44100025d447","ieee_address":"0x54ef44100025d447"},"type":"device_joined"}'
Zigbee2MQTT:info 2022-04-12 10:00:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x54ef44100025d447"},"type":"device_connected"}'
Zigbee2MQTT:info 2022-04-12 10:00:21: Starting interview of '0x54ef44100025d447'
Zigbee2MQTT:info 2022-04-12 10:00:21: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef44100025d447","ieee_address":"0x54ef44100025d447","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info 2022-04-12 10:00:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x54ef44100025d447"},"type":"pairing"}'
Error: Route error

at /app/node_modules/zigbee-herdsman/src/utils/waitress.ts:33:70
at Waitress.forEachMatching (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:83:17)
at Waitress.reject (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:33:21)
at Driver.handleRouteError (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:336:23)

at Driver.handleFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:285:18)

at Ezsp.emit (node:events:526:28)
at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:129:28)

at SerialDriver.emit (node:events:526:28)
at SerialDriver.handleDATA (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:319:14)

at SerialDriver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:272:22)`

agners, 2022-04-12:
@kirovilya sorry I missed your comment https://github.com/Koenkk/zigbee2mqtt/issues/12021#issuecomment-1090600519

Just did a test with driver2.zip, and it looks good here too. I tested [MCCGQ14LM](https://www.zigbee2mqtt.io/devices/MCCGQ14LM.html#xiaomi-mccgq14lm), [RTCGQ11LM](https://www.zigbee2mqtt.io/devices/RTCGQ11LM.html#xiaomi-rtcgq11lm) and [WXKG11LM](https://www.zigbee2mqtt.io/devices/WXKG11LM.html#xiaomi-wxkg11lm). The last two had a IEEE prefix of `0x00158d`, so I had to extend `IEEE_PREFIX_MFG_ID`

```javascript
const IEEE_PREFIX_MFG_ID = [
    { mfgId: 0x115F, prefix: [0x04, 0xcf, 0xfc] },
    { mfgId: 0x115F, prefix: [0x54, 0xef, 0x44] },
    { mfgId: 0x115F, prefix: [0x00, 0x15, 0x8d] },
];
```

With that, all three devices report state and seem properly connected. I leave the system running for a couple of hours to see if they stay associated and responsive.

kirovilya, 2022-04-12:
@agners glad to hear it.
RTCGQ11LM and WXKG11LM are old models and should have worked anyway without setting a specific manufacturer code.

I'll do PR in zigbee-herdsman later...

agners, 2022-04-12:
> RTCGQ11LM and WXKG11LM are old models and should have worked anyway without setting a specific manufacturer code.

Yes I noticed that too, also that prefix is not present in ZHA.

I retested with z2m without the additional prefix, and indeed they seem to work. Maybe interviewing wasn't complete in my previous test.

I'll continue to monitor all devices throughout the day, but so far they look good :+1: 

> I'll do PR in zigbee-herdsman later...

Cool, thanks!

Koenkk, 2022-04-12:
Merged.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

agners, 2022-04-13:
Works here with latest dev. Thank you!

closed 2022-04-13 by agners