Koenkk/zigbee2mqtt #12022: Error converting file

unpack8866, 2022-04-01:
### What happened?

Sorry, there is an error in the conversion file of the one gang switch, which makes it unusable; Please modify it!
[image]

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20210319

### Adapter

cc2652p

unpack8866, 2022-04-01:
exposes: [e.switch().withEndpoint('l1')],
    endpoint: (device) => {
        return {'l1': 16};
    },    

Need to add endpoint: 16; It cannot be used normally without adding. The default status will be obtained from endpoint 1

Koenkk, 2022-04-02:
Done!

closed 2022-04-02 by Koenkk

unpack8866, 2022-04-03:
> Done!

The switch is normal, thank you!