Koenkk/zigbee2mqtt #12023: TuYa U86KCJ-ZP（zigbeeModel: 005f0c3）

unpack8866, 2022-04-01:
### What happened?

Not added in version 1.25.0
TuYa U86KCJ-ZP there is another model of this device. Please add it. thank you！
zigbeeModel: 005f0c3
[image]

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20210319

### Adapter

cc2652p

unpack8866, 2022-04-02:
Sorry, the zigbeeModel is wrong

[image]

correct zigbeeModel: 005f0c3b

Koenkk, 2022-04-02:
fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-04-02 by Koenkk