Koenkk/zigbee2mqtt #12025: Curtain Motor Xiaomi ZNCLDJ11LM state is always closed (position attribute works fine)

ASNNetworks, 2022-04-01:
### What happened?

Since update 1.25.0-1 the state is always closed. The position attribute however works fine. So when it's open, it shows position 100, but state still closed. When closing, it shows position 0, and also closed. So this isn't a revert bug, but the position is always closed.

Before 1.25.0-1 I already updated the Zigbee firmware, which did not give this issue. Only after installing 1.25.0-1 did this issue occur. So it has to do with Zigbee2mqtt and not the released Zigbee coordinator firmware.

### What did you expect to happen?

Show the correct state.

### How to reproduce it (minimal and precise)

Open curtain, it will still say closed, while the position does update correctly.

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20220219

### Adapter

Zigstar 4

### Debug log

Closing and opening the curtain gives these logs:
```
Received MQTT message on 'zigbee2mqtt/Kantoor Gordijnen/set' with data 'CLOSE'
Publishing 'set' 'state' to 'Kantoor Gordijnen'
Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440550,"presentValue":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440550,"presentValue":99}' from endpoint 1 with groupID 0
Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":0,"presentValue":0}' from endpoint 1 with groupID 0
Received MQTT message on 'zigbee2mqtt/Kantoor Gordijnen/set' with data 'OPEN'
Publishing 'set' 'state' to 'Kantoor Gordijnen'
Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440551,"presentValue":100}' from endpoint 1 with groupID 0
Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440551,"presentValue":46}' from endpoint 1 with groupID 0
Received Zigbee message from 'Kantoor Gordijnen', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":0,"presentValue":100}' from endpoint 1 with groupID 0
```

ASNNetworks, 2022-04-01:
> Hello! Thanks for your report. Is the values `motor_state` and `running` updated when cover is moving? See [Koenkk/zigbee-herdsman-converters/pull/4027](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4027)

Hi,

It always shows:
`motor_state: null`
`running: false`

Btw, before 1.25 I always got `running: false`. It never went to true when the cover was actually running. This applies to all my curtains (I have this Aqara one, two TuYa motors and two Tuya roller blind), they always show `running: false`, but state and position work fine.

Drafteed, 2022-04-01:
What firmware version is installed on the device? 

[image]

ASNNetworks, 2022-04-01:
> What firmware version is installed on the device?
> 
> [image]

04-13-2017

I have no way to update this, OTA says no update found.

Drafteed, 2022-04-01:
>> EDIT: woah, wait now it does say new update available. Never showed an update before. Let me try and update first, and then I'll get back if that solved it!

Nice! Please leave feedback after update.

ASNNetworks, 2022-04-01:
> > > EDIT: woah, wait now it does say new update available. Never showed an update before. Let me try and update first, and then I'll get back if that solved it!
> 
> Nice! Please leave feedback after update.

It seems the curtain motor is not updating. After it reached 100%, the firmware was still the same and I can update it again. Doesn't seem to actually update it unfortunately...

Drafteed, 2022-04-01:
You may try to re-pair device in Z2M. 

Can you please also send me state from the device tab:

[image]

ASNNetworks, 2022-04-01:
> You may try to re-pair device in Z2M. Maybe this will help.
> 
> Can you please also send me state from the device tab:
> 
> [image]

I deleted the device and paired it again. It's even worse now, the state is `unknown`. Also somehow it also has a temperature attribute now, which it doesn't even have the hardware for.

[image]

```
{
    "last_seen": "2022-04-01T19:32:10+02:00",
    "linkquality": 72,
    "motor_state": null,
    "position": 100,
    "running": null,
    "update": {
        "state": null
    },
    "update_available": null,
    "power_outage_count": 27,
    "state": "OFF",
    "temperature": 35
}
```

Also restarted Z2M and deleted and paired the curtain again, same results.

Drafteed, 2022-04-01:
I [pushed the fix](https://github.com/Koenkk/zigbee2mqtt/pull/12028) of the state handle in HA, in case the motor_state was not defined. This should fix this problem and everything will be ok as in 1.24.0. 

However, if you figure out how to update your device, you will get additional [opening or closing states](https://github.com/Koenkk/zigbee2mqtt/pull/11789).

>> Also somehow it also has a temperature attribute now, which it doesn't even have the hardware for.
Yes, the device does return its internal temperature, but this value is not yet exposed.

ASNNetworks, 2022-04-01:
> [I pushed the fix](https://github.com/Koenkk/zigbee2mqtt/pull/12028) of the state handle in HA, in case the motor_state was not defined. This should fix this problem and everything will be ok as in 1.24.0.

> Just do something with device to update state in HA.

I opened and closed the curtain a few times and set positions, but it remained `unknown`. I even set the state manually to open through the dev tools. Then when changing position, it went to `uknown` immediately.

Should I install the dev branch in a few minutes?

Drafteed, 2022-04-01:
> Should I install the dev branch in a few minutes?

We need to wait for Koen to accept this PR and update the repository, then you can update from dev branch.

> Btw: this is the only curtain that now shows motor_state: null. My other 4 don't even have motor_state (which don't have this state issue).

Right. `motor_state` exposed in limit devices.

DivanX10, 2022-04-02:
У меня стоит последняя версия и все ок

[image]
[image]
[image]
[image]

Koenkk, 2022-04-02:
Should be fixed now in latest dev, can someone confirm? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

ASNNetworks, 2022-04-02:
> Should be fixed now in latest dev, can someone confirm? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Tested the Edge release, works great now!

PR #12028 (2022-04-02): Improve handle `motor_state` in HA
This is a hotfix of #11926. 

I've added an extra check if `motor_state` suddenly returns `null` or `false`, which breaks the state in HA.

Closes: #12025
closed 2022-04-02 by ASNNetworks