Koenkk/zigbee2mqtt #12029: [New device support]: Aqara H1 Pro 3 gang switch QBKG32LM

ShadowFistOkami, 2022-04-01:
### Link

https://www.aqara.com/cn/productDetail/d27

### Database entry

{"id":11,"type":"Router","ieeeAddr":"0x54ef4410002c45ef","nwkAddr":4148,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.n3acn1","epList":[1,2,3,21,41,42,43,51,52,53,61,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,9,1794,2820],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"dateCode":"08-14-2020","modelId":"lumi.switch.n3acn1","zclVersion":3}},"genOnOff":{"attributes":{"245":117440545,"onOff":0}},"aqaraOpple":{"attributes":{"238":3883,"240":0,"247":{"type":"Buffer","data":[100,16,0,101,16,0,102,16,0,3,40,32,152,57,0,0,0,0,149,57,67,29,215,57,150,57,7,114,23,69,151,57,0,0,0,0,5,33,26,0,154,32,0,9,33,0,4,11,32,0,13,35,43,15,0,0,14,35,0,0,0,0]},"252":0,"256":1,"512":1,"513":1,"515":0,"mode":1}},"genMultistateInput":{"attributes":{"numberOfStates":6,"presentValue":0}}},"binds":[],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"245":51393536,"onOff":0}},"aqaraOpple":{"attributes":{"512":1}}},"binds":[],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"245":51393536,"onOff":0}},"aqaraOpple":{"attributes":{"256":1,"512":1}}},"binds":[],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"21":{"epId":21,"inClusterList":[],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"configuredReportings":[],"meta":{}},"41":{"epId":41,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":2}}},"binds":[],"configuredReportings":[],"meta":{}},"42":{"epId":42,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"43":{"epId":43,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":2}}},"binds":[],"configuredReportings":[],"meta":{}},"51":{"epId":51,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"52":{"epId":52,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":2}}},"binds":[],"configuredReportings":[],"meta":{}},"53":{"epId":53,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":2}}},"binds":[],"configuredReportings":[],"meta":{}},"61":{"epId":61,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":2}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":43,"stackVersion":2,"hwVersion":1,"dateCode":"08-14-2020","zclVersion":3,"interviewCompleted":true,"meta":{"configured":503270415},"lastSeen":1648821170215,"defaultSendRequestWhen":"immediate"}

### Comments

First post here, so please be patient with me.

I have successfully managed to bodge an integration of the Aqara H1 Pro 3 gang switch QBKG32LM (essentially a 3 gang version of https://www.zigbee2mqtt.io/devices/QBKG31LM.html).

Managed to get pretty much everything working through trial and (mostly) error, but all features of the device have been integrated successfully, except for the following:

-  hold function (the QBKG31LM 2gang equivalent doesn't seem to support it either, although wireless versions like the WXKG14LM seem to do so)
- triple click (same as above)
- preventReset (was throwing an UnhandledPromiseRejectionWarning error, so it's missing from the code below. Not sure what that does, but if it prevents resetting the switch on hold, that could be the clue to having hold actions supported)

Before I even attempt to submit a PR, would anyone like to take a look and see if the above can be added in order to unlock more features? I don't currently have access to a xiaomi hub nor a zigbee sniffer, and I'm not getting my hopes up due to the fact that these switches are expensive, restricted mostly to China, and extremely rare to find in stock at Ali.

Comments where changes need to be applied to get this device supported have been added to the code below. 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const herdsman = require('zigbee-herdsman');
const {
    precisionRound, mapNumberRange, isLegacyEnabled, toLocalISOString, numberWithinRange, hasAlreadyProcessedMessage,
    calibrateAndPrecisionRoundOptions, addActionGroup, postfixWithEndpointName, getKey,
    batteryVoltageToPercentage, getMetaValue,
} = require('zigbee-herdsman-converters/lib/utils');
const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};
const tzLocal = {
    xiaomi_switch_power_outage_memory: {
        key: ['power_outage_memory'],
        convertSet: async (entity, key, value, meta) => {
//add 'QBKG32LM' to line 2123 in toZigbee.js
            if (['SP-EUC01', 'ZNCZ04LM', 'ZNCZ15LM', 'QBCZ14LM', 'QBCZ15LM', 'SSM-U01', 'SSM-U02', 'DLKZMK11LM', 'DLKZMK12LM',
                'WS-EUK01', 'WS-EUK02', 'WS-EUK03', 'WS-EUK04', 'QBKG19LM', 'QBKG20LM', 'QBKG25LM', 'QBKG26LM',
                'QBKG31LM', 'QBKG32LM', 'QBKG34LM', 'QBKG38LM', 'QBKG39LM', 'QBKG40LM', 'QBKG41LM', 'ZNDDMK11LM'].includes(meta.mapped.model)) {
                await entity.write('aqaraOpple', {0x0201: {value: value ? 1 : 0, type: 0x10}}, manufacturerOptions.xiaomi);
            } else if (['ZNCZ02LM', 'QBCZ11LM', 'LLKZMK11LM'].includes(meta.mapped.model)) {
                const payload = value ?
                    [[0xaa, 0x80, 0x05, 0xd1, 0x47, 0x07, 0x01, 0x10, 0x01], [0xaa, 0x80, 0x03, 0xd3, 0x07, 0x08, 0x01]] :
                    [[0xaa, 0x80, 0x05, 0xd1, 0x47, 0x09, 0x01, 0x10, 0x00], [0xaa, 0x80, 0x03, 0xd3, 0x07, 0x0a, 0x01]];

                await entity.write('genBasic', {0xFFF0: {value: payload[0], type: 0x41}}, manufacturerOptions.xiaomi);
                await entity.write('genBasic', {0xFFF0: {value: payload[1], type: 0x41}}, manufacturerOptions.xiaomi);
            } else if (['ZNCZ11LM'].includes(meta.mapped.model)) {
                const payload = value ?
                    [0xaa, 0x80, 0x05, 0xd1, 0x47, 0x00, 0x01, 0x10, 0x01] :
                    [0xaa, 0x80, 0x05, 0xd1, 0x47, 0x01, 0x01, 0x10, 0x00];

                await entity.write('genBasic', {0xFFF0: {value: payload, type: 0x41}}, manufacturerOptions.xiaomi);
            } else {
                throw new Error('Not supported');
            }
            return {state: {power_outage_memory: value}};
        },
[... 95 lines cut ...]
	  exposes: [
	  e.switch().withEndpoint('left'), e.switch().withEndpoint('center'), e.switch().withEndpoint('right'),
	  e.power(), e.energy(), e.voltage(), e.temperature(), e.power_outage_memory(), e.led_disabled_night(), e.flip_indicator_light(),
	  e.action([
        'single_left', 'double_left', 'single_center', 'double_center',
        'single_right', 'double_right', 'single_left_center', 'double_left_center',
        'single_left_right', 'double_left_right', 'single_center_right', 'double_center_right',
        'single_all', 'double_all','hold_left', 'hold_right', 'hold_both']), 
//hold commands above not working
	  exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
	  .withDescription('Decoupled mode for left button').withEndpoint('left'),
	  exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
[... cut ...]
	  exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
	  .withDescription('Decoupled mode for right button').withEndpoint('right')],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
// set "event" mode. No idea what this does, but needed it to receive mqtt messages from physical switch
        await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
    },
 
};

module.exports = definition;
```

Koenkk, 2022-04-02:
@z-x-f @otnow do you maybe already have experience with this device?

ShadowFistOkami, 2022-04-02:
tagging @jglim too if no one minds, since he seems to be the one who initially added support for the QBKG31LM

Otnow, 2022-04-02:
> @Otnow do you maybe already have experience with this device?

Unfortunately not, so I can't help 🤷

jglim, 2022-04-03:
> tagging @jglim too if no one minds, since he seems to be the one who initially added support for the QBKG31LM

Sorry, I (still) do not own a hub and could not investigate beyond the basic single/double actions.

z-x-f, 2022-04-05:
Hold/release is not available in Aqara T1 and H1 switches

ShadowFistOkami, 2022-04-05:
Damn, thanks @z-x-f

mpashka, 2022-05-13:
Probably the issue with hold/release is the same as we have in #9611 .

I have Aqara D1 QBKG24LM and looks like people experience problems with hold and release actions support. I collected release button logs. Looks like there are some issues with parsing data - probably because of some proprietary extensions.
So currently I can `{"65520":{"data":[170,16,5,65,135,1,1,16,0],"type":"Buffer"}}` message to identify hold action but I suppose it is not the best solution and it will be better to parse data correctly.
Here is hold log example:

```
z:a:zStack:unpi:parser <-- [254,38,68,129,0,0,0,0,2,125,1,1,0,36,0,38,197,211,0,0,18,28,95,17,178,10,240,255,65,9,170,16,5,65,135,1,1,16,0,203,82,28,219]
z:a:zStack:unpi:parser --- parseNext [254,38,68,129,0,0,0,0,2,125,1,1,0,36,0,38,197,211,0,0,18,28,95,17,178,10,240,255,65,9,170,16,5,65,135,1,1,16,0,203,82,28,219]
z:a:zStack:unpi:parser --> parsed 38 - 2 - 4 - 129 - [0,0,0,0,2,125,1,1,0,36,0,38,197,211,0,0,18,28,95,17,178,10,240,255,65,9,170,16,5,65,135,1,1,16,0,203,82,28] - 219
z:a:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":32002,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":36,"securityuse":0,"timestamp":13878566,"transseqnumber":0,"len":18,"data":{"type":"Buffer","data":[28,95,17,178,10,240,255,65,9,170,16,5,65,135,1,1,16,0]}}
z:c:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":178,"manufacturerCode":4447,"commandIdentifier":10},"Payload":[{"attrId":65520,"dataType":65,"attrData":{"type":"Buffer","data":[170,16,5,65,135,1,1,16,0]}}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":32002,"endpoint":1,"linkquality":36,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug Received Zigbee message from '0x00158d0005049bc1', type 'attributeReport', cluster 'genBasic', data '{"65520":{"data":[170,16,5,65,135,1,1,16,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
z:controller:endpoint Write 0x00158d0005049bc1/1 genBasic({"65520":{"value":[170,16,5,65,71,1,1,16,1],"type":65}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false})
z:a:zStack:adapter sendZclFrameToEndpointInternal 0x00158d0005049bc1:32002/1 (0,0,1)
z:a:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":32002,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":5,"options":0,"radius":30,"len":18,"data":{"type":"Buffer","data":[20,95,17,4,2,240,255,65,9,170,16,5,65,71,1,1,16,1]}}
z:a:zStack:unpi:writer --> frame [254,28,36,1,2,125,1,1,0,0,5,0,30,18,20,95,17,4,2,240,255,65,9,170,16,5,65,71,1,1,16,1,252]
z:a:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:debug Converted to {}
z:a:zStack:unpi:parser <-- [254,1,100,1,0,100]
z:a:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
z:a:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
z:a:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
z:a:zStack:unpi:parser --- parseNext []
z:a:zStack:unpi:parser <-- [254,3,68,128,0,1,5,195]
z:a:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,5,195]
z:a:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,5] - 195
z:a:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":5}
z:a:zStack:unpi:parser --- parseNext []
z:a:zStack:unpi:parser <-- [254,26,68,129,0,0,0,0,2,125,1,1,0,66,0,159,210,211,0,0,6,28,95,17,4,4,0,2,125,29,74]
z:a:zStack:unpi:parser --- parseNext [254,26,68,129,0,0,0,0,2,125,1,1,0,66,0,159,210,211,0,0,6,28,95,17,4,4,0,2,125,29,74]
z:a:zStack:unpi:parser --> parsed 26 - 2 - 4 - 129 - [0,0,0,0,2,125,1,1,0,66,0,159,210,211,0,0,6,28,95,17,4,4,0,2,125,29] - 74
z:a:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":32002,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":66,"securityuse":0,"timestamp":13882015,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[28,95,17,4,4,0]}}
z:c:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":4,"manufacturerCode":4447,"commandIdentifier":4},"Payload":[{"status":0}],"Command":{"ID":4,"name":"writeRsp","parameters":[{"name":"status","type":32},{"name":"attrId","type":33,"conditions":[{"type":"statusNotEquals","value":0}]}]}},"address":32002,"endpoint":1,"linkquality":66,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
z:a:zStack:unpi:parser --- parseNext []
```

I will put more details into #9611

closed 2022-05-21 by github-actions[bot]