Koenkk/zigbee2mqtt #12031: Z2M Keeps crashing

Inrego, 2022-04-01:
### What happened?

Very recently, my Z2M keeps crashing, with this error in the log:

```
RangeError: Offset is outside the bounds of the DataView
    at DataView.getUint32 (<anonymous>)
    at parseGreenPowerDataIndication (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:339:33)
    at parseFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:466:29)
    at processFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:414:59)
    at Driver.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:69:59)
    at Driver.emit (node:events:526:28)
    at Driver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:784:14)
    at Parser.emit (node:events:526:28)
    at Parser.onMessage (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
    at slip.Decoder.p.handleEnd (/app/node_modules/zigbee-herdsman/node_modules/slip/src/slip.js:189:18)
Using '/app/data' as data directory
CREATED DECONZ ADAPTER
```

It reboots roughly every 10 seconds.

### How to reproduce it (minimal and precise)

I'm not sure, I don't know what caused the issue.

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

0x26680700

### Adapter

Conbee II

Inrego, 2022-04-02:
Ok, I noticed that watchtower had updated zigbee2mqtt to 1.25.0
That seems to be the cause. I've rolled back to 1.24.0, and now it seems stable again.

andrasg, 2022-04-02:
I am seeing a similar issue. Watchtower updated overnight, now z2m doesn't start. For me it the error was along the line fatal error, unreachable code detected.
Rolled back to 1.24.0 for now.

markprins, 2022-04-02:
seeing same issues. Zigbee2mqtt running as docker container.

Log states:
```
#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0xbecd263c
```

andrasg, 2022-04-02:
Happens on `latest-dev` too.

Twit123, 2022-04-02:
Did you see the release notes:

> Important for users running Raspbian Buster: due to https://github.com/Koenkk/zigbee2mqtt/pull/11804, libseccomp2 is required which needs to be installed manually. If you do not do this the container will crash on startup. See [the docs](https://www.zigbee2mqtt.io/guide/installation/02_docker.html#notes-for-raspberry-pi-users) on how to do this.

Simon

Inrego, 2022-04-02:
I'm running in a Docker container on Ubuntu

Koenkk, 2022-04-02:
```
Using '/app/data' as data directory
#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0xbeb6863c
```

As @markprins mentioned this is due to libsecomp2, read the first note from https://github.com/Koenkk/zigbee2mqtt/releases

@chrishae could you look at the `RangeError: Offset is outside the bounds of the DataView` error?

andrasg, 2022-04-02:
> As @markprins mentioned this is due to libsecomp2, read the first note from https://github.com/Koenkk/zigbee2mqtt/releases

Thanks, this solved it for me. Even read it on the april's fool post yesterday, but completely forgot about it...

ChrisHae, 2022-04-04:
> could you look at the `RangeError: Offset is outside the bounds of the DataView` error?

The message is printed because it received a faulty frame. I will look at it and change the message.

bkupidura, 2022-04-22:
So after update to 1.25.0 i start observing random crashes of z2m. It already happened twice in a week.
1.24.0 was rock solid.

```
RangeError: Offset is outside the bounds of the DataView
    at DataView.getUint8 (<anonymous>)
    at parseEnqueueSendDataResponse (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:292:28)
    at parseFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:466:29)
    at processFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:414:59)
    at Driver.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:69:59)
    at Driver.emit (node:events:526:28)
    at Driver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:784:14)
    at Parser.emit (node:events:526:28)
    at Parser.onMessage (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
    at slip.Decoder.p.handleEnd (/app/node_modules/zigbee-herdsman/node_modules/slip/src/slip.js:189:18)
```

If this is caused by faulty frame, any reason for that? I didnt do anything (no new device) to zigbee network.
Also can we somehow handle those broken frames instead of just throwing panic?

ChrisHae, 2022-04-22:
We would need to see a zigbee sniffer log to see the real origin and kind of the (faulty) frame that causes those messages.

Could anybody provide that?

bkupidura, 2022-04-22:
I never did any zb sniffing, but i have spare cc2652p. So if there is any howto+firmware i can try.

One thing is that i need to avoid breaking my current network. So couple of questions:
1) Sniffer should be part of network?
2) Can i safetly add sniffer to my network without re-pairing everything else? :)

ChrisHae, 2022-04-22:
here is a howto: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

You keep your network as it is and use only the other stick for sniffing. It is a distinct process.

nebhead, 2022-04-27:
For what it's worth, I am experiencing crashes with v1.25.0-dev as well.  I'm running in a docker container on top of Ubuntu 20.04 Server.  Did not experience these issues with v1.24.0 release either.  I can't seem to pinpoint the error, but if someone could point me in the right direction as to what log to look at to get more information, that would be great.

bkupidura, 2022-04-29:
Since my last comment, i didnt notice any restarts. So nothing to share yet. 
Its strange, as i didnt do any changes to my zigbee devices (not before/right after update). But now looks like everything is working like before...

nebhead, 2022-04-29:
> Since my last comment, i didnt notice any restarts. So nothing to share yet. Its strange, as i didnt do any changes to my zigbee devices (not before/right after update). But now looks like everything is working like before...

Weirdly, mine has been up for two days now (since I last posted) and hasn't crashed.  I'll keep an eye on it, but so far, so good.

nebhead, 2022-04-30:
> Weirdly, mine has been up for two days now (since I last posted) and hasn't crashed. I'll keep an eye on it, but so far, so good.

NM... crashed again about 15 hours ago, maybe right after posting my last response.  Unfortunately, still no sign of what might be the issue.  At least it's a silent crash, and docker spins right back up.

nebhead, 2022-05-08:
FWIW - Observed another crash after about 4-days of up-time on the latest stable release v1.25.1.

ChrisHae, 2022-05-10:
@nebhead How serious are those crashes? Aren't they just some error messages in the logs or does the whole program crash?

And could you provide a sniffer log maybe?

Inrego, 2022-05-10:
In my case, a crash means the Docker container crashes and stops. It's set to restart automatically, which it does. And as such, it would be ok if it just happened like once per day or so.
But every 10 seconds is a big problem.

Inrego, 2022-05-11:
I just tried the latest version, and every 10 seconds might have been an overstatement I see. Here's a snippet from my log. The actual number is 33 crashes in 8,5 hours.

And none of my 100 zigbee devices are functional atm. And I'm very sad I left it running for so long. Because all my 100 zigbee devices have dropped connection. Even after rolling back to 1.24.0, all my devices are unreachable now.. Any good ideas how to fix my network without having to re-pair and rename every single device?
```
2022-05-11T12:05:33.404617663Z Using '/app/data' as data directory
2022-05-11T12:05:34.587391799Z CREATED DECONZ ADAPTER
2022-05-11T12:06:24.533524992Z RangeError: Offset is outside the bounds of the DataView
2022-05-11T12:06:24.533566930Z     at DataView.getUint32 (<anonymous>)
2022-05-11T12:06:24.533574265Z     at parseGreenPowerDataIndication (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:339:33)
2022-05-11T12:06:24.533589887Z     at parseFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:466:29)
2022-05-11T12:06:24.533595767Z     at processFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:414:59)
2022-05-11T12:06:24.533601550Z     at Driver.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:69:59)
2022-05-11T12:06:24.533627307Z     at Driver.emit (node:events:527:28)
2022-05-11T12:06:24.533637613Z     at Driver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:784:14)
2022-05-11T12:06:24.533646613Z     at Parser.emit (node:events:527:28)
2022-05-11T12:06:24.533654992Z     at Parser.onMessage (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
2022-05-11T12:06:24.533663669Z     at slip.Decoder.p.handleEnd (/app/node_modules/slip/src/slip.js:189:18)
2022-05-11T12:06:25.331743353Z Using '/app/data' as data directory
2022-05-11T12:06:26.543065269Z CREATED DECONZ ADAPTER
2022-05-11T12:07:24.459733989Z RangeError: Offset is outside the bounds of the DataView
2022-05-11T12:07:24.459778386Z     at DataView.getUint32 (<anonymous>)
2022-05-11T12:07:24.459786613Z     at parseGreenPowerDataIndication (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:339:33)
2022-05-11T12:07:24.459794087Z     at parseFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:466:29)
2022-05-11T12:07:24.459799540Z     at processFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:414:59)
2022-05-11T12:07:24.459805030Z     at Driver.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:69:59)
2022-05-11T12:07:24.459810624Z     at Driver.emit (node:events:527:28)
2022-05-11T12:07:24.459815652Z     at Driver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:784:14)
2022-05-11T12:07:24.459821886Z     at Parser.emit (node:events:527:28)
2022-05-11T12:07:24.459829683Z     at Parser.onMessage (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
2022-05-11T12:07:24.459838286Z     at slip.Decoder.p.handleEnd (/app/node_modules/slip/src/slip.js:189:18)
2022-05-11T12:07:25.183134216Z Using '/app/data' as data directory
2022-05-11T12:07:26.371235747Z CREATED DECONZ ADAPTER
2022-05-11T12:07:54.464481897Z RangeError: Offset is outside the bounds of the DataView
2022-05-11T12:07:54.464524090Z     at DataView.getUint32 (<anonymous>)
2022-05-11T12:07:54.464531833Z     at parseGreenPowerDataIndication (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:339:33)
2022-05-11T12:07:54.464542550Z     at parseFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:466:29)
2022-05-11T12:07:54.464548386Z     at processFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:414:59)
2022-05-11T12:07:54.464554131Z     at Driver.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:69:59)
2022-05-11T12:07:54.464560046Z     at Driver.emit (node:events:527:28)
2022-05-11T12:07:54.464565350Z     at Driver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:784:14)
2022-05-11T12:07:54.464585112Z     at Parser.emit (node:events:527:28)
2022-05-11T12:07:54.464591498Z     at Parser.onMessage (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
2022-05-11T12:07:54.464597348Z     at slip.Decoder.p.handleEnd (/app/node_modules/slip/src/slip.js:189:18)
2022-05-11T12:07:55.210339466Z Using '/app/data' as data directory
2022-05-11T12:07:56.412011636Z CREATED DECONZ ADAPTER
2022-05-11T12:08:02.816513257Z RangeError: Offset is outside the bounds of the DataView
2022-05-11T12:08:02.816575984Z     at DataView.getUint32 (<anonymous>)
2022-05-11T12:08:02.816594954Z     at parseGreenPowerDataIndication (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:339:33)
2022-05-11T12:08:02.816620644Z     at parseFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:466:29)
2022-05-11T12:08:02.816626043Z     at processFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:414:59)
2022-05-11T12:08:02.816630924Z     at Driver.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:69:59)
2022-05-11T12:08:02.816635919Z     at Driver.emit (node:events:527:28)
[... cut ...]
2022-05-11T12:20:25.224067265Z     at Driver.emit (node:events:527:28)
2022-05-11T12:20:25.224072889Z     at Driver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:784:14)
2022-05-11T12:20:25.224078906Z     at Parser.emit (node:events:527:28)
2022-05-11T12:20:25.224084379Z     at Parser.onMessage (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
2022-05-11T12:20:25.224090163Z     at slip.Decoder.p.handleEnd (/app/node_modules/slip/src/slip.js:189:18)
```

Inrego, 2022-05-11:
> > could you look at the `RangeError: Offset is outside the bounds of the DataView` error?
> 
> The message is printed because it received a faulty frame. I will look at it and change the message.

I just tried 1.25.1 and the message is the same. I don't have a sniffer at hand, but 1.24.0 is still working for me. So something changed from 1.24.0 to 1.25.0 which affects this.

ChrisHae, 2022-05-13:
Well these information pinpoint the problem pretty precisely. I will try to find a fix and upload soon.

Just to be curious: do you have any Green Power Devices?

Inrego, 2022-05-13:
Yes I do. And I also have a small suspicion of the device that's causing it. Are you on Discord? If so, hit me up there. Else, I'll try to give a long explanation here

bkupidura, 2022-05-13:
@ChrisHae just FYI i dont own any Green Power devices and i observed those issues also.

Inrego, 2022-05-13:
@bkupidura does your error messages look the same? With `parseGreenPowerDataIndication` in the stacktrace?

bkupidura, 2022-05-13:
Nope, my last stack trace:
```
RangeError: Offset is outside the bounds of the DataView
    at DataView.getUint8 (<anonymous>)
    at parseReceivedDataNotification (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:306:30)
    at parseFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:466:29)
    at processFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:414:59)
    at Driver.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:69:59)
    at Driver.emit (node:events:527:28)
    at Driver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:784:14)
    at Parser.emit (node:events:527:28)
    at Parser.onMessage (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
    at slip.Decoder.p.handleEnd (/app/node_modules/slip/src/slip.js:189:18)
```

ChrisHae, 2022-05-13:
@Inrego perhaps you could test your network without that device.
@bkupidura Thats the same error but origin from another function.
So I will try something and if it fixes the crashes. Perhaps would be interesting to see what devices causes that error in your network.

When you enable zigbee-herdsman debug logging we should be able to see the network address of that device.
https://www.zigbee2mqtt.io/guide/usage/debug.html

Inrego, 2022-05-13:
Alright, so that indicates it's not related to GreenPower. Although I do believe I have a GreenPower device gone rogue that's causing the crashes, the crashes seem to happen when receiving a bad frame (GreenPower or not).

More info on my rogue GreenPower device:
A GreenPower device keeps trying to join my Zigbee network. Once joined, it's spamming errors (on 1.24.0):

```
Zigbee2MQTT:debug 2022-05-13 08:48:08: Received Zigbee message from '0x00000000e202b874', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[111,50,182,171,142,127,40,100,124,30,245,138,8,43,121,37,192,51,88,212,114,94,56,207,252,89,20,237,92,237,129,55,196,110,107,124,196,87,164,154,33,123,185,204,141,146,122,156,204,66,34,134,5,225,131,222,79,185,211,116,113,208,121,118,129,22],"type":"Buffer"}},"commandID":75,"frameCounter":3371391,"gppGddLink":182,"gppNwkAddr":12911,"options":2608,"payloadSize":59,"srcID":3791829108}' from endpoint 242 with groupID null
Zigbee2MQTT:warn  2022-05-13 08:48:08: Received message from unsupported device with Zigbee model 'GreenPower_117' and manufacturer name 'undefined'
Zigbee2MQTT:warn  2022-05-13 08:48:08: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```

On 1.25.0+, the device doesn't join, Z2M just crashes.

The Model (and I think network address) of the joined device keeps changing, so even if I block it from joining again, it joins again shortly after.

I am pretty certain that this rogue GreenPower device is a Schneider Electric PowerTag. Strange thing is, it's already connected to a separate Zigbee network (the Schneider Electric hub).

Also, Z2M version 1.25.1 adds support for these PowerTags, so I'm not sure why they keep crashing on that version. Maybe because they behave differently when connected to their own hub.

bkupidura, 2022-05-13:
@ChrisHae i enabled debug, lets see. But for me those crashes are not that often (like once or twice per week). So probably i need couple of days to share anything.

Inrego, 2022-05-13:
I can now confirm the rogue device. I switched it off, and the errors stopped.

Another strange thing, I have 2 of them. Only one does this. But that's more of a device related problem, not a Z2M problem I figure.

Inrego, 2022-05-13:
Ok, I've updated to 1.25.1 and paired the device successfully. However, I'm still getting a lot of crashes.

ChrisHae, 2022-05-13:
So I have added some quick fixes. Those errors should now be catched and hopefully does not cause crashes anymore.
You could test it. It should be available in the mqtt dev branch in the next release (which should be out soon I think?)

When posting some logs it would be helpfull when zigbee-herdsman debug logging is enabled.

@Inegro Do you have checked if the two GP devices have a different firmware version? Perhaps a newer firmware could be the solution there.

Inrego, 2022-05-13:
On the new latest-dev, and now I'm getting this error:

```
Error: Unhandled error. ('APS TIMEOUT')
    at new NodeError (node:internal/errors:372:5)
    at DeconzAdapter.emit (node:events:516:17)
    at emitUnhandledRejectionOrErr (node:events:384:10)
    at processTicksAndRejections (node:internal/process/task_queues:85:21)
```

Inrego, 2022-05-13:
Trying a few different things on my end. Will give more info tomorrow

Inrego, 2022-05-14:
I've left it overnight, and the container has been running for 8 hours without crash.

With the small caveat that none of my ZigBee devices are working, lol

Inrego, 2022-05-14:
Ok so far it seems to be running alright until I "Permit Join" on my network.

Here's something from my log. It seems to be running ok, although being spammed a lot with some greenpower messages (which is says is from the coordinator?).
After I Permit Join, shortly after an error is thrown followed by Z2M crash

```
Zigbee2MQTT:debug 2022-05-14 07:38:28: Received Zigbee message from 'Coordinator', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[235,43,91,125,161,90,120,82,36,60,113,218,252,8,181,75,100,13,183,216,5,199,165,4,230],"type":"Buffer"}},"commandID":0,"frameCounter":877802526,"gpdEndpoint":226,"gpdIEEEAddr":"0xc705d8b702b84b58","options":13394,"payloadSize":238}' from endpoint 242 with groupID 2948, ignoring since it is from coordinator
Zigbee2MQTT:debug 2022-05-14 07:38:28: Received Zigbee message from 'Coordinator', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[46,52,226,48,161,94,16,0,0,5,0,66,5,82,57,77,52,50,1,224,66,11,48,48,52,46,48,48,48,46,52,50,52,4,224,66,15,83,77,45,50,48,52,52,50,48,49,48,56,45,76,50],"type":"Buffer"}},"commandID":48,"frameCounter":45642752,"options":4190,"payloadSize":48}' from endpoint 242 with groupID 2948, ignoring since it is from coordinator
Zigbee2MQTT:debug 2022-05-14 07:38:29: Received Zigbee message from 'Strømmålende Stikkontakt', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":2482,"rmsCurrent":132}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2022-05-14 07:38:29: MQTT publish: topic 'zigbee2mqtt/Strømmålende Stikkontakt', payload '{"current":1.32,"linkquality":255,"power":248.2,"state":"ON","voltage":232.83}'
Zigbee2MQTT:debug 2022-05-14 07:38:29: Retrieving state of 'Hus Motion Sensor 1' after reconnect
Zigbee2MQTT:debug 2022-05-14 07:38:33: Received Zigbee message from 'Coordinator', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[47,244,218,158,165,175,85,181,141,169,77,40,225,178,47,161,61,75,84,108,93,27,248,252,234,165,207,9,196,170,165,165,249,6,155,239,101,138,231,113,120,186,234,160,21,95,12,208,85,201,199],"type":"Buffer"}},"commandID":0,"frameCounter":877803064,"gpdEndpoint":226,"gpdIEEEAddr":"0x1b5d6c5402b84b18","options":13394,"payloadSize":50}' from endpoint 242 with groupID 2948, ignoring since it is from coordinator
Zigbee2MQTT:debug 2022-05-14 07:38:33: Received Zigbee message from 'Coordinator', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[122,29,114,205,104,95,174,244,164,190,212,208,70,121,85,103,59,230,222,223,231,9,89,98,185],"type":"Buffer"}},"commandID":0,"frameCounter":877803294,"gpdEndpoint":226,"gpdIEEEAddr":"0x09e7dfde02b84b58","options":13394,"payloadSize":72}' from endpoint 242 with groupID 2948, ignoring since it is from coordinator
Zigbee2MQTT:debug 2022-05-14 07:38:37: Received Zigbee message from 'Strømmålende Stikkontakt', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":2432,"rmsVoltage":23293}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2022-05-14 07:38:37: MQTT publish: topic 'zigbee2mqtt/Strømmålende Stikkontakt', payload '{"current":1.32,"linkquality":255,"power":243.2,"state":"ON","voltage":232.93}'
Zigbee2MQTT:debug 2022-05-14 07:38:37: Received Zigbee message from 'Coordinator', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[220,242,17,167,117,229,60,16,82,220,127,220,38,5,212,240,136,83,245,246,94,24,28,203,16,10,252,48,233,95,149,175,229,201,21,157,143,26,116,65,174,227,86,201,97,246,66,31,70,47,245],"type":"Buffer"}},"commandID":0,"frameCounter":877803832,"gpdEndpoint":226,"gpdIEEEAddr":"0x185ef6f502b84b18","options":13394,"payloadSize":125}' from endpoint 242 with groupID 2948, ignoring since it is from coordinator
Zigbee2MQTT:debug 2022-05-14 07:38:37: Received Zigbee message from 'Coordinator', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[107,95,172,126,175,133,14,26,185,120,75,240,51,9,34,103,98,7,137,142,160,10,236,114,123],"type":"Buffer"}},"commandID":0,"frameCounter":877804062,"gpdEndpoint":226,"gpdIEEEAddr":"0x0aa08e8902b84b58","options":13394,"payloadSize":54}' from endpoint 242 with groupID 2948, ignoring since it is from coordinator
Zigbee2MQTT:debug 2022-05-14 07:38:37: Received Zigbee message from 'Coordinator', type 'commandCommisioningNotification', cluster 'greenPower', data '{"commandFrame":{"raw":{"data":[46,52,226,48,161,94,16,0,0,5,0,66,5,82,57,77,52,50,1,224,66,11,48,48,52,46,48,48,48,46,52,50,52,4,224,66,15,83,77,45,50,48,52,52,50,48,49,48,56,45,76,50],"type":"Buffer"}},"commandID":48,"frameCounter":45642752,"options":4190,"payloadSize":48}' from endpoint 242 with groupID 2948, ignoring since it is from coordinator
Zigbee2MQTT:debug 2022-05-14 07:38:38: Received Zigbee message from 'Hus Motion Sensor 1', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107321517,"imageType":269,"manufacturerCode":4107}' from endpoint 2 with groupID null
Zigbee2MQTT:debug 2022-05-14 07:38:38: Device 'Hus Motion Sensor 1' requested OTA
Zigbee2MQTT:debug 2022-05-14 07:38:38: Check if update available for '0x001788010949d3ed' (SML002)
Zigbee2MQTT:debug 2022-05-14 07:38:38: ZigbeeOTA: downloaded main index
[... cut ...]
CREATED DECONZ ADAPTER
Zigbee2MQTT:info  2022-05-14 07:42:41: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-05-14 07:42:41: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":104,"product":0,"revision":"0x26680700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:debug 2022-05-14 07:42:41: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Zigbee2MQTT:info  2022-05-14 07:42:41: Currently 100 devices are joined:
```

Inrego, 2022-05-14:
I'm gonna try now with both of those GreenPower devices switched off. It's just not that feasible for longer periods of time due to the nature of their installation. It requires me to switch off the mains that they're monitoring.

Inrego, 2022-05-14:
Ok, I've now paired those green power devices back to their own hub. And Z2M seems to run stable again. So this issue can be closed I guess.

I'm still having lots of problems when trying to pair them to Z2M, but I guess that's another issue.

Inrego, 2022-05-14:
Scratch that, my ZigBee network is down again.

ChrisHae, 2022-05-17:
There was a pull request that is related to those GP devices I think:
https://github.com/Koenkk/zigbee-herdsman/commit/146997b42dc5b228862cc1057efb8e74be72c556
And there will be one more fix in the near future.

And because of that `Error: Unhandled error. ('APS TIMEOUT')`
could you paste the zigbee-herdsman debug log here? See https://www.zigbee2mqtt.io/guide/usage/debug.html I would need the zigbee-herdsman debug logs to see more. Mqtt debug log is not enough in this case.

Inrego, 2022-05-17:
Ooh that's great news. I've also ordered another zigbee coordinator. But if a fix is coming to Conbee, I'll keep that, and maybe flash the incoming one with sniffer firmware for other purposes.

Thanks for linking to how to enable herdsman log output. I was looking for it in settings, without luck.

Lets put a pin in this issue and wait and see if I'm still having issues with above changes.

closed 2022-07-26 by github-actions[bot]