Koenkk/zigbee2mqtt #12034: Raspberry pi zero ran out of ram during installation

fl0rian19, 2022-04-02:
### What happened?

I followed the zigbee2mqtt install guide for linux on fresh rasbian lite install on a raspberry pi zero wh 1.1 (starter kit)
When running npm start it shows this

pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.25.0-dev start
> node index.js

Building Zigbee2MQTT... (initial build), failed

Build failed; ran out-of-memory, free some memory (RAM) and start again

(node:935) UnhandledPromiseRejectionWarning: Error: Command failed: npm run build
FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory
Aborted
npm ERR! code 134
npm ERR! path /opt/zigbee2mqtt
npm ERR! command failed
npm ERR! command sh -c tsc && node index.js writehash

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-04-02T08_28_02_001Z-debug.log

    at ChildProcess.exithandler (child_process.js:308:12)
    at ChildProcess.emit (events.js:314:20)
    at maybeClose (internal/child_process.js:1022:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5)
(node:935) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:935) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

### Zigbee2MQTT version

latest

### Adapter firmware version

 Z-Stack_3.x.0_20211217

### Adapter

CC2652P SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

0 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'run', 'build' ]
1 info using npm@7.5.2
2 info using node@v12.22.5
3 timing config:load:defaults Completed in 60ms
4 timing config:load:file:/usr/share/nodejs/npm/npmrc Completed in 50ms
5 timing config:load:builtin Completed in 52ms
6 timing config:load:cli Completed in 59ms
7 timing config:load:env Completed in 16ms
8 timing config:load:file:/opt/zigbee2mqtt/.npmrc Completed in 7ms
9 timing config:load:project Completed in 23ms
10 timing config:load:file:/home/pi/.npmrc Completed in 5ms
11 timing config:load:user Completed in 7ms
12 timing config:load:file:/etc/npmrc Completed in 2ms
13 timing config:load:global Completed in 4ms
14 timing config:load:cafile Completed in 6ms
15 timing config:load:validate Completed in 27ms
16 timing config:load:setUserAgent Completed in 12ms
17 timing config:load:setEnvs Completed in 27ms
18 timing config:load Completed in 303ms
19 verbose npm-session c9557d4f23e55672
20 timing npm:load Completed in 546ms
21 timing command:run-script Completed in 571195ms
22 verbose stack Error: command failed
22 verbose stack     at ChildProcess.<anonymous> (/usr/share/nodejs/@npmcli/pro>
22 verbose stack     at ChildProcess.emit (events.js:314:20)
22 verbose stack     at maybeClose (internal/child_process.js:1022:16)
22 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_pro>
23 verbose pkgid zigbee2mqtt@1.25.0-dev
24 verbose cwd /opt/zigbee2mqtt
25 verbose Linux 5.10.92+
26 verbose argv "/usr/bin/node" "/usr/bin/npm" "run" "build"
27 verbose node v12.22.5
28 verbose npm  v7.5.2
29 error code 134
30 error path /opt/zigbee2mqtt
31 error command failed
32 error command sh -c tsc && node index.js writehash
33 verbose exit 134

Trashwarez, 2022-04-02:
I can confirm that for my Raspberry Pi 1B running nodejs v14.19.1

richimaint, 2022-04-03:
Same issue for my Beaglebone Black running nodejs v14.19.1 on Debian 10.3

Koenkk, 2022-04-03:
Try to stop some other applications to free up RAM.

Trashwarez, 2022-04-03:
Already tried that. Didn't help

richimaint, 2022-04-03:
I have downgrade to 1.24.0 and it´s works aigain.

gonzzzales, 2022-04-03:
> I have downgrade to 1.24.0 and it´s works aigain.
I have same problem with orange pi zero. How can i install 1.24.0. It works fine, but 1.25.0 can not start

Trashwarez, 2022-04-03:
> I have same problem with orange pi zero. How can i install 1.24.0. It works fine, but 1.25.0 can not start

git reset --hard 1.24.0
npm ci
npm start

gonzzzales, 2022-04-03:
Thanks, i will try it!
And for clean install can i do this? 

sudo git clone https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
sudo chown -R root:root /opt/zigbee2mqtt
cd /opt/zigbee2mqtt
git reset --hard 1.24.0
npm ci
npm start

Trashwarez, 2022-04-03:
> Thanks, i will try it! And for clean install can i do this?

Yes

RaNo99, 2022-04-05:
Same problem here. 1.24.0 works. 1.25.0 fails with the memory issue.

Koenkk, 2022-04-06:
Could you  try executing `export NODE_OPTIONS=--max_old_space_size=256` before doing `npm start`?

Trashwarez, 2022-04-06:
> Could you try executing `export NODE_OPTIONS=--max_old_space_size=256` before doing `npm start`?

I can confirm that it builds by using this option. Thanks, Koen!

richimaint, 2022-04-06:
i can confirm works on my Beaglebone Black

fl0rian19, 2022-04-07:
> Could you try executing `export NODE_OPTIONS=--max_old_space_size=256` before doing `npm start`?

Thanks for your reply. It works with that option

closed 2022-04-07 by fl0rian19

Koenkk, 2022-04-20:
Pushed a fix to automatically add this variable on system with < 600MB RAM.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)