Koenkk/zigbee2mqtt #12035: missing web ui after update to Matter, cannot see anything

Dylantje, 2022-04-02:
### What happened?

Like the head says

### What did you expect to happen?

works as before

### How to reproduce it (minimal and precise)

I did the upgrade like always, include the backup and restore option.
Is this perhas wrong?

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

do not kwow, cannot see this at the moment

### Adapter

same as firmware

### Debug log

Use domoticz, device works fine
Only where to find the webui?

dev-a, 2022-04-02:
I have the same problem 
I don't have a 404 error or anything else just a blank page 
the node client seems to be running (via ps aux and systemctl status zigbee2mqtt.service)

Dylantje, 2022-04-02:
> I have the same problem I don't have a 404 error or anything else just a blank page the node client seems to be running (via ps aux and systemctl status zigbee2mqtt.service)

What domotica system you are running? on what hardware?

dev-a, 2022-04-02:
> > I have the same problem I don't have a 404 error or anything else just a blank page the node client seems to be running (via ps aux and systemctl status zigbee2mqtt.service)
> 
> What domotica system you are running? on what hardware?

in my case I do not use a home automation system, I use only nodered & zigbee2mqtt ui  directly on my raspberry pi

Dylantje, 2022-04-02:
u used als the webui form z2m?

ondrejbaranek, 2022-04-02:
Same issue on rpi3b+ with bullseye, standalone z2m on it

aarpi3, 2022-04-03:
Same here. Rpi3b+, buster, standalone z2m.

ondrejbaranek, 2022-04-03:
Looks to be chrome(ium) related. Works fine in Firefox.

dev-a, 2022-04-04:
Indeed , on firefox work correctly

ondrejbaranek, 2022-04-04:
Clearing cache in Chrome app settings on Android helps. So this release could break something in the cache for some people :)

dev-a, 2022-04-05:
Hello , 
as far as I am concerned, I use brave
I looked at the functions available on the brave shield
I have deactivated. "update connections to https"
since i have access to ui correctly

I did a small search on google "chrome "+ "https"
It appears that there has been an update of chrome on this subject in March

samme78, 2022-05-10:
Clearing the cache in my Chrome browser solved the issue for me.

closed 2022-06-18 by github-actions[bot]