Koenkk/zigbee2mqtt #12036: Aquara motion sensors vanished overnight

hammondses, 2022-04-02:
### What happened?

At around 10pm last night both versions of the aqara motion sensors vanished from my ZigBee network. 

I have multiple other devices still working perfectly, a lot of which are aqara contact sensors.

### What did you expect to happen?

Nothing 

### How to reproduce it (minimal and precise)

Not sure

### Zigbee2MQTT version

1.25, but unsure of what version I was using prior to updating this morning 

### Adapter firmware version

    20211217

### Adapter

Tubes POE cc2652

habitoti, 2022-04-03:
This happens all the time with Xiaomi/Aqara devices. Just remove it from Z2M and reconnect it (potentially remove battery for a moment also), and you should be good again.

hammondses, 2022-04-03:
Oh it's totally gone from Z2M also, pressing the little button on the motion sensors doesn't seem to make them remember to connect again, I reconnected one manually and it's got no memory of its previous config. 

Seems odd that literally 18 devices of the same type (motion) vanished, yet everything else is perfect :(

closed 2022-05-11 by github-actions[bot]