Koenkk/zigbee2mqtt #12037: HGZB-04D / HGZB-4D-UK not reporting dimmer level

tomlut, 2022-04-02:
### What happened?

I just connected a dual dimmer module recognised as two HGZB-04D / HGZB-4D-UK devices. 

For the most part it operates well. I can control the on/off state and brightness from home assistant. 

The light reports the on/off state to home assistant if changes are made using the dimmer but does not report changes in brightness.

### What did you expect to happen?

Changing the brightness on the dimmer wall plate to be reported to home assistant.

### How to reproduce it (minimal and precise)

Change the brightness level using the wall plate.
Observe no change in the brightness slider in home assistant or in the zigbee2mqtt control panel.

### Zigbee2MQTT version

1.25.0-1 

### Adapter firmware version

20210120

### Adapter

Tubes CC2652P2 based Zigbee to Ethernet Serial Coordinator

tomlut, 2022-05-03:
Still an issue.

Koenkk, 2022-05-03:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['FB56+ZSC05HG1.0', 'FNB56-ZBW01LX1.2', 'LXN56-DS27LX1.3', 'LXN60-DS27LX1.3'],
    model: 'HGZB-04D / HGZB-4D-UK',
    vendor: 'Nue / 3A',
    description: 'Smart dimmer wall switch CUSTOM',
    extend: extend.light_onoff_brightness({disableEffect: true, noConfigure: true}),
    whiteLabel: [{vendor: 'Sunricher', model: 'SR-ZG9001K8-DIM'}],
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.brightness(device.getEndpoint(1)); // Added this
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

tomlut, 2022-05-03:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/4197

closed 2022-05-03 by tomlut