Koenkk/zigbee2mqtt #12038: garage_door_entity of PJ-ZGD01 not supported by Home Assistant

fsevilla3, 2022-04-02:
### What happened?

### Background and issue description
I'm testing changes to support a new device (refer to [[New device support]: LoraTap garage door opener #11570](https://github.com/Koenkk/zigbee2mqtt/issues/11570)), which is very similar to the supported MatSee Plus PJ-ZGD01. This works on the Zigbee2Mqtt side. However, the `garage_door_contact` entity that this creates is rejected by Home Assistant.

```
2022-04-02 23:53:16 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_binary_sensor_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}], 'device': {'configuration_url': 'http://HOSTNAME:8080/#/device/0xDEVICEID/info', 'identifiers': ['zigbee2mqtt_0xDEVICEID'], 'manufacturer': 'TuYa', 'model': 'LoraTap garage door opener with wireless sensor (GDC311ZBQ1)', 'name': 'LoraTap Garage Door Opener'}, 'device_class': 'garage', 'name': 'LoraTap Garage Door Opener garage door contact', 'payload_off': False, 'payload_on': True, 'state_topic': 'zigbee2mqtt/LoraTap Garage Door Opener', 'unique_id': '0xDEVICEID_garage_door_contact_zigbee2mqtt', 'value_template': '{{ value_json.garage_door_contact }}', 'platform': 'mqtt'},)
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 246, in async_discover
config = schema(discovery_payload)
File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__
return self._compiled([], data)
File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict
return base_validate(path, iteritems(data), out)
File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: expected BinarySensorDeviceClass for dictionary value @ data['device_class']
```

### Summary of fix
I am able to resolve this issue by changing the `matsee_garage_door_opener` `fromZigbee` converter to return a `contact` result instead of the custom `garage_door_contact`.

### Original `fromZigbee.js`
Below is the [snippet of lines 7754 - 7778 of `fromZigbee.js`](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js):

```
    matsee_garage_door_opener: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataReport', 'raw'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const value = tuya.getDataValue(dpValue);
                switch (dpValue.dp) {
                case tuya.dataPoints.garageDoorTrigger:
                    result.action = 'trigger';
                    break;
                case tuya.dataPoints.garageDoorContact:
                    result.garage_door_contact = Boolean(!value);
                    break;
                case tuya.dataPoints.garageDoorStatus:
                    // This reports a garage door status (open, closed), but it is very naive and misleading
                    break;
                default:
                    meta.logger.debug(`zigbee-herdsman-converters:matsee_garage_door_opener: NOT RECOGNIZED ` +
                        `DP #${dpValue.dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    },
```

### Proposed change to `fromZigbee.js`
Below is the [snippet of lines 7754 - 7778 of `fromZigbee.js`](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js), but this time with the proposed change to line 7766:

```
    matsee_garage_door_opener: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataReport', 'raw'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const value = tuya.getDataValue(dpValue);
                switch (dpValue.dp) {
                case tuya.dataPoints.garageDoorTrigger:
                    result.action = 'trigger';
                    break;
                case tuya.dataPoints.garageDoorContact:
                    result.contact = Boolean(!value);
                    break;
                case tuya.dataPoints.garageDoorStatus:
                    // This reports a garage door status (open, closed), but it is very naive and misleading
                    break;
                default:
                    meta.logger.debug(`zigbee-herdsman-converters:matsee_garage_door_opener: NOT RECOGNIZED ` +
                        `DP #${dpValue.dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    },
```

### New external converter for LoraTap GDC311ZBQ1
Below is the revised external converter (proposed for LoraTap GDC311ZBQ1) that leverages the exposed `contact` and works with Home Assistant. However, similar changes would need to be made to other converters that use `fz.matsee_garage_door_opener` like the one for the MatSee Plus PJ-ZGD01.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_wfxuhoea'}],
    model: 'GDC311ZBQ1',
    vendor: 'TuYa',
    description: 'LoraTap garage door opener with wireless sensor',
    fromZigbee: [fz.matsee_garage_door_opener, fz.ignore_basic_report],
    toZigbee: [tz.matsee_garage_door_opener, tz.tuya_data_point_test],
    whiteLabel: [{vendor: 'LoraTap', model: 'GDC311ZBQ1'}],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [exposes.binary('trigger', ea.STATE_SET, true, false).withDescription('Trigger the door movement'),
      e.contact()],
};

module.exports = definition;
```

### What did you expect to happen?

Exposed Z2M entities should work in Home Assistant through the standard integration via MQTT.

### How to reproduce it (minimal and precise)

1. Use converter that is similar to MatSee Plus PJ-ZGD01.
2. Confirm this works in Zigbee2Mqtt.
3. Start Home Assistant. Confirm that device's `garage_door_contact` entity doesn't exist.
4. Check logs and see similar error to above.

### Zigbee2MQTT version

1.25.0 commit: 6f1460e

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee USB Dongle Plus (zStack3x0)

### Debug log

This doesn't result in an error on the Zigbee2Mqtt end. Refer to above for error from Home Assistant.

Koenkk, 2022-04-02:
Should be fixed now, can you confirm?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

fsevilla3, 2022-04-03:
Thanks for the quick action @Koenkk. I loved how you fixed the problem and learned something new today. More appropriate icon than with my hack using `e.contact()`, as well.

I confirm that my original issue is fixed using `1.25.0-dev commit: [b84ad45](https://github.com/Koenkk/zigbee2mqtt/commit/b84ad45)`. However, this raises a new issue with the converter for the MatSee Plus PJ-ZGD01 ([lines 1909 - 1923 of zigbee-herdsman-converters/devices/tuya.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js)).

Line 1922 reads as follows: `exposes.binary('garage_door_contact', ea.STATE, true, false)],`

In Z2M `garage_door_contact` is `True` when closed and `False` when open. However -- using my custom converter for my LoraTap GDC311311ZBQ1 but using the MatSee Plus PJ-ZGD01 converter as a pattern -- this results in an incorrect status in HA. HA shows `Open` when `garage_door_contact` is `True` (which is when the garage door is physically closed).

By changing the line to read as follows, which is aligned with what `e.contact()` uses, the HA status is correct: `exposes.binary('garage_door_contact', ea.STATE, false, true).withDescription('Indicates if the garage door contact is closed (= true) or open (= false)')],`

I'll be using this in the custom converter I'll be submitting for the LoraTap GDC311311ZBQ1, but I think it's worth checking and fixing with the MatSee Plus PJ-ZGD01 as well. Please let me know if you agree. Thanks!

Koenkk, 2022-04-03:
`garage_door_contact` should be true when the door is closed, since there is contact when the garage door is closed.

fsevilla3, 2022-04-03:
Thanks @Koenkk. I'm sorry if I wasn't clear. In Z2M, `garage_door_contact` is `true` when the door is closed with both cases of `exposes.binary(...)` that I mentioned. However, Home Assistant interprets them differently:
- Home Assistant gets the incorrect status with the way line 1922 of `tuya.js` is currently set up (`exposes.binary('garage_door_contact', ea.STATE, true, false)],`), so `true` in Z2M becomes `Open` in HA and `false` in Z2M become `Closed` in HA
- Home Assistant gets the correct status with my proposed change (`exposes.binary('garage_door_contact', ea.STATE, false, true).withDescription('Indicates if the garage door contact is closed (= true) or open (= false)')],`), so `true` in Z2M becomes `Closed` in HA and `false` in Z2M becomes `Open` in HA

fsevilla3, 2022-04-03:
Thanks @Koenkk. I don't have a MatSee Plus PJ-ZGD01 so I can't test this. However, I've tested my pull request that you merged for the LoraTap GDC311ZBQ1 using Zigbee2MQTT 1.25.0-dev commit: [e251fc0](https://github.com/Koenkk/zigbee2mqtt/commit/e251fc0), and it works perfectly, so this latest change doesn't break that.

Hi @octa22. Reaching out to you since you contributed the support for the MatSee Plus PJ-ZGD01. Firstly, thanks for your contribution, which I leveraged for the LoraTap GDC311ZBQ1. @Koenkk has made some changes that will affect how the PJ-ZGD01 status is reported to Home Assistant. Is this something you're able to test, or are we all happy to just close this issue for now? Thanks!

TEF2one, 2022-04-04:
Oh, looks I may have reported the same issue again [here](https://github.com/Koenkk/zigbee2mqtt/issues/12066), however It my case it was working fine earlier and not anymore... was this introduced with the new version of Z2M?

filin20, 2022-04-05:
hi, just checked it with PJ-ZGD01 and it is working correctly. now door contact has correct icon and Open/Closed status

fsevilla3, 2022-04-09:
Thanks @filin20!

closed 2022-04-09 by fsevilla3