Koenkk/zigbee2mqtt #12039: OTA upgrade of tuya-tv02-zigbee

21stce, 2022-04-02:
### What happened?

I clicked on "Check for updates" on the OTA web site being hosted by zigbee2mqtt, the result is:

> Failed to check if update available for 'my_trv_name' (No image available for imageType '5634')

### What did you expect to happen?

I expected a successful check and then some kind of interface saying whether I really want to do a firmware upgrade.

### How to reproduce it (minimal and precise)

I clicked on "Check for updates" on the OTA web site being hosted by zigbee2mqtt. 

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

6.7.8.0 build 373

### Adapter

Hacked Silvercrest 

### Debug log

Manufacturer: _TZE200_hue3yfsn
Model: https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#tuya-tv02-zigbee

Koenkk, 2022-04-02:
This means we don't have an OTA update yet, someone has to grab it from the TuYa bridge, see: https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html

21stce, 2022-04-02:
Thank you. 

I noticed the following spelling error: `If an firmware upgarde` on that page.

Koenkk, 2022-04-02:
thanks, fixed