Koenkk/zigbee2mqtt #12050: #FailureMessage Object: 0xbebe163c

erikvh, 2022-04-03:
### What happened?

After update Z2M will not start. I get the following error:

#FailureMessage Object: 0xbef6163c
Using '/app/data' as data directory

#
# Fatal error in , line 0
# unreachable code

### What did you expect to happen?

That Z2M would start without error

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20211210

### Adapter

Slaesh's CC2652RB stick

Koenkk, 2022-04-03:
See first note from 1 (https://www.zigbee2mqtt.io/guide/installation/02_docker.html#notes-for-raspberry-pi-users)

samvanbrussel, 2022-04-06:
@Koenkk, I also have the same issue: 1.24 was working, 1.25 is not. Library was already installed:

> pi@rp4-buster-4g:~/Projects/Zigbee2Mqtt $ sudo apt install libseccomp2
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> libseccomp2 is already the newest version (2.3.3-4).
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

I did `docker-compose pull && docker-compose up -d` and here is the output of `docker logs <container_id>`:

pi@rp4-buster-4g:~/Projects/Zigbee2Mqtt $ docker logs 29
Using '/app/data' as data directory

> #
> # Fatal error in , line 0
> # unreachable code
> #
> #
> #
> #FailureMessage Object: 0xbe9a063c
> Using '/app/data' as data directory
> 
> 
> #
> # Fatal error in , line 0
> # unreachable code
> #
> #
> #
> #FailureMessage Object: 0xbeb3963c
> Using '/app/data' as data directory
> 
> 
> #
> # Fatal error in , line 0
> # unreachable code
> #
> #
> #
> #FailureMessage Object: 0xbea6b63c
> Using '/app/data' as data directory
> 
> 
> #
> # Fatal error in , line 0
> # unreachable code
> #
> #
> #
> #FailureMessage Object: 0xbeb4063c
> Using '/app/data' as data directory
> 
> 
> #
> # Fatal error in , line 0
> # unreachable code
> #
> #
> #
> #FailureMessage Object: 0xbef3c63c
> Using '/app/data' as data directory
> 
> 
> #
> # Fatal error in , line 0
> # unreachable code
> #
> #
> #
> #FailureMessage Object: 0xbeaf863c
> Using '/app/data' as data directory
> 
> 
> #
> # Fatal error in , line 0
> # unreachable code
> #
> #
> #
> #FailureMessage Object: 0xbea0463c
> Using '/app/data' as data directory
> 
> 
> #
> # Fatal error in , line 0
> # unreachable code
> #
> #
> #
> #FailureMessage Object: 0xbebea63c
> Using '/app/data' as data directory
> 
> 
> #
> # Fatal error in , line 0
> # unreachable code
> #
> #
> #
> #FailureMessage Object: 0xbef5c63c
> Using '/app/data' as data directory
> 
> 
> #
> # Fatal error in , line 0
> # unreachable code
> #
> #
> #
> #FailureMessage Object: 0xbee5163c
> Using '/app/data' as data directory
> 
> 
> #
> # Fatal error in , line 0
> # unreachable code
> #
> #
> #
> #FailureMessage Object: 0xbebd663c
> Using '/app/data' as data directory

Koenkk, 2022-04-06:
@samvanbrussel did you execute all commands? (not just the apt install libseccomp2)

```
sudo apt-key adv --keyserver hkps://keyserver.ubuntu.com:443 --recv-keys 04EE7237B7D453EC 648ACFD622F3D138
echo "deb http://httpredir.debian.org/debian buster-backports main contrib non-free" | sudo tee -a "/etc/apt/sources.list.d/debian-backports.list"
sudo apt update
sudo apt install libseccomp2 -t buster-backports
```

Koenkk, 2022-04-07:
Could you try upgrading to raspbian bullseye?

BabyDino, 2022-04-09:
Same issue here. Lib was already installed.

Just installed bullseye. Seems to fix this issue

closed 2022-05-17 by github-actions[bot]

segaura, 2022-06-12:
> Don't know why it doesn't work, could you try upgrading to raspbian bullseye?

In my case the suggested solution (adoption of the backported library in buster) worked, thank you.
...actually it also allow me to update my pihole image, which had the same issue :-)