Koenkk/zigbee2mqtt #12052: Hive SLR2b without thermostat never changes running state

phoenix1589, 2022-04-03:
### What happened?

I've just got a Hive SLR2b receiver; no thermostat (yet) which I can get into z2m and I can publish the settings I want to it (emergency heating, heat to this temperature etc) for both the heating and water - however in all cases, the running_state remains at idle:

My current state is:
`{
    "linkquality": 240,
    "local_temperature_heat": 21,
    "local_temperature_water": 21,
    "occupied_heating_setpoint_heat": 29,
    "occupied_heating_setpoint_water": 32,
    "running_state_heat": "idle",
    "running_state_water": "idle",
    "system_mode_heat": "heat",
    "system_mode_water": "heat",
    "temperature_setpoint_hold_duration_heat": 32,
    "temperature_setpoint_hold_duration_water": 65535,
    "temperature_setpoint_hold_heat": true,
    "temperature_setpoint_hold_water": true
}`

Neither light comes on the receiver when I publish the message which I'm doing in a single message like `-m "{\"system_mode\":\"emergency_heating\",\"temperature_setpoint_hold_duration\":\"5\",\"temperature_setpoint_hold\":\"1\",\"occupied_heating_setpoint\":\"29\"}"`

If I press the button on the front of the receiver; it lights up, I hear the relay click, and I get an update to `"running_state_water": "heat"`

I've seen other issues people have had without a thermostat not being able to set the target temperature of the heating (always at 1) - I'm not having this issue and the occupied heating point is 29 as I set.

Anyone seen this / got any thoughts?

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

0x26720700

### Adapter

Conbee2

robertalexa, 2022-04-05:
> Neither light comes on the receiver when I publish the message which I'm doing in a single message like `-m "{\"system_mode\":\"emergency_heating\",\"temperature_setpoint_hold_duration\":\"5\",\"temperature_setpoint_hold\":\"1\",\"occupied_heating_setpoint\":\"29\"}"`

I am assuming you are sending this to the specific endpoint, in this case heat : `zigbee2mqtt/FRIENDLY_NAME/heat/set`

As this is a dual channel receiver, you can send the same commands to the generic endpoint but explicitly choosing specific endpoint attributes. e.g. system_mode_heat, temperature_setpoint_hold_duration_heat etc

I am not sure if this will be the solution you are after, Hive is very secretive about the operation of their devices and i has been a bit of a nightmare with a lot of testing involved when I have "revamped" all their products in z2m. The documentation that i wrote has been tested and proven to work for everyone involved, so if it continues failing, my suspicion would point to the lack of remote.

As mentioned above, I suspect (though not personally tested) that the receivers (devices connected to boiler) MUST be paired with the remotes (the thermostats) for the receivers to work correctly and get the correct local_temperature_heat. In absence of it, it will default to 21C behaving just like the water endpoint. This is also a property that you can't write to, so you won't be able to 

Just for clarity, emergency_heating should be used only for the purpose of native boosting, where the duration is the time in minutes for the boost to run, followed by a return to the previous state.

Try https://www.zigbee2mqtt.io/devices/SLR2b.html#set-heating-mode-to-on-heat-endpoint and see if your boiler starts as expected without the remote. In a setup without the remote (that is if the receiver will work as expected), you would need an automation media like HA with a separate temp sensor and have automations turn heating on and off based on that. Otherwise, with a remote, you let the receiver deal with this, your automation just handles the desired temperature. Hope this helps

phoenix1589, 2022-04-05:
Hi,

Thanks for the reply

> I am assuming you are sending this to the specific endpoint, in this case heat : `zigbee2mqtt/FRIENDLY_NAME/heat/set`

That's correct, I'm using the specific endpoint name in the topic.

> Try https://www.zigbee2mqtt.io/devices/SLR2b.html#set-heating-mode-to-on-heat-endpoint and see if your boiler starts as expected without the remote

I had tried this already, and I get exactly the same behaviour. The emergency heating countdown does count down, but at no point does the state actually go to heat (and the light on the receiver never turns green)

>  my suspicion would point to the lack of remote.

That is what I was guessing; so thanks for coming to the same conclusion I did. 

I've just ordered a thermostat off ebay; was hoping to do more testing with just the cheap receiver before paying more but hey-ho. I would want to use the scheduling built into the receiver so any issues with HomeAssistant couldn't impact the house heating/hot water.

I'd also be interested in trying to get the receiver scheduling configurable via z2m; which would then open up the option of using a Hive SLT6 thermostat.

robertalexa, 2022-04-05:
Oh boy you will be up to a great experience :) There is a whole thread with me saying all the bad stuff about the programming on the devices :) and you won't like any of it. But I will give you a summary.

- I have started by using all my Hive products in the Hive software and have discovered flaws, big ones, a lot of inefficiencies. So i decided to do it myself using z2m
- the devices were poorly supported in z2m, so i spent a reeeally long time making sure that as much stuff is exposed to the user
- the 1 single thing that i refused to expose was the native scheduling. It is awful, it gets reset and forgotten by the device, it is actually very restrictive, e.g. very limited number of actions per dayset. There is no documentation (because Hive has a closed environment) and everything is guessing and sniffing. The scheduling that you see in the Hive app, I always suspected it to be software scheduling, and not hardware one :)

So that being said, my personal advice when it comes to heating is to brace yourself :) You will need a "brain" (automation coordinator). In my case, i use HA, but all my automations are in Node-Red (and they work brilliantly for my desire)

Have a look at this thread. It has hundreds of comments showing everything i mentioned above in detail, and it is not just about Hive SLR1, in fact it is about all of them :) You will also find other software solutions for heating scheduling, you will also meet other people that use the dual channel (so you can ask for opinions) and you can also find screenshots and information about my setup and some other people in this community.
https://github.com/Koenkk/zigbee2mqtt/issues/8184

Overall I find the receiver and the remote I have super at doing what i need them to do. But I tell them what to do and when, instead of using internal schedulers that get erased and throw tantrums :)

I hope this helps you on your journey

phoenix1589, 2022-04-05:
I've skimmed 8184 before; noticed it was long, and that it started with 'device unsupported' and ended with 'device supported'. I'll go through it in more detail and see what fun to expect.

>the 1 single thing that i refused to expose was the native scheduling. It is awful, it gets reset and forgotten by the device, it is actually very restrictive, e.g. very limited number of actions per dayset

That's disappointing; and without the 'hive online service' to potentially fix the scheduling on the device might cause me some issues for what I had planned. I'll certainly do some testing around this.

> Basically it becomes a "classic" thermostat instead of a smart one (with online functionality), but does not break your heating or leave you cold :)

I'll need to think this through a bit more; we currently have a fairly limited schedule time (when the boiler could be on) and make use of the existing honeywell ST9400c 'extra hours' function (equivalent of emergency_heat in hive) so I was thinking of duplicating this with a bit more intelligence of 'where' we are or being able to boost remotely before we get back home.

> Let me know when the remote arrives, if you are still struggling then, i don't mind trying to help you :)

Thank you very much. Hopefully here end of this week, so I can play next week and then cry the week after!

phoenix1589, 2022-04-11:
For this specific issue; as soon as I paired a thermostat then the same messages cause the receiver to turn on/off.

closed 2022-04-11 by phoenix1589