Koenkk/zigbee2mqtt #12053: [New device support]:

gozdagno, 2022-04-03:
### Link

Aliexpress generic GU10 LED RGBW bulb. Manufacturer: it says "KURVIA". Freaking auction expired, so I cannot provide exact link. The internet remains silent about this product.

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xa4c1385178629cf7","nwkAddr":30130,"manufId":4107,"manufName":"KURVIA","powerSource":"Mains (single phase)","modelId":"ZB-CL01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"hwVersion":0,"manufacturerName":"KURVIA","zclVersion":3,"modelId":"ZB-CL01","dateCode":"20211229","deviceEnabled":1}},"lightingColorCtrl":{"attributes":{"currentHue":0,"currentSaturation":0,"colorCapabilities":25,"colorTempPhysicalMin":250,"currentY":24701,"currentX":24939,"enhancedColorMode":2,"colorTemperature":454}},"genIdentify":{"attributes":{"identifyTime":0}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"startUpOnOff":1,"offWaitTime":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":2,"hwVersion":0,"dateCode":"20211229","swBuildId":"2.0.1","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1648981628183,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to add this device, but I failed because I don't know how to fill the  fromZigbee: [], toZigbee: [] and exposes:.
I can read the data via the z2m console, but I cannot write anything. E.g. I can read the min and max levels of color temp (Info Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}), but I cannot set it to another value (Error Publish 'set' 'write' to '0xa4c1385178629cf7' failed: 'Error: Write 0xa4c1385178629cf7/1 lightingColorCtrl({"colorTemperature":420}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'READ_ONLY')')

My e-mail address in case you need it: gozdagno@gmail.com

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZB-CL01'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'Shit', // Vendor model number, look on the device for a model number
    vendor: 'KURVIA', // Vendor of the device (only used for documentation and startup logging)
    description: 'Chinese shitty led bulb', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off], // We will add this later
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.light()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;

i tried some other versions. From and to Zigbee blank, other entries than on_off etc but everything failed.
```

### Supported color modes

Color temperature and RGBW

### Color temperature range

{"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}

Koenkk, 2022-04-03:
Can you try with:

```js
const definition = {
    zigbeeModel: ['ZB-CL01'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'Shit', // Vendor model number, look on the device for a model number
    vendor: 'KURVIA', // Vendor of the device (only used for documentation and startup logging)
    description: 'Chinese shitty led bulb', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
   extend: extend.light_onoff_brightness_colortemp_color(),
};
```

azuanaa, 2022-04-04:
I have 3 of these bulb. I use the following:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const definition = {
    zigbeeModel: ['ZB-CL01'],
    model: 'ZN323128',
    vendor: 'KURVIA',
    description: 'GU10 Spotlight',
    extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [250, 454]}),
    meta: {applyRedFix: true, enhancedHue: false},
};
module.exports = definition;
```

So far so good

gozdagno, 2022-04-04:
@azuanaa Thank you! It worked.

Now it is the time for me to understand the code. Why there is no fromZigbee, toZigbee and exposes keywords like dhey did here: [https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device](url)?
Instead you both used extend and azuanaa used meta. Can you tell me why is it like that or provide some documentation?

sjorge, 2022-04-04:
The `extend: ` is a little code saving feature, it extends some existing templates.
e.g. the one used is found here https://github.com/Koenkk/zigbee-herdsman-converters/blob/dda3ab3d584f46dc6d48a91aba0bedf36386d30a/lib/extend.js#L70

We mostly have these for lights as almost all lights fit into one of the following categories:
- on/off
- on/off + brightness
- on/off + brightness + colorTemp
- on/off + brightness + color
- on/off + brightness + colorTemp + color

So instead of repeating the same thing over and over again, you use the extend. The added bonus if you need to update one of the templates it applies to all devices using the extend!

As for meta, some devices have some quirks and we tell the converts about that using the meta item.
They are listed in the readme.md in the zhc repo https://github.com/Koenkk/zigbee-herdsman-converters

```
applyRedFix: basically avoid using a particular set of XY values that represents 'pure red' that get ignored by some devices
enhancedHue: by default the convert assumes the device uses enhancedHue for finer color control, but not all devices support this so unsetting will just make it use XY, there is a paired option to enable to fallback to non-enhancedHue... granted this one could use a bit of a rework at somepoint I guess.
```

closed 2022-04-04 by gozdagno