Koenkk/zigbee2mqtt #12057: [Feature request]: Support option to enable "--unhandled-rejections=strict" on startup of zigbee2mqtt

Moltey, 2022-04-03:
### Is your feature request related to a problem? Please describe

I experience problems in my configuration shown below:

- **podman** is utlized to run a zigbee2mqtt container
- zigbee2mqtt container is **unprivileged** 
- **Conbee II** is used on /dev/ttyACM0

Once in a while (e.g. sometimes after a bootup), no zigbee device is responsive. 
Zigbee2mqtt is throwing this message:

```
(node:2) UnhandledPromiseRejectionWarning: Error: Error while opening serialport 'Error: Error: Permission denied, cannot open /dev/ttyACM0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:182:28)
    at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
(node:2) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 61)
```

If I restart the container, everything works fine (for weeks / months):
```
$ podman restart zigbee2mqtt
```

### Describe the solution you'd like

Implement an option to activate "--unhandled-rejections=strict" on startup of npm, as suggested in the error message.

Reason: Zigbee2MQTT should fail and be restarted automatically.

### Describe alternatives you've considered

I considered to check the log file consistently via script and restart the container if the error message is thrown. However, this is a very ugly option.

### Additional context

--- none ---

Impact123, 2022-04-03:
I'm not familiar with podman but this might already be possible by overriding [the cmd](https://github.com/Koenkk/zigbee2mqtt/blob/6f1460e47b430f8e1fe22819da56a238f8c92174/docker/Dockerfile#L37) like this
```
podman run ... koenkk/zigbee2mqtt /sbin/tini -- node --unhandled-rejections=strict index.js
```
Not that I think this is the right solution

Moltey, 2022-04-03:
Thank you for your answer. I will try your advice as soon as I am back home.

> Not that I think this is the right solution

I am curious: How would your suggested approach look like? I thought it would be beneficial to have a failing container to utilize the internal watchdog of docker / podman / k8s to restart the container automatically.

Impact123, 2022-04-03:
Sorry if I was too vague. I expect fatal/unrecoverable errors like this to exit the app/container so it can be restarted too, I'm just not sure if this flag is the best approach. I'm not sure what's the "best" solution.

Moltey, 2022-05-04:
I tried it and it works very well. The container restarts itself as itended.

How about defining this as a standard for the container startup?

closed 2022-06-12 by github-actions[bot]