Koenkk/zigbee2mqtt #12062: Nue 3A Single/Double Gang Switch Not receiving Data request failed with error: 'MAC no ack' (233)

trentbraidner, 2022-04-03:
### What happened?

Recently my Nue/3A Single and Double gang switches are failing.
When the switch is manually activated the state and LQI are updated in Z2M however when trying to send a command I am receiving the error `Data request failed with error: 'MAC no ack' (233)`

All my other Nue/3A switches work as expected and havent had any issues.

its only the 
3x LXN-2S27LX1.0
1x LXN-1S27LX1.0 with the issue

I have completely deleted the db and started with a fresh config, still occurring.

### What did you expect to happen?

Up until last week the switches were operating correctly all commands were sent and received, no new devices added to the network, ZZH is on an USB extension, have tried additional ports and extension cables.

Expectation is that a command is sent and that it is received.
The switch can report all states and LQI to the coordinator no issues

### How to reproduce it (minimal and precise)

Attempt any send command to the device and they fail with `Data request failed with error: 'MAC no ack' (233)`

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

 20220219

### Adapter

zzh

### Debug log

2022-04-03T22:31:34.614Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,14,36,1,149,242,1,1,25,0,60,0,30,4,25,21,2,152,229]
2022-04-03T22:31:34.614Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-04-03T22:31:34.626Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-04-03T22:31:34.626Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-04-03T22:31:34.626Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-04-03T22:31:34.626Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-04-03T22:31:34.626Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-04-03T22:31:34.635Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,60,250]
2022-04-03T22:31:34.636Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,60,250]
2022-04-03T22:31:34.636Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,60] - 250
2022-04-03T22:31:34.636Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":60}
2022-04-03T22:31:34.636Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-04-03T22:31:35.608Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x149f0111e00bfdb0:1107/1 (0,4,1)
2022-04-03T22:31:35.608Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":1107,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":61,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,81,11,10,0]}}
2022-04-03T22:31:35.609Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,83,4,1,1,6,0,61,0,30,5,24,81,11,10,0,21]
2022-04-03T22:31:35.621Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-04-03T22:31:35.621Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-04-03T22:31:35.621Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-04-03T22:31:35.621Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-04-03T22:31:35.621Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-04-03T22:31:35.687Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,233,1,61,18]
2022-04-03T22:31:35.687Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,233,1,61,18]
2022-04-03T22:31:35.687Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [233,1,61] - 18
2022-04-03T22:31:35.687Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":233,"endpoint":1,"transid":61}
2022-04-03T22:31:35.687Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-04-03T22:31:35.687Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x149f0111e00bfdb0:1107,233,4)
2022-04-03T22:31:35.688Z zigbee-herdsman:controller:endpoint DefaultResponse 0x149f0111e00bfdb0/1 6(10, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))
2022-04-03T22:31:35.688Z zigbee-herdsman:controller:device:error Default response to 0x149f0111e00bfdb0 failed

johnnykang, 2022-04-07:
i have the exact same issues with the brand new Nue 4 gang and 2 gang wall switches. 

It never really worked reliably and i tried the devices with both Philip Hue and zigbee2mqtt and I have the same issues on both coordinator. 

any clue?

trentbraidner, 2022-04-07:
Kevin from 3asmarthome seemed to think there was an issue with the Herdsman converters and provided me with this file
[nue_3a.txt](https://github.com/Koenkk/zigbee2mqtt/files/8439649/nue_3a.txt)

I didnt save my final revision as it had some validation errors, but i replaced my herdsman converter file with this on a separate system and had the same issues.

Ill pair the device, it works for maybe 5-10mins and then it wont work anymore.

My 3 gang switch and all my dimmers are perfect however.

I _suspect_ this isnt a herdsman converter/device profile issue due to the fact that even upon restoring from an earlier backup and restoring an earlier version of my zzh firmware from when this did work I couldnt resolve the issue.

Kevin has said he is going to test the devices in stock for the same issue but was waiting for his Conbee II to arrive

johnnykang, 2022-04-07:
ok. We are emailing to the same guy then.

Well, i don't think it is herdman or zigbee2mqtt fault. Because I tried the switches with Philips Hue and same issues happened. It paired and worked (I am able to turn on and off ), then after 5 mins or so,  I can't issue commands from Philip Hue to the switches. Philips Hue shows 'device unavailable' and while on zigbee2mqtt ' it displayed 'mac no ack' .

i bought the 2 and 4 gang from Kevin brand new, and they have never worked reliably.

trentbraidner, 2022-04-07:
thanks for doing the test with the hue bridge, I was about to dig mine out to test. Glad I don’t have to.

Interesting yours never worked reliably. I’ve had nothing but excellent performance since I installed them expect for last week they started failing. But only my 1 gang, 2 gang and power points.

My 3 gang which has been installed for 6 months and 1/2 gang dimmers still work perfect

Regards,
Trent Braidner

johnnykang, 2022-04-11:
@trentbraidner did you sort out the switch issue? Have you heard back from Kevin as well?

jyavenard, 2022-04-14:
I opened #12169 and mine too worked fine for several weeks and only started to fail last week  

unbelievable that they would all stop working around the same time with different coordinators/stack

did they request a firmware update somehow? (Didn’t think that was possible)

jyavenard, 2022-04-14:
> thanks for doing the test with the hue bridge, I was about to dig mine out to test. Glad I don’t have to. Interesting yours never worked reliably. I’ve had nothing but excellent performance since I [...]

Is this the new version that supports 2-way? (They have an extra button to sync a gang on two different switches)

trentbraidner, 2022-04-17:
> Is this the new version that supports 2-way? (They have an extra button to sync a gang on two different switches)

Nah this is the non two way switches.
It's happening with the 1/2 gang switches as well as the 2 gpo powerpoints

jyavenard, 2022-04-18:
> Nah this is the non two way switches.
> It's happening with the 1/2 gang switches as well as the 2 gpo powerpoints

Weird. The only failing for me are those new 2-ways one. I have 14 other Nue switches in my house, all working fine.

smulle48, 2022-04-22:
Could be interference on the channel.

Have had that problem in one end of our house. Found out that my neighbors wifi was interfering a few devices in that end. After scanning the channels in the entire house, I found a channel (26) with almost no interference. After changing the channel all my problems were gone :-)

https://github.com/Koenkk/zigbee2mqtt/issues/2789

jyavenard, 2022-04-22:
> Could be interference on the channel.

Out of my 14 switches; only two are behaving like that; both purchases last months and new generation (support 2-way). 
so I doubt it’s a wireless issue. 

Additionally; zigbee2mqtt perfectly reads the status of the switches. It just can’t set it.

smulle48, 2022-04-22:
> > Could be interference on the channel.
> 
> 
> 
> Out of my 14 switches; only two are behaving like that; both purchases last months and new generation (support 2-way). 
> 
> so I doubt it’s a wireless issue. 
> 
> 
> 
> Additionally; zigbee2mqtt perfectly reads the status of the switches. It just can’t set it. 

I had the same with two bulbs, and they were not even next to each other. I moved them closer to the zigbee stick and within a day, it repaired it self and everything was working.

After that I found the interference and after the channel change they all work

jyavenard, 2022-04-25:
Well, that was a massively painful waste of time.
Changed from channel 11 to 18 (I use wifi on channel 1 and 11).

Over an hour re-pairing everything, those two Nue switches stopped responding to request after just 5 minutes.

And now I have to crawl under various places to repair leak-detector and various thermometers :(

trentbraidner, 2022-04-30:
> @trentbraidner did you sort out the switch issue? Have you heard back from Kevin as well?

@johnnykang Nothing back yet from Kevin, I've not been able to do anything to get these switches to work. Will update this if i have any more info

jyavenard, 2022-04-30:
Kevin said he was going to pass the information to his engineer a couple of weeks ago.  No news since

jyavenard, 2022-05-18:
I posted an update in #12169 . TLDR: Nue firmware update fixed it

trentbraidner, 2022-05-30:
I can confirm that the issue is resolved with a firmware update and not an issue with Z2M

**To resolve if you have this issue**

You will need to have your switches reflashed by Kevin from 3A Smart Home.
I dropped in over the weekend and had the switches flashed in 10 mins.

OTA updates are available for these switches via the Nue Zigbee Bridge (no others at this time)
Im not sure if this can be done OTA though (needs to be confirmed)

For future firmware updates I will be getting a Nue Bridge to temp connect the switches and perform OTA updates

This fixed the issue for my 1 Gang, 2 Gang and Dual GPO power points
3x LXN-2S27LX1.0
1x LXN-1S27LX1.0

closed 2022-05-30 by trentbraidner