Koenkk/zigbee2mqtt #12065: [New device support]:

unpack8866, 2022-04-04:
### Link

http://www.dnake.com/#/product-list?cate2=11&cate1=10

### Database entry

Device type： Router ；Zigbee Model ：CS-Z-CZ-2402； Zigbee Manufacturer ：DNAKE ； IEEE Address：0x00124b000ffcdd41

### Comments

Request to add device support; DNAKE Smart Socket,It works perfectly under the external converter.
[image]
[image]

### External converter

```shell
const device = {
    zigbeeModel: ['CS-Z-CZ-2402'],
    model: 'CS-Z-CZ-2402',
    vendor: 'DNAKE',
    description: 'Smart Socket',
    fromZigbee: [fz.ignore_basic_report, fz.on_off,],
    toZigbee: [tz.ptvo_switch_trigger, tz.on_off,],
    exposes: [e.switch().withEndpoint('l1'),
],
    meta: {
        multiEndpoint: true,
        
    },
    endpoint: (device) => {
        return {
            l1: 1,
        };
    },

module.exports = device;
```

Koenkk, 2022-04-04:
Added!

unpack8866, 2022-04-05:
> Added!
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)
[image]

It works; But the icon is not displayed

Koenkk, 2022-04-05:
This will be added in the next release (1 May)