Koenkk/zigbee2mqtt #12066: [Problem] TuYa PJ-ZGD01 garage_door_contact unavailable

TEF2one, 2022-04-04:
### What happened?

Within Home Assistant using Mosquitto broker, the entity relative to the "garage_door_contact" appears as unavailable.
Within Zigbee2mqtt., the "garage_door_contact" is however correctly updating.

I disable the availability in the case that would have an impact but it did not.

### What did you expect to happen?

Home Assistant to reflect information as Zigbee2mqtt.

### Zigbee2MQTT version

1.25.0-1

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Debug log

Debug Received Zigbee message from 'Gate Opener', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":70}' from endpoint 1 with groupID null
Debug Received Zigbee message from 'Gate Opener', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":3}],"seq":20480}' from endpoint 1 with groupID null

fsevilla3, 2022-04-05:
Hi @TEF2one. I believe this is indeed a duplicate of  #12038.  Are you in a position to test the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)? That should fix the issue with Home Assistant rejecting the information due to an incorrect device class. If you could also please check whether the open/closed status in Home Assistant matches the physical world that would be great. You can see the discussion between me and @Koenkk in #12038 for the background, but because I don't have the MatSee Plus PJ-ZGD01 I'm unable to test with that to compare. Thanks.

filin20, 2022-04-05:
hi @fsevilla3 as mentioned in the original ticket, i've checked fix in the dev branch and it is working flawlessly

TEF2one, 2022-04-06:
Hi @fsevilla3, I'm using home assistant with the Z2M add-on, I could install "Zigbee2mqtt Edge" however I am unsure what would happen with regard to the original Z2M add-on:
- Can they run in parallel, accessing the same Conbee2 dongle at the same time?
- Do they share the same configuration?
- will both add-ons interfere with each other?
- Will I need to migrate/re-sync all, some or none of the devices?
- Do you have a good reference I could use to get started?
Regards,

snoopsbne, 2022-04-10:
@fsevilla3 I just tried via the :latest-dev container and Home assistant reports correctly.