Koenkk/zigbee2mqtt #12071: CC2531: Error: Error while opening serialport

hih4323, 2022-04-04:
### What happened?

Hello,

After reboot my CC2531 won't restart en give me a error (see link):

https://gathering.tweakers.net/forum/list_messages/2084092 (my reaction hih4323).

Thanks

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

last

### Adapter

CC2531

Koenkk, 2022-04-04:
`Zigbee2MQTT:error 2022-04-02 15:38:31: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'` -> another process is locking the port, see https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-resource-temporarily-unavailable-cannot-lock-port

hih4323, 2022-04-04:
What i get is this:

```
pi@raspberrypi:~ $ ls -l /proc/[0-9]/fd/ |grep /dev/ttyACM0
ls: kan map '/proc/1/fd/' niet openen: Toegang geweigerd
ls: kan map '/proc/2/fd/' niet openen: Toegang geweigerd
ls: kan map '/proc/3/fd/' niet openen: Toegang geweigerd
ls: kan map '/proc/4/fd/' niet openen: Toegang geweigerd
ls: kan map '/proc/6/fd/' niet openen: Toegang geweigerd
ls: kan map '/proc/7/fd/' niet openen: Toegang geweigerd
ls: kan map '/proc/8/fd/' niet openen: Toegang geweigerd
ls: kan map '/proc/9/fd/' niet openen: Toegang geweigerd
pi@raspberrypi:~ $ sudo ls -l /proc/[0-9]/fd/ |grep /dev/ttyACM0
pi@raspberrypi:~ $
```

Only with a sudo i get nothing back

closed 2022-05-14 by github-actions[bot]