Koenkk/zigbee2mqtt #12073: Aqara E1 door & window contact sensor [MCCGQ14LM not working properly

RaulVKoNe, 2022-04-04:
### What happened?

The pairing of the (MCCGQ14LM) seems to be done correctly, however, although the interview says that it is done correctly, I do not receive any status. Sometimes I have gotten the LQI, but nothing else, no closed/open status, voltage, battery....

When you gave support for this device months ago I installed 3 without any problem. Now I'm trying with 2 and it's impossible, I've tried anyway.

As you can see in the screenshot, it is a Conbee2 with the latest firmware and the latest version of Zigbee2MQTT. I have it in a USB 2.0 port with an extension cable and I have tried to try to synchronize it both next to it, further away, but there is no way.

[image]
[image]

### Zigbee2MQTT version

1.25

### Adapter firmware version

0x26720700

### Adapter

Conbee2

geoffreylagaisse, 2022-04-08:
same problem here. Restarting zigbee2mqtt helps for about 2 hours. After that the aqura device becomes unresponsive and zigbee2mqtt doesn't handle the device updates anymore.

Koenkk, 2022-04-08:
@geoffreylagaisse do you see any `Received zigbee message` debug logging from the device after no reports come in? ( https://www.zigbee2mqtt.io/guide/usage/debug.html)

- No: seems device stopped reporting, check: https://www.zigbee2mqtt.io/devices/WXKG01LM.html#troubleshooting-device-stops-sending-messages-disconnects-from-network
- Yes: may be caused by https://github.com/Koenkk/zigbee-herdsman-converters/pull/3994

closed 2022-05-17 by github-actions[bot]