Koenkk/zigbee2mqtt #12074: After updating HA addon to 1.25.0-1 and to 20220219 slaesh's firmware most of the network is unreachable

fireport, 2022-04-05:
### What happened?

I updated the addon to 1.25-0.1 on ha OS and at the same time, I update slaesh firmware to  20220219. After restart, I see only 4 devices the rest are unreachable. I tried to downgrade the addon and firmware but this doesn't help 

### What did you expect to happen?

To upgrade without loss of devices

### Zigbee2MQTT version

1.25.0.1

### Adapter firmware version

20220219

### Adapter

Slaesh's _cc2652rb

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8415724/log.txt)

RazoredRaven, 2022-04-05:
90% of my devices are now unreachable (LQI "N/A") since upgrading.

Even rolling back, cannot get them back.

fireport, 2022-04-05:
My devices are slowly starting to become available. Now are about 70% with an LQI value but, sometimes, they  timeout

fireport, 2022-04-07:
now my network is stable. This wasn't the first update that I did but is the first time that I observed this behavior

closed 2022-04-07 by fireport