Koenkk/zigbee2mqtt #12076: SMHM-I1 fix

unpack8866, 2022-04-05:
### What happened?

Request deletion'' e.tamper()'',There is no such thing.
[image]

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20210309

### Adapter

cc2652p

Koenkk, 2022-04-05:
Done.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-04-05 by Koenkk

unpack8866, 2022-04-17:
> Done.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Repair the battery level display！
[image]

const device = {
    zigbeeModel: ['SMHM-I1', 'PIR_TPV12'],
    model: 'SMHM-I1',
    vendor: 'HEIMAN',
    description: 'Smart motion sensor',
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.battery()],  
    meta: {battery: {voltageToPercentage: '3V_2500'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
        await reporting.batteryVoltage(endpoint);
    },
};

module.exports = device;