Koenkk/zigbee2mqtt #12078: [FC80CC] Legrand Energy not reported ?

kevincaradant, 2022-04-05:
### What happened?

In the documentation of the manufacter, I can see the following sentence :

[image]
( highlight in blue ) 
If I translated it, it's something like: "measurement and monitoring of the consumption in (kWh) of the controlled circuit"
But I don't see this reported value from the device. ( only live consumption in watt, as we can also saw in debug log  ) 

Do you know if it's something which is missing as exposed attribute ? 

### What did you expect to happen?

Having the consumption (energy) in Kwh exposed

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20211210

### Adapter

Launchpad CC1352-P2

### Debug log

Debug Received Zigbee message from 'power_air_conditioner_livingroom', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":240}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'power_air_conditioner_livingroom', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":267}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/power_air_conditioner_livingroom', payload '{"device_mode":"switch","linkquality":91,"power":267,"state":"ON"}'
Debug Received Zigbee message from 'power_air_conditioner_livingroom', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":277}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/power_air_conditioner_livingroom', payload '{"device_mode":"switch","linkquality":91,"power":277,"state":"ON"}'

sjorge, 2022-04-06:
Can you try with this external convertor?

```node
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition =     {
        zigbeeModel: [' Contactor\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'+
            '\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'],
        model: 'FC80CC',
        description: 'Legrand (or Bticino) DIN contactor module',
        vendor: 'Legrand',
        extend: extend.switch(),
        fromZigbee: [fz.identify, fz.on_off, fz.electrical_measurement, fz.legrand_cluster_fc01, fz.ignore_basic_report, fz.ignore_genOta, fz.metering],
        toZigbee: [tz.legrand_deviceMode, tz.on_off, tz.legrand_identify, tz.electrical_measurement_power, tz.currentsummdelivered],
        exposes: [exposes.switch().withState('state', true, 'On/off (works only if device is in "switch" mode)'),
            e.power().withAccess(ea.STATE_GET), exposes.enum('device_mode', ea.ALL, ['switch', 'auto'])
                .withDescription('switch: allow on/off, auto will use wired action via C1/C2 on contactor for example with HC/HP'), e.energy().withAccess(ea.STATE_GET)],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.onOff(endpoint);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
            await reporting.activePower(endpoint);

            await reporting.readMeteringMultiplierDivisor(endpoint);
            await reporting.currentSummDelivered(endpoint, {min: 60, change: 1});
        },
    };

module.exports = definition;
```

kevincaradant, 2022-04-09:
Hi @sjorge 

No problem but sorry I didn't find a doc which explain exactly how to do it. 
I found this link :
https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/externalConvertersExample/freepad_ext.js

How can I do it with that ? 

Thanks

sjorge, 2022-04-09:
You can save the above as legrand.js and follow https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters

You replace the example filename(s) with legrand.js

kevincaradant, 2022-04-09:
OK I tried .

I added this following line 
`const extend = require('zigbee-herdsman-converters/lib/extend');`
in the legrand.js file because otherwise the function was not found

Futhermore since the UI , I clicked on the refresh button ( blue ) about energy:
[image]

I got this issue 
```
zigbee2mqtt    | Zigbee2MQTT:info  2022-04-09 20:53:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'energy' to 'power_air_conditioner_livingroom' failed: 'Error: Read 0x0004xxxxxxxxxx/1 seMetering([\"currentSummDelivered\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"power_air_conditioner_livingroom"},"type":"zigbee_publish_error"}'
```

And also, I saw eveytime (even without this custom converter) when I start Z2M, this issue on the device;
```
Zigbee2MQTT:error 2022-04-09 20:49:37: Failed to configure 'power_air_conditioner_livingroom', attempt 2 (ReferenceError: reporting is not defined
zigbee2mqtt    |     at Object.configure (/app/dist/util/externally-loaded.js:23:13)
zigbee2mqtt    |     at Configure.configure (/app/lib/extension/configure.ts:115:37)
zigbee2mqtt    |     at EventEmitter.<anonymous> (/app/lib/extension/configure.ts:80:62)
zigbee2mqtt    |     at EventEmitter.emit (node:events:538:35)
zigbee2mqtt    |     at EventBus.emitLastSeenChanged (/app/lib/eventBus.ts:52:22)
zigbee2mqtt    |     at Controller.<anonymous> (/app/lib/zigbee.ts:66:27)
zigbee2mqtt    |     at Controller.emit (node:events:526:28)
zigbee2mqtt    |     at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:483:14)
zigbee2mqtt    |     at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:694:18)
zigbee2mqtt    |     at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:135:70))
```

I don't know if it's normal or if it's not.

sjorge, 2022-04-09:
Ah good spot, the extend was also needed! Looks like the device does not report Energy via seMetering :( as we get `UNSUPPORTED_ATTRIBUTE` meaning the device tells z2m I don't support this when trying to read the value.

Not really sure where else to try and get the value from, perhaps Legrand is doing something with a manufacturer specific cluster, hard to say. Not sure I can help further with this.

The missing `reporting` error isn't normal, and should probably be looked at too.

kevincaradant, 2022-04-09:
OK I guess seeing yours results, they use the 'current power' to calculate it using their application or something like that.
It's not directly reported by the device :(
It's not a big deal because this feature is implemented in HA directly. I saw this yesterday :)

Mhh about this:
 Edit: figured why it complains about reporting, also adding this near the top should fix that.
const reporting = require('zigbee-herdsman-converters/lib/reporting');

I will try now but when I paired my device the first time in the Z2M UI,  I got a red toast, I thought it was 'reporting' issue. Finally,  I guess it was an other attribute issue if here the 'reporting' issue is just because I forgot a import ;)

pirracas77, 2022-04-13:
@kevincaradant do you have also this issue #12152 ?

kevincaradant, 2022-04-13:
To answer at your question, not sure what should happens. But if I have the switch Off and then I decide to change the mode to auto. In this case nothing happens when I trigger the switch ON/OFF.
Hopping that answer at your question.

pirracas77, 2022-04-14:
> To answer at your question, not sure what should happens. But if I have the switch Off and then I decide to change the mode to auto. In this case nothing happens when I trigger the switch ON/OFF. Hopping that answer at your question.
This is the normal behaivor. The switch ON/OFF is disable when automatic mode is set up.

The issue that I'm describing is diferent : 
switch off the device from the web (or MQTT)
Set up to automatic mode.
At this point, the device would turn on the led to blue in order to verify that is in the correct status "auto". <<<< this part doesn't work. The led is not switch to blue and also it is not able to manage via the external imput C1/C2.

Could you verify this part?

kevincaradant, 2022-04-14:
Yes sorry I was not clear on my previous answer but I wanted to say it's stay at off everytime.
Yes I agree that the switch could not respond on auto mode but it never go on blue light even after to set at auto.
On my case its not an issue because I'm currently don't use it. But the behavior is the same as you

pirracas77, 2022-04-14:
> Yes sorry I was not clear on my previous answer but I wanted to say it's stay at off everytime. Yes I agree that the switch could not respond on auto mode but it never go on blue light even after to [...]

Thanks for your confirmation

closed 2022-05-23 by github-actions[bot]