Koenkk/zigbee2mqtt #12080: "Exception while calling fromZigbee converter" when adding Develco/Frient SMSZB-120

okastl, 2022-04-05:
### What happened?

When adding a new Develco/Frient SMSZB-120 smoke detector to the network, z2m shows the exception "Exception while calling fromZigbee converter: Cannot set property 'softwareBuildID' of undefined}"
However, I don't see any functional problem The device is added, and works just fine.

Here is an excerpt from the log:
```
debug 2022-04-05 15:19:50: Received Zigbee message from '0x0015bc00310098f6', type 'readResponse', cluster 'genPowerCfg', data '{"batteryVoltage":29}' from endpoint 35 with groupID 0
info  2022-04-05 15:19:50: MQTT publish: topic 'zigbee2mqtt/0x0015bc00310098f6', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":null,"ias_cie_address":"0x00124b0022a54eff","last_seen":1649164790471,"linkquality":87,"max_duration":null,"reliability":null,"restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":null,"test":false,"update":{"state":null},"update_available":null,"voltage":2900}'
debug 2022-04-05 15:19:50: Received Zigbee message from '0x0015bc00310098f6', type 'readResponse', cluster 'genBasic', data '{"32768":{"data":[4,0,2],"type":"Buffer"},"32784":{"data":[3,4,25],"type":"Buffer"},"32800":{"data":[5,0,2],"type":"Buffer"}}' from endpoint 35 with groupID 0
error 2022-04-05 15:19:50: Exception while calling fromZigbee converter: Cannot set property 'softwareBuildID' of undefined}
info  2022-04-05 15:19:50: MQTT publish: topic 'zigbee2mqtt/0x0015bc00310098f6', payload '{"alarm":null,"battery":80,"battery_low":false,"enrolled":true,"fault":null,"ias_cie_address":"0x00124b0022a54eff","last_seen":1649164790588,"linkquality":83,"max_duration":null,"reliability":null,"restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":null,"test":false,"update":{"state":null},"update_available":null,"voltage":2900}'
```

### What did you expect to happen?

No exception thrown.

### How to reproduce it (minimal and precise)

Add a Develco/Frient SMSZB-120 to the network.

### Zigbee2MQTT version

1.25.0 commit: 6f1460e4

### Adapter firmware version

20211217

### Adapter

Slaesh CC2652RB

Koenkk, 2022-04-05:
Can you check if this is fixed now?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

okastl, 2022-04-05:
> Can you check if this is fixed now?
> 

Just tried 1.25.0-dev commit: [f91d0f6b](https://github.com/Koenkk/zigbee2mqtt/commit/f91d0f6b) and the problem is fixed.
Thank you!

closed 2022-04-05 by okastl

labaland, 2022-10-28:
> 

MY wont event connect!! :( 
Plase help how did you do? It did work in deconz. I holding in the reset button and pressing it multible times but wont connecnt :( 

swede? 
Mvh Labaland