Koenkk/zigbee2mqtt #12085: [New device support]: Innr Outdoor Smart Globe Lights OGL 130C

Mark-Sellwood, 2022-04-05:
### Link

https://www.innr.com/en/product/outdoor-smart-globe-lights-uk/

### Database entry

{"id":8,"type":"Router","ieeeAddr":"0x00158d000197f5ff","nwkAddr":55199,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"OGL 130 C","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"OGL 130 C","manufacturerName":"innr","powerSource":1,"zclVersion":2,"appVersion":41,"stackVersion":33,"hwVersion":2,"dateCode":"20210828","swBuildId":"2.9.9"}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":49246,"epId":2,"devId":4096,"inClusterList":[4096],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":41,"stackVersion":33,"hwVersion":2,"dateCode":"20210828","swBuildId":"2.9.9","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1649167565277,"defaultSendRequestWhen":"immediate"}

### Comments

I suspect that these are very similar to the INNR OPL 130 C or OSL 130 C

Koenkk, 2022-04-05:
Please fill in the supported color modes + color temperature range.

Mark-Sellwood, 2022-04-05:
I have added the color temperature range.
How do I find out the Supported color modes?

sjorge, 2022-04-06:
For the color_temp range you need to read it using z2m:
https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

Mark-Sellwood, 2022-04-06:
Thanks, just done that and got the below:
Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":100,"colorTempPhysicalMax":1000}

closed 2022-04-06 by Koenkk

Mark-Sellwood, 2022-04-09:
Many thanks, works fine.