Koenkk/zigbee2mqtt #12088: GledOpto lights join succesfull and leave network after a few seconds (over and over again)

vinnie99999, 2022-04-06:
### What happened?

I'am a bit new with Zigbee but I have been testing a lot, because I want to replace some MiLight bulbs with GeldOpto bulbs. The problem I am facing is that the GeldOpto bulbs are paired successfully in the first place and I am able to control them. But it looks like as soon as I delete a successfully added light, the issue is starting to occur when I want to rejoin this light. The light will be paired successfully but after a few seconds the message appears "Device <IEEE Address> left the network". This join and disjoin process will keep on going over and over again.

I have this issue with different types:
- GL-S-006P (GU10 RGB+CCT)
- GL-S-004P (MR16 RGB+CCT)
- GL-B-002P (E27 dual white)

I am not sure which part of the Zigbee implemenation is causing this (hardware or software) so I hope someone can help troubleshooting on this issue or maybe better providing a solution. Please let me know if you need any more information?

I have already tried to reset the light by turning it on and off for 5 times (leave it on at the 5th time).

Thank you in  advance

### What did you expect to happen?

When I delete a succesfully added light, I expect that when I join it again it stays added without the light leaving the network.

### How to reproduce it (minimal and precise)

Pair GL-S-006P light (or probably an other type as well)
Remove device/light
Pair device again and the issue should start to occur

### Zigbee2MQTT version

1.25.0-dev

### Adapter firmware version

20211217

### Adapter

zStack3x0  Tube's Zigbee Gateways (CC2652P2 variant) Ethernet/POE model

### Debug log

debug 2022-04-06 13:49:04: Retrieving state of '0xa4c138bd77e72d9f' after reconnect
debug 2022-04-06 13:49:04: Received Zigbee message from '0xa4c138bd77e72d9f', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 11 with groupID 0
info  2022-04-06 13:49:04: MQTT publish: topic 'zigbee2mqtt/0xa4c138bd77e72d9f', payload '{"color_mode":"xy","linkquality":102,"state":"ON"}'
warn  2022-04-06 13:49:24: Device '0xa4c138bd77e72d9f' left the network
info  2022-04-06 13:49:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138bd77e72d9f","ieee_address":"0xa4c138bd77e72d9f"},"type":"device_leave"}'
info  2022-04-06 13:49:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xa4c138bd77e72d9f"},"type":"device_removed"}'
debug 2022-04-06 13:49:24: Received MQTT message on 'zigbee2mqtt/bridge/config/devices/get' with data ''
info  2022-04-06 13:49:24: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"dateCode":"20211217","friendly_name":"Coordinator","ieeeAddr":"0x00124b00259060db","lastSeen":1649245764381,"networkAddress":0,"softwareBuildID":"zStack3x0","type":"Coordinator"},{"dateCode":"20220223","description":"Zigbee 4W MR16 Bulb RGB+CCT (pro)","friendly_name":"Spot01_Souterain_WC","hardwareVersion":2,"ieeeAddr":"0xa4c138e39b946840","lastSeen":1649079717402,"manufacturerID":4687,"manufacturerName":"GLEDOPTO","model":"GL-S-004P","modelID":"GL-S-004P","networkAddress":19095,"powerSource":"Mains (single phase)","softwareBuildID":"10151203","type":"Router","vendor":"Gledopto"},{"dateCode":"20201021","description":"Zigbee 4W MR16 Bulb RGB+CCT (pro)","friendly_name":"unknown","hardwareVersion":0,"ieeeAddr":"0xbc33acfffe8b93f5","lastSeen":1648659430031,"manufacturerID":4687,"manufacturerName":"GLEDOPTO","model":"GL-S-004P","modelID":"GL-S-004P","networkAddress":5085,"powerSource":"Mains (single phase)","softwareBuildID":"V_1_1","type":"Router","vendor":"Gledopto"},{"dateCode":"20210712","description":"-","friendly_name":"0xa4c13829498fd349","hardwareVersion":0,"ieeeAddr":"0xa4c13829498fd349","lastSeen":1648643307648,"manufacturerID":4687,"manufacturerName":"GLEDOPTO","model":"GL-B-002P","modelID":"GL-B-002P","networkAddress":56482,"powerSource":"Mains (single phase)","softwareBuildID":"10051203","type":"Router","vendor":"-"},{"dateCode":"20210414","description":"Zigbee 4W MR16 Bulb RGB+CCT (pro)","friendly_name":"0x84fd27fffe8aa39a","hardwareVersion":2,"ieeeAddr":"0x84fd27fffe8aa39a","lastSeen":1649079807783,"manufacturerID":4687,"manufacturerName":"GLEDOPTO","model":"GL-S-004P","modelID":"GL-S-004P","networkAddress":44972,"powerSource":"Mains (single phase)","softwareBuildID":"V_1_7_A1","type":"Router","vendor":"Gledopto"},{"dateCode":"20220223","description":"Zigbee 4W MR16 Bulb RGB+CCT (pro)","friendly_name":"Spot02_Souterain_WC","hardwareVersion":2,"ieeeAddr":"0xa4c138f856113b9d","lastSeen":1649080174894,"manufacturerID":4687,"manufacturerName":"GLEDOPTO","model":"GL-S-004P","modelID":"GL-S-004P","networkAddress":54216,"powerSource":"Mains (single phase)","softwareBuildID":"10151203","type":"Router","vendor":"Gledopto"},{"dateCode":"20220223","description":"Zigbee 4W MR16 Bulb RGB+CCT (pro)","friendly_name":"Spot03_Souterain_WC","hardwareVersion":2,"ieeeAddr":"0xa4c138ec7c7708f2","lastSeen":1649080366458,"manufacturerID":4687,"manufacturerName":"GLEDOPTO","model":"GL-S-004P","modelID":"GL-S-004P","networkAddress":17224,"powerSource":"Mains (single phase)","softwareBuildID":"10151203","type":"Router","vendor":"Gledopto"},{"dateCode":"20220223","description":"Zigbee 4W MR16 Bulb RGB+CCT (pro)","friendly_name":"Spot04_Souterain_WC","hardwareVersion":2,"ieeeAddr":"0xa4c1388259039869","lastSeen":1649080547872,"manufacturerID":4687,"manufacturerName":"GLEDOPTO","model":"GL-S-004P","modelID":"GL-S-004P","networkAddress":64181,"powerSource":"Mains (single phase)","softwareBuildID":"10151203","type":"Router","vendor":"Gledopto"},{"dateCode":"20220225","description":"Zigbee 5W GU10 Bulb RGB+CCT (pro)","friendly_name":"Spot01_Overloop","hardwareVersion":2,"ieeeAddr":"0xa4c13817495feda6","lastSeen":1649152847536,"manufacturerID":4687,"manufacturerName":"GLEDOPTO","model":"GL-S-006P","modelID":"GL-S-006P","networkAddress":64712,"powerSource":"Mains (single phase)","softwareBuildID":"10251203","type":"Router","vendor":"GledOpto"},{"dateCode":"20220225","description":"Zigbee 5W GU10 Bulb RGB+CCT (pro)","friendly_name":"Spot02_Overloop","hardwareVersion":2,"ieeeAddr":"0xa4c1381d67706cd8","lastSeen":1649233119501,"manufacturerID":4687,"manufacturerName":"GLEDOPTO","model":"GL-S-006P","modelID":"GL-S-006P","networkAddress":45912,"powerSource":"Mains (single phase)","softwareBuildID":"10251203","type":"Router","vendor":"GledOpto"}]'
debug 2022-04-06 13:49:24: Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data '[{"dateCode":"20211217","friendly_name":"Coordinator","ieeeAddr":"0x00124b00259060db","lastSeen":1649245764381,"networkAddress":0,"softwareBuildID":"zStack3x0","type":"Coordinator"},{"dateCode":"20220223","description":"Zigbee 4W MR16 Bulb RGB+CCT (pro)","friendly_name":"Spot01_Souterain_WC","hardwareVersion":2,"ieeeAddr":"0xa4c138e39b946840","lastSeen":1649079717402,"manufacturerID":4687,"manufacturerName":"GLEDOPTO","model":"GL-S-004P","modelID":"GL-S-004P","networkAddress":19095,"powerSource":"Mains (single phase)","softwareBuildID":"10151203","type":"Router","vendor":"Gledopto"},{"dateCode":"20201021","description":"Zigbee 4W MR16 Bulb RGB+CCT (pro)","friendly_name":"unknown","hardwareVersion":0,"ieeeAddr":"0xbc33acfffe8b93f5","lastSeen":1648659430031,"manufacturerID":4687,"manufacturerName":"GLEDOPTO","model":"GL-S-004P","modelID":"GL-S-004P","networkAddress":5085,"powerSource":"Mains (single phase)","softwareBuildID":"V_1_1","type":"Router","vendor":"Gledopto"},{"dateCode":"20210712","description":"-","friendly_name":"0xa4c13829498fd349","hardwareVersion":0,"ieeeAddr":"0xa4c13829498fd349","lastSeen":1648643307648,"manufacturerID":4687,"manufacturerName":"GLEDOPTO","model":"GL-B-002P","modelID":"GL-B-002P","networkAddress":56482,"powerSource":"Mains (single phase)","softwareBuildID":"10051203","type":"Router","vendor":"-"},{"dateCode":"20210414","description":"Zigbee 4W MR16 Bulb RGB+CCT (pro)","friendly_name":"0x84fd27fffe8aa39a","hardwareVersion":2,"ieeeAddr":"0x84fd27fffe8aa39a","lastSeen":1649079807783,"manufacturerID":4687,"manufacturerName":"GLEDOPTO","model":"GL-S-004P","modelID":"GL-S-004P","networkAddress":44972,"powerSource":"Mains (single phase)","softwareBuildID":"V_1_7_A1","type":"Router","vendor":"Gledopto"},{"dateCode":"20220223","description":"Zigbee 4W MR16 Bulb RGB+CCT (pro)","friendly_name":"Spot02_Souterain_WC","hardwareVersion":2,"ieeeAddr":"0xa4c138f856113b9d","lastSeen":1649080174894,"manufacture [... cut ...] le phase)","softwareBuildID":"10251203","type":"Router","vendor":"GledOpto"},{"dateCode":"20220225","description":"Zigbee 5W GU10 Bulb RGB+CCT (pro)","friendly_name":"Spot02_Overloop","hardwareVersion":2,"ieeeAddr":"0xa4c1381d67706cd8","lastSeen":1649233119501,"manufacturerID":4687,"manufacturerName":"GLEDOPTO","model":"GL-S-006P","modelID":"GL-S-006P","networkAddress":45912,"powerSource":"Mains (single phase)","softwareBuildID":"10251203","type":"Router","vendor":"GledOpto"}],"type":"devices"}'

Koterak, 2022-04-16:
Got the same problem. I received second gledopto bulb and problem still is there.

Koterak, 2022-04-16:
Got answer about this problem. For now I flashed my coordinator with mentioned in this post : https://github.com/Koenkk/zigbee2mqtt/issues/11202#issuecomment-1100653490 firmware and it's working

yoution, 2022-04-22:
[image]
tuya device has the same problem

vinnie99999, 2022-04-25:
> Got answer about this problem. For now I flashed my coordinator with mentioned in this post : [#11202 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/11202#issuecomment-1100653490) firmware and it's working

Thanks Koterak for referring to this issue. I flashed my coordinator with the firmware they are referring to in that issue and now all of my lights stay connected. Even it is a work around, I am more than happy and I can go on adding my 80 lights to my Zigbee network :)

closed 2022-04-25 by vinnie99999