Koenkk/zigbee2mqtt #12090: [New device support]: Model 'TS0505B' and manufacturer name '_TZ3210_iystcadi' is NOT supported...

ma-world, 2022-04-06:
### Link

https://www.lidl.de/p/livarno-home-lichtleiste-mit-lichtfarbensteuerung-rgb-led-zigbee-smart-home/p100339565001

### Database entry

{"id":19,"type":"Router","ieeeAddr":"0xb4e3f9fffec9cae1","nwkAddr":20757,"manufId":4098,"manufName":"_TZ3210_iystcadi","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_iystcadi","stackVersion":0,"dateCode":"","zclVersion":3}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1649167685502,"defaultSendRequestWhen":"immediate"}

### Comments

TS0505B is basically supported but latest Livarno LED has an unsupported manufacturer id

Koenkk, 2022-04-06:
- What is the model number of the device? It probably starts with `HG` (e.g. `HG07834A`)
- Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

ma-world, 2022-04-08:
Hi,
it's Lidl 14149505L and when I add 
`{modelID: 'TS0505B', manufacturerName: '_TZ3210_iystcadi'}` to the fingerprint of the existing 14149505L in lidl.js, it works for me

Koenkk, 2022-04-08:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-04-08 by Koenkk