Koenkk/zigbee2mqtt #12093: Zigbee2MQTT reports device not supported even though it is

neriburo, 2022-04-06:
### What happened?

I have a gas/water walve opener just like this one: https://www.zigbee2mqtt.io/devices/PF-PM02D-TYZ.html#iotperfect-pf-pm02d-tyz . It is supposed to be supported. But when I pair it to Zigbee2MQTT it gives me endless errors:
`Warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_d0ypnbvn'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info MQTT publish: topic 'zigbee2mqtt/boilerio_sklende', payload '{"last_seen":"2022-04-06T18:06:33+03:00","linkquality":138}'
Info MQTT publish: topic 'zigbee2mqtt/boileris', payload '{"child_lock":"UNLOCK","current":0,"energy":212.17,"indicator_mode":"off/on","last_seen":"2022-04-06T18:06:34+03:00","linkquality":54,"power":0,"power_outage_memory":null,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":230}'`
I've tried to remove and re-pair - nothing changes.
When searching on the internet for "TS0601" it gives me results for a zigbee radiator smart valve. Might it be that for some reason Zigbee2MQTT recognizes it wrobgly and therefore treats it as unsuported. And if yes, is there a way aroubd this.
Thank you

[image]

### What did you expect to happen?

The device is supposed to be supported:
https://www.zigbee2mqtt.io/images/devices/PF-PM02D-TYZ.jpg

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20211217

### Adapter

Sonoff ZIGBEE 3.0 USB dongle Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8428363/log.txt)

kylinfedora, 2022-04-08:
It's same to me  
Duplicate of #12117

Koenkk, 2022-04-08:
`_TZE200_d0ypnbvn` was not supported yet, added it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-04-08 by Koenkk

kkossev, 2023-03-26:
@neriburo I suppose you are using your TS0601 _TZE200_d0ypnbvn valve now.
Do you see in the logs data packets from this device (the open or closed state) to Z2M every 3 seconds non-stop?

neriburo, 2023-03-26:
> @neriburo I suppose you are using your TS0601 _TZE200_d0ypnbvn valve now. Do you see in the logs data packets from this device (the open or closed state) to Z2M every 3 seconds non-stop?

Hi. Newer noticed this, but yes, it does. Every 2-3 seconds.

[image]