Koenkk/zigbee2mqtt #12095: docker builds for armv7 are not working

dlashua, 2022-04-06:
### What happened?

when pulling `:latest` or `:1.25.0` docker image, the following error is reported. zigbee2mqtt doesn't run.

```
Using '/app/data' as data directory

#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0xbedc564c
```

```
$ uname -m
armv7l
```

Pulling `:1.24.0` works just fine.

### What did you expect to happen?

zigbee2mqtt should run

### How to reproduce it (minimal and precise)

use docker image `koenkk/zigbee2mqtt:latest` or `koenkk/zigbee2mqtt:1.25.0` pull and run from an armv7 device.

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20210319

### Adapter

CC2652R

Impact123, 2022-04-06:
Take a look at [the docker docs' notes for raspberry users](https://www.zigbee2mqtt.io/guide/installation/02_docker.html#notes-for-raspberry-pi-users) and [the `1.25.0` release's `Notes` section](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.25.0).

dlashua, 2022-04-07:
That did the trick. Thank you.

I had no idea that libraries on the host system could affect a docker image.

closed 2022-04-07 by dlashua

jorneilander, 2022-04-07:
> That did the trick. Thank you.
> 
> I had no idea that libraries on the host system could affect a docker image.

Requiring certain packages to run certain features in Docker and/or kernel calls makes sense. 

But since it's not the version provided by the OS, it's a first for me too. 

Fixed for me as well using the documentation, thanks for `issue` that was my entry to fixing this.