Koenkk/zigbee2mqtt #12097: TS011F_TZ3000_gjnozsaz have to poll data

smulle48, 2022-04-06:
### What happened?

Just got a shipment of the orange plugs (NAS-WR01B) and they all got appVersion 74 and the model is identified as TS011F_plug_1 but still I have to poll the plug to get updates about W, kWh, etc.

### Zigbee2MQTT version

1.25.0 commit: unknown

### Adapter firmware version

20220219

### Adapter

Slae.sh_cc2652rb_stick

### Debug log

Apr  6 19:54:24 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:info  2022-04-06 19:54:24: Zigbee: allowing new devices to join.
Apr  6 19:54:24 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:info  2022-04-06 19:54:24: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"kp7go-3"}'
Apr  6 19:54:25 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:info  2022-04-06 19:54:25: Device '0xa4c138fa1431b940' joined
Apr  6 19:54:25 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:info  2022-04-06 19:54:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138fa1431b940","ieee_address":"0xa4c138fa1431b940"},"type":"device_joined"}'
Apr  6 19:54:25 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:info  2022-04-06 19:54:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c138fa1431b940"},"type":"device_connected"}'
Apr  6 19:54:25 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:info  2022-04-06 19:54:25: MQTT publish: topic 'homeassistant/switch/0xa4c138fa1431b940/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c138fa1431b940/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xa4c138fa1431b940/set","device":{"identifiers":["zigbee2mqtt_0xa4c138fa1431b940"],"manufacturer":"TuYa","model":"Smart plug (with power monitoring) (TS011F_plug_1)","name":"0xa4c138fa1431b940"},"json_attributes_topic":"zigbee2mqtt/0xa4c138fa1431b940","name":"0xa4c138fa1431b940","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0xa4c138fa1431b940","unique_id":"0xa4c138fa1431b940_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Apr  6 19:54:25 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:info  2022-04-06 19:54:25: MQTT publish: topic 'homeassistant/sensor/0xa4c138fa1431b940/power/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c138fa1431b940/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138fa1431b940"],"manufacturer":"TuYa","model":"Smart plug (with power monitoring) (TS011F_plug_1)","name":"0xa4c138fa1431b940"},"device_class":"power","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0xa4c138fa1431b940","name":"0xa4c138fa1431b940 power","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138fa1431b940","unique_id":"0xa4c138fa1431b940_power_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power }}"}'
Apr  6 19:54:25 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:info  2022-04-06 19:54:25: MQTT publish: topic 'homeassistant/sensor/0xa4c138fa1431b940/current/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c138fa1431b940/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138fa1431b940"],"manufacturer":"TuYa","model":"Smart plug (with power monitoring) (TS011F_plug_1)","name":"0xa4c138fa1431b940"},"device_class":"current","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0xa4c138fa1431b940","name":"0xa4c138fa1431b940 current","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138fa1431b940","unique_id":"0xa4c138fa1431b940_current_zigbee2mqtt","unit_of_measurement":"A","value_template":"{{ value_json.current }}"}'
Apr  6 19:54:25 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:info  2022-04-06 19:54:25: MQTT publish: topic 'homeassistant/sensor/0xa4c138fa1431b940/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c138fa1431b940/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138fa1431b940"],"manufacturer":"TuYa","model":"Smart plug (with power monitoring) (TS011F_plug_1)","name":"0xa4c138fa1431b940"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0xa4c138fa1431b940","name":"0xa4c138fa1431b940 voltage","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138fa1431b940","unique_id":"0xa4c138fa1431b940_voltage_zigbee2mqtt","unit_of_measurement":"V","value_template":"{{ value_json.voltage }}"}'
Apr  6 19:54:25 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:info  2022-04-06 19:54:25: MQTT publish: topic 'homeassistant/sensor/0xa4c138fa1431b940/energy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c138fa1431b940/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138fa1431b940"],"manufacturer":"TuYa","model":"Smart plug (with power monitoring) (TS011F_plug_1)","name":"0xa4c138fa1431b940"},"device_class":"energy","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0xa4c138fa1431b940","name":"0xa4c138fa1431b940 energy","state_class":"total_increasing","state_topic":"zigbee2mqtt/0xa4c138fa1431b940","unique_id":"0xa4c138fa1431b940_energy_zigbee2mqtt","unit_of_measurement":"kWh","value_template":"{{ value_json.energy }}"}'
Apr  6 19:54:25 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:info  2022-04-06 19:54:25: MQTT publish: topic 'homeassistant/sensor/0xa4c138fa1431b940/power_outage_memory/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c138fa1431b940/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138fa1431b940"],"manufacturer":"TuYa","model":"Smart plug (with power monitoring) (TS011F_plug_1)","name":"0xa4c138fa1431b940"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:power-settings","json_attributes_topic":"zigbee2mqtt/0xa4c138fa1431b940","name":"0xa4c138fa1431b940 power outage memory","state_topic":"zigbee2mqtt/0xa4c138fa1431b940","unique_id":"0xa4c138fa1431b940_power_outage_memory_zigbee2mqtt","value_template":"{{ value_json.power_outage_memory }}"}'
Apr  6 19:54:25 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:info  2022-04-06 19:54:25: MQTT publish: topic 'homeassistant/select/0xa4c138fa1431b940/power_outage_memory/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c138fa1431b940/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xa4c138fa1431b940/set/power_outage_memory","device":{"identifiers":["zigbee2mqtt_0xa4c138fa1431b940"],"manufacturer":"TuYa","model":"Smart plug (with power monitoring) (TS011F_plug_1)","name":"0xa4c138fa1431b940"},"entity_category":"config","icon":"mdi:power-settings","json_attributes_topic":"zigbee2mqtt/0xa4c138fa1431b940","name":"0xa4c138fa1431b940 power outage memory","options":["on","off","restore"],"state_topic":"zigbee2mqtt/0xa4c138fa1431b940","unique_id":"0xa4c138fa1431b940_power_outage_memory_zigbee2mqtt","value_template":"{{ value_json.power_outage_memory }}"}'
Apr  6 19:54:25 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:info  2022-04-06 19:54:25: MQTT publish: topic 'homeassistant/sensor/0xa4c138fa1431b940/indicator_mode/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c138fa1431b940/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138fa1431b940"],"manufacturer":"TuYa","model":"Smart plug (with power monitoring) (TS011F_plug_1)","name":"0xa4c138fa1431b940"},"enabled_by_default":false,"json_attributes_topic":"zigbee2mqtt/0xa4c138fa1431b940","name":"0xa4c138fa1431b940 indicator mode","state_topic":"zigbee2mqtt/0xa4c138fa1431b940","unique_id":"0xa4c138fa1431b940_indicator_mode_zigbee2mqtt","value_template":"{{ value_json.indicator_mode }}"}'
Apr  6 19:54:25 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:info  2022-04-06 19:54:25: MQTT publish: topic 'homeassistant/select/0xa4c138fa1431b940/indicator_mode/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xa4c138fa1431b940/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xa4c138fa1431b940/set/indicator_mode","device":{"identifiers":["zigbee2mqtt_0xa4 [... cut ...]
Apr  6 19:54:32 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:debug 2022-04-06 19:54:32: Received Zigbee message from '0xa4c138fa1431b940', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"53264":2,"switchType":0}' from endpoint 1 with groupID 0
Apr  6 19:54:32 homeassistant e01a20dadb6a[531]: Zigbee2MQTT:debug 2022-04-06 19:54:32: No converter available for 'TS011F_plug_1' with cluster 'manuSpecificTuya_3' and type 'attributeReport' and data '{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"53264":2,"switchType":0}'

Koenkk, 2022-04-07:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

GSzabados, 2022-04-07:
> Should be fixed now.

@smulle48, before you try the new dev version, can you change the reporting interval of the plug like this?

https://github.com/Koenkk/zigbee2mqtt/issues/11800#issuecomment-1069107519

I have Nous A1Z plugs which have the same appVersion:74 and they do report, but they do not respect the level of change value. When the maximum reporting interval is set to 60, they start to report values every minute.

smulle48, 2022-04-07:
> > Should be fixed now.
> 
> @smulle48, before you try the new dev version, can you change the reporting interval of the plug like this?
> 
> [#11800 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/11800#issuecomment-1069107519)
> 
> I have Nous A1Z plugs which have the same appVersion:74 and they do report, but they do not respect the level of change value. When the maximum reporting interval is set to 60, they start to report values every minute.

Seems like it's working 👍 
Had to manually create them all. it's gonna be a long night, adding it to all my plugs :-D

smulle48, 2022-04-07:
> Should be fixed now.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Thanks 👍 

Is it possible to make it create all the reporting automatically, when the plug is joining the network? Have to create them all manually, before it's reporting.

And is it possible to reset the Energy (sum of consumed energy) ?

smulle48, 2022-04-07:
> Should be fixed now.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Debug Received Zigbee message from '0xa4c138fa1431b940', type 'raw', cluster '57344', data '{"data":[8,94,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug No converter available for 'TS011F_plug_3' with cluster '57344' and type 'raw' and data '{"data":[8,94,10,1,208,72,2,0,6],"type":"Buffer"}'
Debug Received Zigbee message from '0xa4c138fa1431b940', type 'raw', cluster '57344', data '{"data":[8,94,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug No converter available for 'TS011F_plug_3' with cluster '57344' and type 'raw' and data '{"data":[8,94,10,1,208,72,2,0,6],"type":"Buffer"}'
Debug Received Zigbee message from '0xa4c138fa1431b940', type 'raw', cluster '57344', data '{"data":[8,94,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug No converter available for 'TS011F_plug_3' with cluster '57344' and type 'raw' and data '{"data":[8,94,10,1,208,72,2,0,6],"type":"Buffer"}'
Debug Received Zigbee message from '0xa4c138fa1431b940', type 'raw', cluster '57344', data '{"data":[8,94,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug No converter available for 'TS011F_plug_3' with cluster '57344' and type 'raw' and data '{"data":[8,94,10,1,208,72,2,0,6],"type":"Buffer"}'
Debug Received Zigbee message from '0xa4c138fa1431b940', type 'raw', cluster '57344', data '{"data":[8,94,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug No converter available for 'TS011F_plug_3' with cluster '57344' and type 'raw' and data '{"data":[8,94,10,1,208,72,2,0,6],"type":"Buffer"}'
Debug Received Zigbee message from '0xa4c138fa1431b940', type 'raw', cluster '57344', data '{"data":[8,95,10,2,208,72,2,0,10],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug No converter available for 'TS011F_plug_3' with cluster '57344' and type 'raw' and data '{"data":[8,95,10,2,208,72,2,0,10],"type":"Buffer"}'
Debug Received Zigbee message from '0xa4c138fa1431b940', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":6,"rmsCurrent":52,"rmsVoltage":226}' from endpoint 1 with groupID 0

Koenkk, 2022-04-07:
@smulle48 not sure what the plug means with these raw messages (seems like a new tuya cluster so someone has to reverse engineer this) but power measurements seems to be reported now.

smulle48, 2022-04-09:
Yeah it measures now. Lets close it here.....then I'll see what I can find with the sniffer :-D

closed 2022-04-09 by smulle48

GSzabados, 2022-05-11:
You should have an option to set the polling interval for the device. And you should be able to manually define any reporting. 

But as I suggested above, I have a different brand's plug with the same application version and reporting is working (I haven't updated Z2M yet.) The only issue is that the device disregards the minimum changes values for reporting, so the min/max time for reporting has to be adjusted to smaller values if you want to have dense periodic updates. 

https://github.com/Koenkk/zigbee2mqtt/issues/12097#issuecomment-1091911712

Koenkk, 2022-05-11:
@smulle48 seems this fix actually broke reporting for some other people with a `_TZ3000_gjnozsaz` plug (https://github.com/Koenkk/zigbee2mqtt/issues/12378). I reverted the fix + added a possible fix to enable reporting of power measurements, can you check with the latest dev?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

smulle48, 2022-05-11:
> @smulle48 seems this fix actually broke reporting for some other people with a `_TZ3000_gjnozsaz` plug (https://github.com/Koenkk/zigbee2mqtt/issues/12378). I reverted the fix + added a possible fix [...]

I can confirm the changes are working fine here :-) still reporting fine :-)