Koenkk/zigbee2mqtt #12102: TS011F_din_smart_relay broadcast different model name in new version

Kimovitzh, 2022-04-07:
### Link

https://www.zigbee2mqtt.io/devices/TS011F_din_smart_relay.html

### Database entry

{"id":79,"type":"Router","ieeeAddr":"0xa4c1384a67a6abea","nwkAddr":34553,"manufId":4417,"manufName":"_TZ3000_ky0fq4ho","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"seMetering":{"attributes":{"currentSummDelivered":[0,0]}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2833325,"53251":136,"53252":10338,"53253":1270,"53264":2,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1649332458754,"defaultSendRequestWhen":"immediate"}

### Comments

Ordered 4 of the TS011F_din_smart_relay, 2 of them worked fine, but to 2 others have what appears to be a date 2022 03 on the back indicating they are new.
The ones that works doesn't have this date stamp.

The ones that work says TS011F_din_smart_relay in model in zigbee2mqtt dashboard
the ones that doesn't just stated TS011F

### External converter

```shell
-
```

Koenkk, 2022-04-07:
Could you provide the data/database.db of the ones who are recognized as `TS011F`? The ones with `_TZ3000_ky0fq4ho` should already be recognized as `TS011F_din_smart_relay `

Kimovitzh, 2022-04-07:
> Could you provide the data/database.db of the ones who are recognized as `TS011F`? The ones with `_TZ3000_ky0fq4ho` should already be recognized as `TS011F_din_smart_relay`

Sure.
Thus one is recognized:

`{"id":80,"type":"Router","ieeeAddr":"0x84ba20fffe6d50cd","nwkAddr":32893,"manufId":4098,"manufName":"_TZ3000_8bxrzyxz","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_8bxrzyxz","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"rmsCurrent":0,"activePower":0,"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":223}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,0]}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b76a39","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b76a39","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0021b76a39","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1649332009254,"defaultSendRequestWhen":"immediate"}`

Koenkk, 2022-04-07:
Ah, `_TZ3000_ky0fq4ho ` has been added a few days ago: https://github.com/Koenkk/zigbee2mqtt/discussions/12051  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-04-07 by Koenkk

Kimovitzh, 2022-04-09:
Ah, so it will work in the next update?

Koenkk, 2022-04-10:
@Kimovitzh yes (1 may)

Kimovitzh, 2022-04-10:
@Koenkk 🙌🏼

Fantastic!