Koenkk/zigbee2mqtt #12104: The "optimistic: false" behavior of Groups is not defined

RaNo99, 2022-04-07:
### What happened?

The [documentation](https://www.zigbee2mqtt.io/guide/usage/groups.html) is not clear what the behavior of a group is when its "optimistic" property is set to "false". Can someone explain what is the expectation here?

From my experimenting, the existing behavior is inconsistent and unusable. A similar confusion has been expressed in [this unanswered discussion](https://github.com/Koenkk/zigbee2mqtt/discussions/10719), too. Perhaps it's a subject of a bug?

### What did you expect to happen?

I expected that the [documentation](https://www.zigbee2mqtt.io/guide/usage/groups.html) clearly explains what is the behavior of a group whose "optimistic" property is set to "false".

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20210319

### Adapter

CC2530+CC2591

SHxKM, 2022-04-16:
Thanks for raising this @RaNo99 (author of the discussion here). Any conclusions so far? Did you try optimistic:false? I’m yet to try it because I don’t know what the effects will be..

maxill1, 2022-04-18:
I have an issue that may be related to this question. I have a group with 3 TRADFRI LED1537R6/LED1739R5, if I turn on the group and then turn off one of the devices, after a 30-60 seconds the device i turned off manually comes back on by itself dimming it gradually from 0% to 100%!!

Same happens when I turn on just one of the devices, the other two gradually dims on..

I think that what I just described may be the behaviour addressed by the "optimistic: true" config, but I'm not sure.

I tried setting optimistic false but nothing changed. I also checked all my Home Assistant automation to make sure they are not involved.

SHxKM, 2022-04-24:
> I have an issue that may be related to this question. I have a group with 3 TRADFRI LED1537R6/LED1739R5, if I turn on the group and then turn off one of the devices, after a 30-60 seconds the device [...]

I don’t think this has anything to do with the optimistic setting.

maxill1, 2022-04-25:
> I don’t think this has anything to do with the optimistic setting.

After a lot of tests I confirm that it wasn't the optimistic settings. The behaviour I described was caused by the HA Flux component.

RaNo99, 2022-05-06:
> Thanks for raising this @RaNo99 (author of the discussion here). Any conclusions so far? Did you try optimistic:false? I’m yet to try it because I don’t know what the effects will be..

No, nothing. I need to wait till this issue gets resolved.

Koenkk, 2022-05-06:
I hope this explains it: https://github.com/Koenkk/zigbee2mqtt/discussions/10719#discussioncomment-2702500

closed 2022-06-14 by github-actions[bot]