Koenkk/zigbee2mqtt #12106: Received Zigbee message but no MQTT publish happened

sanasol, 2022-04-07:
### What happened?

Some time Tuya Wall Switch(TS0041-TS0043) doesn't trigger anything.
First click after few stand by hours usuaully does nothing.

Looks like it

First button click generates 2 messages and nothing happened after it
```
debug 2022-04-07 18:58:29: Received Zigbee message from 'Living Room Switch', type 'raw', cluster 'genOnOff', data '{"data":[1,1,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2022-04-07 18:58:30: Received Zigbee message from 'Living Room Switch', type 'raw', cluster 'genOnOff', data '{"data":[1,1,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
```

Same here, another switch
```
debug 2022-04-07 18:59:48: Received Zigbee message from 'Second living room switch', type 'raw', cluster 'genOnOff', data '{"data":[1,1,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2022-04-07 18:59:49: Received Zigbee message from 'Second living room switch', type 'raw', cluster 'genOnOff', data '{"data":[1,1,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
```

Second click to Second Living Room switch after few seconds, everything works ok now. Light turned off.
```
debug 2022-04-07 18:59:56: Received Zigbee message from 'Second living room switch', type 'raw', cluster 'genOnOff', data '{"data":[1,2,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
info  2022-04-07 18:59:56: MQTT publish: topic 'zigbee2mqtt/Second living room switch', payload '{"action":"1_single","battery":100,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"Second living room switch","hardwareVersion":1,"ieeeAddr":"0x0c4314fffe86b415","manufacturerID":4098,"manufacturerName":"_TZ3000_adkvzooy","model":"TS0042","networkAddress":29456,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"elapsed":12505202,"linkquality":66}'
info  2022-04-07 18:59:56: MQTT publish: topic 'zigbee2mqtt/Second living room switch', payload '{"action":"","battery":100,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"Second living room switch","hardwareVersion":1,"ieeeAddr":"0x0c4314fffe86b415","manufacturerID":4098,"manufacturerName":"_TZ3000_adkvzooy","model":"TS0042","networkAddress":29456,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"linkquality":66}'
info  2022-04-07 18:59:56: MQTT publish: topic 'zigbee2mqtt/Second living room switch/action', payload '1_single'
debug 2022-04-07 18:59:56: Received MQTT message on 'zigbee2mqtt/Living room light/set' with data 'OFF'
debug 2022-04-07 18:59:56: Publishing 'set' 'state' to 'Living room light'
```

### What did you expect to happen?

MQTT publish action

### How to reproduce it (minimal and precise)

I have no idea, like wait few hours and z2m usually skip first click.

### Zigbee2MQTT version

1.24.0-dev commit: 52d151e7

### Adapter firmware version

20210120

### Adapter

https://aliexpress.com/item/1005002823262979.html

### Debug log

```
debug 2022-04-07 18:54:32: Device 'Second living room switch' announced itself
info  2022-04-07 18:54:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Second living room switch","ieee_address":"0x0c4314fffe86b415"},"type":"device_announce"}'
debug 2022-04-07 18:54:34: Retrieving state of 'Second living room switch' after reconnect
debug 2022-04-07 18:54:35: Received Zigbee message from 'Second living room switch', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2022-04-07 18:55:52: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2022-04-07 18:56:42: Device 'Living Room Switch' announced itself
info  2022-04-07 18:56:42: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Living Room Switch","ieee_address":"0x5c0272fffe06728e"},"type":"device_announce"}'
debug 2022-04-07 18:56:44: Retrieving state of 'Living Room Switch' after reconnect
debug 2022-04-07 18:56:46: Received Zigbee message from 'Living Room Switch', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2022-04-07 18:58:29: Received Zigbee message from 'Living Room Switch', type 'raw', cluster 'genOnOff', data '{"data":[1,1,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2022-04-07 18:58:30: Received Zigbee message from 'Living Room Switch', type 'raw', cluster 'genOnOff', data '{"data":[1,1,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2022-04-07 18:59:48: Received Zigbee message from 'Second living room switch', type 'raw', cluster 'genOnOff', data '{"data":[1,1,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2022-04-07 18:59:49: Received Zigbee message from 'Second living room switch', type 'raw', cluster 'genOnOff', data '{"data":[1,1,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2022-04-07 18:59:56: Received Zigbee message from 'Second living room switch', type 'raw', cluster 'genOnOff', data '{"data":[1,2,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
info  2022-04-07 18:59:56: MQTT publish: topic 'zigbee2mqtt/Second living room switch', payload '{"action":"1_single","battery":100,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"Second living room switch","hardwareVersion":1,"ieeeAddr":"0x0c4314fffe86b415","manufacturerID":4098,"manufacturerName":"_TZ3000_adkvzooy","model":"TS0042","networkAddress":29456,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"elapsed":12505202,"linkquality":66}'
info  2022-04-07 18:59:56: MQTT publish: topic 'zigbee2mqtt/Second living room switch', payload '{"action":"","battery":100,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"Second living room switch","hardwareVersion":1,"ieeeAddr":"0x0c4314fffe86b415","manufacturerID":4098,"manufacturerName":"_TZ3000_adkvzooy","model":"TS0042","networkAddress":29456,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"linkquality":66}'
info  2022-04-07 18:59:56: MQTT publish: topic 'zigbee2mqtt/Second living room switch/action', payload '1_single'
debug 2022-04-07 18:59:56: Received MQTT message on 'zigbee2mqtt/Living room light/set' with data 'OFF'
debug 2022-04-07 18:59:56: Publishing 'set' 'state' to 'Living room light'
info  2022-04-07 18:59:56: MQTT publish: topic 'zigbee2mqtt/Living room light', payload '{"color_mode":"xy","device":{"applicationVersion":67,"dateCode":"","friendlyName":"Living room light","hardwareVersion":1,"ieeeAddr":"0xa4c13883836f833a","manufacturerID":4417,"manufacturerName":"_TZ3000_npzfdcof","model":"WHD02","networkAddress":52779,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"linkquality":84,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'
debug 2022-04-07 18:59:56: Received Zigbee message from 'Living room light', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2022-04-07 18:59:56: MQTT publish: topic 'zigbee2mqtt/Living room light', payload '{"color_mode":"xy","device":{"applicationVersion":67,"dateCode":"","friendlyName":"Living room light","hardwareVersion":1,"ieeeAddr":"0xa4c13883836f833a","manufacturerID":4417,"manufacturerName":"_TZ3000_npzfdcof","model":"WHD02","networkAddress":52779,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"elapsed":12505208,"linkquality":84,"power_on_behavior":"previous","state":"OFF","switch_type":"momentary"}'
debug 2022-04-07 18:59:57: Received Zigbee message from 'Second living room switch', type 'raw', cluster 'genOnOff', data '{"data":[1,3,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
info  2022-04-07 18:59:57: MQTT publish: topic 'zigbee2mqtt/Second living room switch', payload '{"action":"1_single","battery":100,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"Second living room switch","hardwareVersion":1,"ieeeAddr":"0x0c4314fffe86b415","manufacturerID":4098,"manufacturerName":"_TZ3000_adkvzooy","model":"TS0042","networkAddress":29456,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"elapsed":1161,"linkquality":69}'
info  2022-04-07 18:59:57: MQTT publish: topic 'zigbee2mqtt/Second living room switch', payload '{"action":"","battery":100,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"Second living room switch","hardwareVersion":1,"ieeeAddr":"0x0c4314fffe86b415","manufacturerID":4098,"manufacturerName":"_TZ3000_adkvzooy","model":"TS004 [... cut ...]
info  2022-04-07 18:59:57: MQTT publish: topic 'zigbee2mqtt/Living room light', payload '{"color_mode":"xy","device":{"applicationVersion":67,"dateCode":"","friendlyName":"Living room light","hardwareVersion":1,"ieeeAddr":"0xa4c13883836f833a","manufacturerID":4417,"manufacturerName":"_TZ3000_npzfdcof","model":"WHD02","networkAddress":52779,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"elapsed":1148,"linkquality":78,"power_on_behavior":"previous","state":"ON","switch_type":"momentary"}'
```

closed 2022-05-16 by github-actions[bot]

HakunaMatmata, 2023-01-29:
Hello @sanasol, did you manage to make this work?
I have the same problem with an energy meter : CCB432

I receive the zigbee messages (seen in debug logs), but they do not trigger the mqtt publish action.
Thank you,

sanasol, 2023-01-30:
@HakunaMatmata I ended up with not being sure that problem in this.

Replaced Tuya with Aqara and all my problems gone.