Koenkk/zigbee2mqtt #12108: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms) with SONOFF Zigbee 3.0 USB Dongle Plus

roeeklinger, 2022-04-07:
### What happened?

Zigbee2mqtt does not start

Zigbee2MQTT version: `1.25.0-1`
Adapter hardware: [SONOFF Zigbee 3.0 USB Dongle Plus](https://itead.cc/product/sonoff-zigbee-3-0-usb-dongle-plus/) (`CC2652P`)
Adapter firmware version: `CC1352P2_CC2652P_launchpad_coordinator_20220219.hex`
I am using the `Home Assistant addon`

The adapter had been tested and is verified to work just fine with `Zigbee Home Assistant`

conf:
```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: '-d -d'
  log: false
mqtt:
  server: mqtt://192.168.1.215:1883
  user: mqtt
  password: mqtt
  base_topic: zigbee2mqtt
serial:
  port: /dev/ttyUSB0
zigbee_herdsman_debug: true
```

Log response:
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[23:25:17] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[23:25:20] INFO: Zigbee Herdsman debug logging enabled
[23:25:29] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.25.0 start
> node index.js
Zigbee2MQTT:debug 2022-04-07 23:25:52: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-04-07 23:25:52: Logging to console and directory: '/config/zigbee2mqtt/log/2022-04-07.23-25-43' filename: log.txt
Zigbee2MQTT:debug 2022-04-07 23:25:52: Removing old log directory '/config/zigbee2mqtt/log/2022-04-07.23-13-17'
Zigbee2MQTT:info  2022-04-07 23:25:52: Starting Zigbee2MQTT version 1.25.0 (commit #unknown)
Zigbee2MQTT:info  2022-04-07 23:25:52: Starting zigbee-herdsman (0.14.20)
Zigbee2MQTT:debug 2022-04-07 23:25:52: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB0"}}'
2022-04-07T20:25:53.839Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2022-04-07T20:25:53.854Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-04-07T20:25:53.859Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2022-04-07T20:25:53.902Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2022-04-07T20:25:53.915Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2022-04-07T20:25:53.928Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2022-04-07T20:25:54.193Z zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload
2022-04-07T20:25:54.194Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2022-04-07T20:25:55.200Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2022-04-07T20:25:55.202Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2022-04-07T20:25:55.454Z zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352
2022-04-07T20:25:56.011Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2022-04-07T20:25:56.014Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2022-04-07T20:26:02.018Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2022-04-07T20:26:02.020Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2022-04-07T20:26:08.029Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2022-04-07T20:26:08.031Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
Zigbee2MQTT:error 2022-04-07 23:26:14: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-04-07 23:26:14: Failed to start zigbee
Zigbee2MQTT:error 2022-04-07 23:26:14: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-04-07 23:26:14: Exiting...
Zigbee2MQTT:error 2022-04-07 23:26:15: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:100:27)
    at start (/app/index.js:101:5)
[23:26:16] INFO: Zigbee Herdsman debug logging enabled
[23:26:25] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.25.0 start
> node index.js
```

### What did you expect to happen?

[zigbee2mqtt](https://github.com/Koenkk/zigbee2mqtt) would start normally

### How to reproduce it (minimal and precise)

Use my configuration and matching versions and dongle.

### Zigbee2MQTT version

1.25.0-1

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20220219.hex

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

OlsiIgitt, 2022-04-08:
I have the same problem.
Tested with the original (delivery state) Sonoff firmware;
tested: CC1352P2_CC2652P_launchpad_coordinator_20220219;
tested: CC1352P2_CC2652P_launchpad_router_20220125.

I tried several different settings:
serial:
  adapter: zstack
  rtscts: false/true (with 2 different settings of the DIP switch)
  all baudrates
#advanced:
  adapter_concurrent: 16

Always the same result:
... Error: SRSP - SYS - ping after 6000ms ...

OlsiIgitt, 2022-04-08:
Might be that the following observation is helpful for this problem?

When I tried to flash the Koenkk TStack firmware, I initially used the Zigstar Multi Tool - and had no success with that.
Then I used the TI SmartRF Flash Programmer; this worked without any problem.

smulle48, 2022-04-10:
https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-srsp-sys-ping-after-6000ms

OlsiIgitt, 2022-04-11:
I've read it, of course. All hints were not helpful in this case; Sonoff Zigbee 3.0 USB Stick => CC2652P.

I am using Windows, by the way.

Are there configuration parameters for no. of data/stop bits & parity bit setting available in Zigbee2MQTT -> for testing?

MihailRizov, 2022-04-12:
I am also having the same issue. I brought new Sonoff stick, flashed it with "CC1352P2_CC2652P_launchpad_router_20220125" firmware. My version of zigbee2mqtt is 1.25.0

OlsiIgitt, 2022-04-13:
Hope dies last:
Today I repeated all actions with Ubuntu (21.10)
 -> Result = Same problem as with Windows.

See my debug protocol with zigbee-herdsman log (which was not accessible with Windows):
--------------------------------------------------------------------------------
Zigbee2MQTT:info  2022-04-13 11:09:25: Starting Zigbee2MQTT version 1.25.0 (commit #6f1460e4)
Zigbee2MQTT:info  2022-04-13 11:09:25: Starting zigbee-herdsman (0.14.20)
Zigbee2MQTT:debug 2022-04-13 11:09:25: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB0"}}'
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[130,221,81,183,19,163,176,144,162,249,162,132,9,41,88,171],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +49ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +0ms
  zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload +254ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +252ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +1s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352 +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +815ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +815ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
Zigbee2MQTT:error 2022-04-13 11:09:45: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-04-13 11:09:45: Failed to start zigbee
Zigbee2MQTT:error 2022-04-13 11:09:45: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-04-13 11:09:45: Exiting...
Zigbee2MQTT:error 2022-04-13 11:09:45: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
npm ERR! code 1
--------------------------------------------------------------------------------

My question now is:
Is there anybody at all out there who could successfully use his Sonoff USB stick with Zigbee2MQTT???

For clarification:
I had success in the past with Zigbee2MQZZ in connection with the Conbee II stick (under windows); it was a little bit petulant, but it worked lastly - and it's still working! But I wanted to switch over to the more powerfull CC2652P stick.

OlsiIgitt, 2022-04-13:
Problem solved!
1) I erased the NVRAM with Zigstar Multi Tool - I am not sure that this is needed.
2) I switched of my Conbee II stick (power off) which was still running next door.
Then I had success, finally.

I think there was a strong interference on Zigbee channel 11, which was used by both, Conbee II stick and new Sonoff Zigbee stick.
Bve...

closed 2022-05-21 by github-actions[bot]