Koenkk/zigbee2mqtt #12110: Supervisor log full of warnings from Zigbee2mqtt

KitHubek, 2022-04-07:
### What happened?

Supervisor logs has many many warnings from Zigbee2mqtt after update to newest ver 
[image]

### How to reproduce it (minimal and precise)

no warnings at log

### Zigbee2MQTT version

1.25.0-1

### Adapter firmware version

0x26700700

### Adapter

ConBee2/RaspBee2

### Debug log

`22-04-08 00:03:44 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:03:44 WARNING (MainThread) [supervisor.addons.options] Option 'external_converters' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:03:44 WARNING (MainThread) [supervisor.addons.options] Option 'devices' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:03:44 WARNING (MainThread) [supervisor.addons.options] Option 'groups' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:03:44 WARNING (MainThread) [supervisor.addons.options] Option 'homeassistant' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:03:44 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:03:44 WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:03:44 WARNING (MainThread) [supervisor.addons.options] Option 'device_options' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:03:44 WARNING (MainThread) [supervisor.addons.options] Option 'blocklist' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:03:44 WARNING (MainThread) [supervisor.addons.options] Option 'passlist' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:03:44 WARNING (MainThread) [supervisor.addons.options] Option 'queue' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:03:44 WARNING (MainThread) [supervisor.addons.options] Option 'frontend' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:03:44 WARNING (MainThread) [supervisor.addons.options] Option 'experimental' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:03:44 WARNING (MainThread) [supervisor.addons.options] Option 'availability' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:01 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:01 WARNING (MainThread) [supervisor.addons.options] Option 'external_converters' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:01 WARNING (MainThread) [supervisor.addons.options] Option 'devices' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:01 WARNING (MainThread) [supervisor.addons.options] Option 'groups' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:01 WARNING (MainThread) [supervisor.addons.options] Option 'homeassistant' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:01 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:01 WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:01 WARNING (MainThread) [supervisor.addons.options] Option 'device_options' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:01 WARNING (MainThread) [supervisor.addons.options] Option 'blocklist' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:01 WARNING (MainThread) [supervisor.addons.options] Option 'passlist' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:01 WARNING (MainThread) [supervisor.addons.options] Option 'queue' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:01 WARNING (MainThread) [supervisor.addons.options] Option 'frontend' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:01 WARNING (MainThread) [supervisor.addons.options] Option 'experimental' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:01 WARNING (MainThread) [supervisor.addons.options] Option 'availability' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:02 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:02 WARNING (MainThread) [supervisor.addons.options] Option 'external_converters' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:02 WARNING (MainThread) [supervisor.addons.options] Option 'devices' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:02 WARNING (MainThread) [supervisor.addons.options] Option 'groups' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:02 WARNING (MainThread) [supervisor.addons.options] Option 'homeassistant' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:02 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:02 WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:02 WARNING (MainThread) [supervisor.addons.options] Option 'device_options' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:02 WARNING (MainThread) [supervisor.addons.options] Option 'blocklist' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:02 WARNING (MainThread) [supervisor.addons.options] Option 'passlist' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:02 WARNING (MainThread) [supervisor.addons.options] Option 'queue' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:02 WARNING (MainThread) [supervisor.addons.options] Option 'frontend' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:02 WARNING (MainThread) [supervisor.addons.options] Option 'experimental' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:02 WARNING (MainThread) [supervisor.addons.options] Option 'availability' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:03 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:03 WARNING (MainThread) [supervisor.addons.options] Option 'external_converters' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:03 WARNING (MainThread) [supervisor.addons.options] Option 'devices' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:03 WARNING (MainThread) [supervisor.addons.options] Option 'groups' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:03 WARNING (MainThread) [supervisor.addons.options] Option 'homeassistant' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
[... 12 lines cut ...]
22-04-08 00:05:03 WARNING (MainThread) [supervisor.addons.options] Option 'groups' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:03 WARNING (MainThread) [supervisor.addons.options] Option 'homeassistant' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:03 WARNING (MainThread) [supervisor.addons.options] Option 'permit_join' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:03 WARNING (MainThread) [supervisor.addons.options] Option 'advanced' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:03 WARNING (MainThread) [supervisor.addons.options] Option 'device_options' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:03 WARNING (MainThread) [supervisor.addons.options] Option 'blocklist' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
[... cut ...]
22-04-08 00:05:03 WARNING (MainThread) [supervisor.addons.options] Option 'queue' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:03 WARNING (MainThread) [supervisor.addons.options] Option 'frontend' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:03 WARNING (MainThread) [supervisor.addons.options] Option 'experimental' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)
22-04-08 00:05:03 WARNING (MainThread) [supervisor.addons.options] Option 'availability' does not exist in the schema for Zigbee2mqtt (45df7312_zigbee2mqtt)`

Impact123, 2022-04-08:
See #12040

KitHubek, 2022-04-08:
thanks. solved

closed 2022-04-08 by KitHubek