Koenkk/zigbee2mqtt #12111: [New device support]: Nue HGZB-09A Zigbee Smart Curtain Motor Controller

plodmanau, 2022-04-08:
### Link

https://www.lectory.com.au/p/Nue-Zigbee-Smart-Curtain-Controller-Work-with-Stan/NUECURTAINC

### Database entry

{"id":14,"type":"Router","ieeeAddr":"0x00124b001418ab16","nwkAddr":18608,"manufId":0,"manufName":"Feibit Co.Ltd   ","powerSource":"Mains (single phase)","modelId":"FB56+CUR17SB1.4","epList":[14],"endpoints":{"14":{"profId":260,"epId":14,"devId":514,"inClusterList":[0,3,4,5,258],"outClusterList":[10],"clusters":{"genBasic":{"attributes":{"modelId":"FB56+CUR17SB1.4","manufacturerName":"Feibit Co.Ltd   ","powerSource":1,"zclVersion":1,"hwVersion":1,"dateCode":"20170227        "}},"genOnOff":{"attributes":{"onOff":0}},"genTime":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"hwVersion":1,"dateCode":"20170227        ","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1649397422943,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to follow the support new device page but got stuck at adding converters.  I am a noob at javascritpt and I could not work out how to create custom converters to convert from / to Zigbee:

type 'attributereport', cluster 'genOnOff", data '{"onOff":X} ' where X is:
    0 - UP / OPEN button
    1 - DOWN / CLOSE button
    2 - STOP button

[image]
[image]
[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['FB56+CUR17SB1.4'],
    model: 'HGZB-09A',
    vendor: 'Nue',
    description: 'Nue Zigbee Smart Curtain Motor Controller',
    fromZigbee: [fz.OnOFF],
    toZigbee: [],
    exposes: [],
};
module.exports = definition;
```

smulle48, 2022-04-09:
Right now, z2m are only receiving messages from the switch (fromZigbee) but do not send anything as toZigbee is empty.

Try to change it to `toZigbee: [tz.OnOff]` and see what happens

smulle48, 2022-04-10:
Try this

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['FB56+CUR17SB1.4'],
    model: 'HGZB-09A',
    vendor: 'Nue',
    description: 'Nue Zigbee Smart Curtain Motor Controller',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [],
};
module.exports = definition;`

plodmanau, 2022-04-11:
Using _'fromZigbee: [fz.on_off]'_ results in the MQTT payload 'OFF' for both the top and the bottom buttons and a payload of 'ON' for the middle button (see below log output).  However the top button should ideally result in a payload of "OPEN", the bottom button "CLOSE" and the middle button "STOP".

Testing with _'toZigbee: [tz.on_off]'_ suggests that you are correct that this device cannot be remotely controlled.  I could not hear any click of relays when sending ON and OFF commands to the state variable.  Unfortunately this does mean that the device is of no use to me.

### **LOG OUTPUT FOR TOP, BOTTOM AND THEN MIDDLE BUTTON PRESSES (IN SEQUENCE)**

Debug Received Zigbee message from 'Theatre Curtains', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 14 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Theatre Curtains/state', payload 'OFF'
Info MQTT publish: topic 'zigbee2mqtt/Theatre Curtains/linkquality', payload '142'
Debug Received Zigbee message from 'Theatre Curtains', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 14 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Theatre Curtains/state', payload 'ON'
Info MQTT publish: topic 'zigbee2mqtt/Theatre Curtains/linkquality', payload '142'
Debug Received Zigbee message from 'Theatre Curtains', type 'attributeReport', cluster 'genOnOff', data '{"onOff":2}' from endpoint 14 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Theatre Curtains/state', payload 'OFF'
Info MQTT publish: topic 'zigbee2mqtt/Theatre Curtains/linkquality', payload '134'

plodmanau, 2022-04-12:
Hi,
Log is below.  Order of button presses was top, bottom then middle which resulted in "onOff" returned values of 0, 1 & 2 respectively.

Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"fmzbg-1","value":true}'
Info Zigbee: allowing new devices to join.
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"fmzbg-1"}'
Info Device '0x00124b001418ab16' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b001418ab16","ieee_address":"0x00124b001418ab16"},"type":"device_joined"}'
Info Starting interview of '0x00124b001418ab16'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b001418ab16","ieee_address":"0x00124b001418ab16","status":"started"},"type":"device_interview"}'
Debug Device '0x00124b001418ab16' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b001418ab16","ieee_address":"0x00124b001418ab16"},"type":"device_announce"}'
Info Successfully interviewed '0x00124b001418ab16', device has successfully been paired
Warning Device '0x00124b001418ab16' with Zigbee model 'FB56+CUR17SB1.4' and manufacturer name 'Feibit Co.Ltd ' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x00124b001418ab16","ieee_address":"0x00124b001418ab16","status":"successful","supported":false},"type":"device_interview"}'
Debug Received Zigbee message from '0x00124b001418ab16', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 14 with groupID 0
Warning Received message from unsupported device with Zigbee model 'FB56+CUR17SB1.4' and manufacturer name 'Feibit Co.Ltd '
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0x00124b001418ab16', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 14 with groupID 0
Warning Received message from unsupported device with Zigbee model 'FB56+CUR17SB1.4' and manufacturer name 'Feibit Co.Ltd '
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0x00124b001418ab16', type 'attributeReport', cluster 'genOnOff', data '{"onOff":2}' from endpoint 14 with groupID 0
Warning Received message from unsupported device with Zigbee model 'FB56+CUR17SB1.4' and manufacturer name 'Feibit Co.Ltd '
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

plodmanau, 2022-04-12:
Also I just found the manual for this switch which talks about remote control through the Nue Home App so remote control through Zigbee2MQTT should be possible.
[image]

closed 2022-05-20 by github-actions[bot]