Koenkk/zigbee2mqtt #12113: TS011F  (_TZ3000_typdpbpg) strange behaviour

Masterz69, 2022-04-08:
### What happened?

It switching off suddenly, with no any automation from HA side or manual action. Just become OFF....

At same time in Z2M log appears up to 10 records with same time (same second), but different data, like LQI.
```
info  2022-04-08 04:55:33: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T04:55:33+03:00","linkquality":126,"power":0,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":235}'
...
info  2022-04-08 05:00:33: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:00:33+03:00","linkquality":123,"power":0,"power_outage_memory":"on","state":"ON","update":{"state":null},"update_available":null,"voltage":235}'
... ... ...
info  2022-04-08 05:04:31: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:04:29+03:00","linkquality":105,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":235}'
info  2022-04-08 05:04:31: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:04:31+03:00","linkquality":111,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":235}'
info  2022-04-08 05:04:31: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:04:31+03:00","linkquality":117,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":235}'
info  2022-04-08 05:04:31: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:04:31+03:00","linkquality":120,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":235}'
info  2022-04-08 05:04:31: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:04:31+03:00","linkquality":117,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":235}'
info  2022-04-08 05:04:31: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:04:31+03:00","linkquality":117,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":235}'
info  2022-04-08 05:04:31: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:04:31+03:00","linkquality":111,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":235}'
info  2022-04-08 05:04:31: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:04:31+03:00","linkquality":111,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":235}'
info  2022-04-08 05:04:31: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:04:31+03:00","linkquality":108,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":235}'
info  2022-04-08 05:04:31: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:04:31+03:00","linkquality":108,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":235}'
info  2022-04-08 05:04:31: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:04:31+03:00","linkquality":111,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":235}'
...
info  2022-04-08 05:04:31: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:04:31+03:00","linkquality":111,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":0}'
... ...
info  2022-04-08 05:04:36: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:04:36+03:00","linkquality":105,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":0}'
info  2022-04-08 05:04:37: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:04:37+03:00","linkquality":108,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":233}'
...
info  2022-04-08 05:05:33: MQTT publish: topic 'zigbee2mqtt/plg Washer', payload '{"child_lock":"UNLOCK","current":0,"energy":17.71,"indicator_mode":"off/on","last_seen":"2022-04-08T05:05:33+03:00","linkquality":108,"power":0,"power_outage_memory":"on","state":"OFF","update":{"state":null},"update_available":null,"voltage":233}'
```

Just discovered - "voltage":0 on first and second message when outlet suddenly switched OFF.

Let me know if can provide more information.

Checked by connecting similar device (bought at same time) to Tuya hub: plug have latest software.

### How to reproduce it (minimal and precise)

Just leave plug in ON state for some hours. It will go OFF suddenly.

### Zigbee2MQTT version

1.24.0-1

### Adapter firmware version

20210708 

### Adapter

sOnOff

Masterz69, 2022-04-08:
[image]
Just screenshot to memorize.

RHAD1969, 2022-04-08:
I think this is a duplicate: https://github.com/Koenkk/zigbee2mqtt/issues/11648

Masterz69, 2022-04-08:
@RHAD1969 - yes, You right. Closing this and will follow #11648.

closed 2022-04-08 by Masterz69