Koenkk/zigbee2mqtt #12114: Routers are off-grid

Pablo-or-Pablo, 2022-04-08:
### What happened?

I have SONOFF Zigbee 3.0 USB Dongle Plus working with Home Assistant.
All software / drivers and plugins are always updated.

Every time, after a few hours of normal operation, two of my routers disconnect (off-grid), it shows that they are off the network.
I originally only had one but couldn't "fix" it so I bought a second one and now they both shut down at the same time. In the current version this time seems longer to me, but it's still only a few hours.
One of the routers is a TuYa WHD02 and the other is a SONOFF S26R2ZB.

### What did you expect to happen?

Find the source of problems, maybe some automatic restart of my addon, to have my routers always 

### How to reproduce it (minimal and precise)

There is always the same situation. It does not is connected with any special situation.

### Zigbee2MQTT version

1.25.0-1

### Adapter firmware version

20220125

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

Zigbee2MQTT:warn  2022-04-08 10:17:40: Failed to ping 'Światło konturowe' (attempt 1/1, Read 0xa4c138e7554a0558/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
Zigbee2MQTT:warn  2022-04-08 10:21:06: Failed to ping '0x00124b0024cb1bc0' (attempt 1/1, Read 0x00124b0024cb1bc0/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))

Pablo-or-Pablo, 2022-05-11:
Reopen please

Chippi, 2022-05-13:
Did you try these steps?
https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-srsp-sys-ping-after-6000ms

Pablo-or-Pablo, 2022-05-17:
Hi,

yes. All is ok. I don't have problem with ping, connection with any router.
All my routers disconnect after some time, once a few hours, once after a
full day, always all at once (I've never seen one router unavailable and
the rest are ok).
All everything becomes visible again only after I reset the add-on. It
never fixes itself.
I also check all docs and any founded forum about it, I didn't find any
reason or solution.

Chippi, 2022-05-17:
Do you live in a crowded area? I haven't received this error, but i've experienced similar issues because of high congestion on the 2.4 Ghz band. I wrote a post about it here.

https://github.com/Koenkk/zigbee2mqtt/discussions/12467

closed 2022-06-28 by github-actions[bot]