Koenkk/zigbee2mqtt #12115: Heiman HS2WD-E Siren do not respond to Warning modes

jhonny001, 2022-04-08:
### What happened?

Siren Heiman HS2WD-Esucessfully added to Z2MQTT but when publishing the Warning message, the siren does not react as expected. It always triggers the same alarm, no matter which mode, level, strobe_level, strobe, strobe_duty_cycle, or duration is set in the message. It only switches the alarm ON with full volume (which is painful...). Hopefully, it reacts to the STOP mode...

### What did you expect to happen?

Siren should react accordingly to the message and exposes sent via MQTT. Being able to select the mode, volume, strobe light etc...

### How to reproduce it (minimal and precise)

Add Heiman HS2WD-E to Zigbee2MQTT and try setting any alarm message via the Exposes section.

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

Debug Received MQTT message on 'zigbee2mqtt/Sirena/set' with data '{"warning":{"level":"low","mode":"stop","strobe":false,"strobe_level":"low"}}'
Debug Publishing 'set' 'warning' to 'Sirena'
Debug Received MQTT message on 'zigbee2mqtt/Sirena/set' with data '{"warning":{"duration":2,"level":"low","mode":"burglar","strobe":true,"strobe_level":"low"}}'
Debug Publishing 'set' 'warning' to 'Sirena'
Debug Received Zigbee message from 'Sirena', type 'attributeReport', cluster 'ssIasWd', data '{"61440":3}' from endpoint 1 with groupID 0
Debug No converter available for 'HS2WD-E' with cluster 'ssIasWd' and type 'attributeReport' and data '{"61440":3}'
Debug Received MQTT message on 'zigbee2mqtt/Sirena/set' with data '{"warning":{"duration":2,"level":"low","mode":"stop","strobe":true,"strobe_level":"low"}}'
Debug Publishing 'set' 'warning' to 'Sirena'
Debug Received Zigbee message from 'Sirena', type 'attributeReport', cluster 'ssIasWd', data '{"61440":2}' from endpoint 1 with groupID 0
Debug No converter available for 'HS2WD-E' with cluster 'ssIasWd' and type 'attributeReport' and data '{"61440":2}'
Debug Received Zigbee message from 'Sirena', type 'attributeReport', cluster 'ssIasWd', data '{"61440":0}' from endpoint 1 with groupID 0
Debug No converter available for 'HS2WD-E' with cluster 'ssIasWd' and type 'attributeReport' and data '{"61440":0}'
Debug Saving state to file /config/zigbee2mqtt/state.json
Debug Received Zigbee message from 'Sirena', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":25,"batteryVoltage":35}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Sirena', payload '{"battery":12.5,"linkquality":76,"voltage":3500}'

writepatrick, 2022-04-08:
Same as  #11036 but not solved and no progress from my side

Bartekn86, 2022-04-09:
https://github.com/Koenkk/zigbee2mqtt/issues/9380

NeeeeB, 2022-04-09:
@jhonny001 sorry it is a bit off-topic but how did you manage to pair your siren with Z2M ?
I have one since 6 months, i used to work like a charm (paired at first try).
I've had to "remove" it from Z2M, and since then, I'm totally unable to pair it anymore.
I've bought a new one (thinking the old one was broken), received yesterday.
Same problem, I can't pair it...
Always the same thing happen, device is found by by Z2M, interview starts, it takes almost a minute, and then it fails with "device not successfully paired"...
Also, as soon as the device is detected by Z2M, it is listed, but immediatly with a little red panel...
Any help would be greatly appreciated, even @Koenkk 's help lol

Mutagennn, 2022-04-10:
Same problem here with same adapter on v1.24

closed 2022-05-19 by github-actions[bot]

J-Bot87, 2023-04-15:
⬆️