Koenkk/zigbee2mqtt #12121: IKEA LED1624G9 Brightness After Setting XY Color Issue

markheloking, 2022-04-08:
### What happened?

Pairing the device, no problem. Turning the device off/on, no problem. Changing brightness, no problem. Changing the color on the device - it now seems to go into some "color mode" reducing the brightness for plain white significantly. Going to  RGB(255,255,255) on full brightness now only gives about 40% of the usual brightness of the LED bulb, plus the color is way off (much too blue). It seems that in the XY color mode, it disables the white LED's and just uses the RGB LED's, reducing overall maximum brightness.
As far as I can tell, there's no way to revert back to "just white" mode in zigbee2mqtt GUI.

### What did you expect to happen?

I expect the device to be able to be set back to "just white" mode reverting back to it's usual color pallet and brightness after setting a color mode.

### How to reproduce it (minimal and precise)

1. Pair device
2. Turn device on
3. Change color using Zigbee2MQTT color picker from GUI
4. Set color back to plain white
5. Observe faulty white level (much too blue) and reduced brightness

### Zigbee2MQTT version

1.25.0

### Adapter firmware version

20220219 (although the stick still reports 20211217 somehow)

### Adapter

Slaesh's CC2652RB

smulle48, 2022-04-09:
some debug log would be nice to pinpoint the problem :-)

https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging

markheloking, 2022-04-22:
I've added the debug log when changing the colors.

closed 2022-05-30 by github-actions[bot]

Semmu, 2023-06-07:
Duplicate of #3738 , still no solution unfortunately.