Koenkk/zigbee2mqtt #12123: extension data doesnt provide id

Gibbz, 2022-04-09:
### What happened?

I have updated to zigbee2mqtt 1.25 and noticed the extensions no longer work. The ID always returns undefined:

In the example extension onStateChange:
const { ID, update } = data;
console.log(ID) // this returns undefined

### What did you expect to happen?

Id should provide device id

### How to reproduce it (minimal and precise)

as above log the ID to console

### Zigbee2MQTT version

1.25

### Adapter firmware version

20220219

### Adapter

zzh

nurikk, 2022-04-17:
Try to `console.log(data)` and inspect object structure

Gibbz, 2022-04-30:
`{
  entity: Device {
    zh: Device {
      _events: [Object: null prototype] {},
      _eventsCount: 0,
      _maxListeners: undefined,
      ID: 14,
      _type: 'EndDevice',
      _ieeeAddr: '0x847127fffe0a4028',
      _networkAddress: 26437,
      _manufacturerID: 4098,
      _endpoints: [Array],
      _manufacturerName: '_TZ3000_bi6lpsew',
      _powerSource: 'Battery',
      _modelID: 'TS0043',
      _applicationVersion: 67,
      _stackVersion: 0,
      _zclVersion: 3,
      _hardwareVersion: 1,
      _dateCode: '',
      _softwareBuildID: undefined,
      _interviewCompleted: true,
      _interviewing: false,
      _skipDefaultResponse: false,
      _skipTimeResponse: false,
      meta: {},
      _lastSeen: 1651277855552,
      _defaultSendRequestWhen: 'immediate',
      _linkquality: 163,
      [Symbol(kCapture)]: true
    },
    _definition: {
      zigbeeModel: [Array],
      model: 'TS0043',
      vendor: 'TuYa',
      description: 'Wireless switch with 3 buttons',
      whiteLabel: [Array],
      exposes: [Array],
      fromZigbee: [Array],
      toZigbee: [Array],
      configure: [AsyncFunction: configure],
      options: []
    },
    _definitionModelID: 'TS0043'
  },
  from: {},
  to: {},
  reason: null,
  update: { action: '' }
}
`

Heres the data im getting. ID is undefined, also from, to , reason and update are all empty. 
I can grab ID from data.entity.zh._ieeeAddr but the others I dont have access too... Any ideas?

Gibbz, 2022-04-30:
update: going back this issue looks like it was introduced in the changes from 1.23 and onwards. None of those versions work.

Gibbz, 2022-04-30:
Ive managed to get it working. If anyone has the same issue, my extension is attached:
[automation.zip](https://github.com/Koenkk/zigbee2mqtt/files/8595506/automation.zip)

closed 2022-04-30 by Gibbz