Koenkk/zigbee2mqtt #12127: Device class garage

ton6i, 2022-04-09:
### What happened?

When i add Garage door opener TS0601, _TZE200_nklqjk62 i get error message from Home assistant voluptuous.error.MultipleInvalid: expected BinarySensorDeviceClass for dictionary value @ data['device_class'].
complete message:
2022-04-09 21:01:43 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_binary_sensor_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Garage deur/availability'}], 'availability_mode': 'all', 'device': {'identifiers': ['zigbee2mqtt_0xa4c1385cab13a2d7'], 'manufacturer': 'TuYa', 'model': 'Garage door opener (PJ-ZGD01)', 'name': 'Garage deur'}, 'device_class': 'garage', 'json_attributes_topic': 'zigbee2mqtt/Garage deur', 'name': 'Garage deur garage door contact', 'payload_off': False, 'payload_on': True, 'state_topic': 'zigbee2mqtt/Garage deur', 'unique_id': '0xa4c1385cab13a2d7_garage_door_contact_zigbee2mqtt', 'value_template': '{{ value_json.garage_door_contact }}', 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 246, in async_discover
    config = schema(discovery_payload)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: expected BinarySensorDeviceClass for dictionary value @ data['device_class']

Can it be that there is no 'device_class': 'garage'?
In Home Assistant i can find garage_door as device class, so shouldn't it be 'device_class': 'garage_door'?

### What did you expect to happen?

I expected a contact with the status of the garage door sensor in mqtt.
I can see the status of the garage door sensor in Zigbee2Mqtt device.

### How to reproduce it (minimal and precise)

Pair the sensor.

### Zigbee2MQTT version

Current version: 1.25.0-1

### Adapter firmware version

0x26720700

### Adapter

ConBee2

ton6i, 2022-04-09:
After change config in:
{
    "availability": [{
            "topic": "zigbee2mqtt/bridge/state"
        }, {
            "topic": "zigbee2mqtt/Garage deur/availability"
        }
    ],
    "availability_mode": "all",
    "device": {
        "identifiers": ["zigbee2mqtt_0xa4c1385cab13a2d7"],
        "manufacturer": "TuYa",
        "model": "Garage door opener (PJ-ZGD01)",
        "name": "Garage deur"
    },
    "device_class": "**garage_door**",
    "json_attributes_topic": "zigbee2mqtt/Garage deur",
    "name": "Garage deur garage door contact",
    "payload_off": false,
    "payload_on": true,
    "state_topic": "zigbee2mqtt/Garage deur",
    "unique_id": "0xa4c1385cab13a2d7_garage_door_contact_zigbee2mqtt",
    "value_template": "{{ value_json.garage_door_contact }}"
}
{
    "garage_door_contact": false,
    "last_seen": "2022-04-09T20:19:21.260Z",
    "linkquality": 255,
    "trigger": true
}

the state of the door is known.

Koenkk, 2022-04-10:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/12038

closed 2022-04-10 by Koenkk