Koenkk/zigbee2mqtt #12131: [Feature request]: (Virtual) Groups of Groups

PaulMcMillan, 2022-04-10:
### Is your feature request related to a problem? Please describe

I have several hundred devices. Many of them represent logical groups (e.g. "Bulb A, B, C in a lamp fixture" - I usually want to control the whole fixture at once). However, I also want to control larger logical partitions (e.g. "All lamps in Room A"). This larger group logically exists as a group of groups, but to manage it today, I have to add each individual lamp from every group. If I swap a bulb in a fixture, I may have to adjust several different groups to fix things back up.

### Describe the solution you'd like

I want to be able to include an existing group as an element of another group. I know Zigbee doesn't support this directly, but Z2M could keep track of "groups within groups" and flatten/dedupe before sending out changes.

A few complexities:
* Updating a group requires a separate "and update any groups which include this group" call
* We need to gracefully handle recursion (total number of groups is small, so just track and skip already visited groups)

### Describe alternatives you've considered

This can be managed manually, but it encourages subtle errors which aren't immediately obvious.

### Additional context

Right now if I add, remove, or replace a device, I need to update as many as half a dozen groups individually. With this feature, I could adjust the single most logical grouping, and depend on all others to catch up.

An example of a deeply nested logical group that ends up maintained in many individual groups today:
* Lightbulb A is a member of the Dining Room Chandelier group
* Dining Room Chandelier is a member of Dining Room
* Dining Room is a member of Downstairs
* Downstairs is a member of Inside (in contrast to the Exterior and Pool groups)
* Inside is a member of the Everything group (used for whole-property scenes and bulk turnoff/on)

Today, If I replace Lightbulb A, I have to update each of these groups individually by hand to add the new bulb, and if I miss a layer of logical grouping, I might not notice until those automations leave a single bulb in the wrong state. If we add this feature, I can update the Dining Room Chandelier group and be confident that all groupings down the chain will be properly updated.

smulle48, 2022-04-10:
Ii have never had any problems with this. I got about 50 groups, and if I want to control more then one group at once, I just hook them up in node-red. ie. in my living room I got 3 groups; spot, dining light and coffee table light. All groups got 3-4 light each and are controlled by their own switch. Then I got an Aqara Apple 6 gang switch can control them all, even dimming.

PaulMcMillan, 2022-04-10:
The problem isn't triggering one or two groups at a time, but if you have a command which goes to all 50 groups at once, you'll clog up your zigbee network with a bunch of unnecessary commands that are relayed through node-red, adding a bunch of latency and causing the popcorn effect. Devices can generally happily listen for a handful of broadcast groups, so it makes sense to add them to several as you need them. This is especially true if you're also using zigbee scenes.

smulle48, 2022-04-10:
Maybe it's more of an frontend feature request (surely it could be done by mqtt, but properly not that convenient).

https://github.com/nurikk/zigbee2mqtt-frontend

PaulMcMillan, 2022-04-10:
Hmmm, yeah when I filed the ticket I was debating about which project to file with. My impression was that the frontend is almost entirely stateless on the server side, which is why I filed it in this project. My thinking was that keeping track of "this group is a member of that group" was something that would need state, but you're right that it's more of a "UI/UX" thing than an underlying "mqtt" thing.

I'll re-file the ticket over there. Thanks!

closed 2022-04-10 by PaulMcMillan

EmJay276, 2024-01-05:
@PaulMcMillan
Using the yaml config and friendly names, it is working fine for me to nest groups into a group.
In the UI they are listed as individual devices in the nexted group, but I am fine with that.
I normally only change devices in the lowest level auf groups.

```
groups:
  '8':
    friendly_name: Deckenlicht Wohnzimmer Sofa
    transition: 1
    devices:
      - 0xa4c138be5b86a398/11
  '9':
    friendly_name: Deckenlicht Wohnzimmer Tisch
    transition: 1
    devices:
      - 0x001788010b7fb70b/11
      - 0x001788010b7fa675/11
      - 0x001788010b7f8b7d/11
      - 0x001788010b7f9a5b/11
  '10':
    friendly_name: Deckenlicht Wohnzimmer
    transition: 1
    devices:
      - 'Deckenlicht Wohnzimmer Sofa'
      - 'Deckenlicht Wohnzimmer Tisch'
```