Koenkk/zigbee2mqtt #12134: rob_200-017-1

WBoerhoop, 2022-04-10:
### What happened?

After the installation of ROBB ROB_200-017-1de pairing performed and everything is visible normally. However, in the zigbee log both ROB_200-017-1 states "Failed to configure 'RobPlug0x'" resulting in no kWh notation.

### What did you expect to happen?

Normal use notation kWh

### Zigbee2MQTT version

v.1.25.0 (zStack3x0 20210120)

### Adapter firmware version

20210120

### Adapter

Slaesh's CC2652RB

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8458671/log.txt)

WBoerhoop, 2022-04-10:
inclusive debug
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8458742/log.txt)

WBoerhoop, 2022-04-10:
[image]

smulle48, 2022-04-11:
Try to update to lates dev. It should be fixed there. Same issue as this:

https://github.com/Koenkk/zigbee2mqtt/issues/12000

WBoerhoop, 2022-04-11:
Ok, I updated to latest dev, and as far as i can see now , it's fixed
Thanks

closed 2022-04-11 by WBoerhoop