Koenkk/zigbee2mqtt #12135: JavaScript heap out of memory when installing 1.2.5.0 in Raspberry Zero 2W

uli-rpi, 2022-04-10:
### What happened?

I get the following error while installing V 1.2.5.0 from scratch at raspberry OS bullseye (32bit)

pi@mqttbroker:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.25.0 start
> node index.js

Building Zigbee2MQTT... (initial build), failed
Build failed; ran out-of-memory, free some memory (RAM) and start again

(node:6828) UnhandledPromiseRejectionWarning: Error: Command failed: npm run build
FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory
Aborted
npm ERR! code 134
npm ERR! path /opt/zigbee2mqtt
npm ERR! command failed
npm ERR! command sh -c tsc && node index.js writehash

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-04-10T11_42_46_544Z-debug.log

    at ChildProcess.exithandler (child_process.js:308:12)
    at ChildProcess.emit (events.js:314:20)
    at maybeClose (internal/child_process.js:1022:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5)
(node:6828) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:6828) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

### Zigbee2MQTT version

1.2.5.0

### Adapter firmware version

Z-Stack_3.x.0 20220219

### Adapter

CC2652P

lux73, 2022-04-10:
there is an closed [Issue](https://github.com/Koenkk/zigbee2mqtt/issues/12081)

Koenkk, 2022-04-10:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/12081

closed 2022-04-10 by Koenkk

uli-rpi, 2022-04-12:
I´ve try to use the NODE_OPTIONS=--max_old_space_size=256 option
Likewise with 512 (the nano 2W has 512 MB memory)

However, I get the same error

pi@mqttbroker:/opt/zigbee2mqtt $ NODE_OPTIONS=--max_old_space_size=256
pi@mqttbroker:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.25.0 start
> node index.js

Building Zigbee2MQTT... (initial build), failed

Build failed; ran out-of-memory, free some memory (RAM) and start again

(node:6160) UnhandledPromiseRejectionWarning: Error: Command failed: npm run build
FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory
Aborted
npm ERR! code 134
npm ERR! path /opt/zigbee2mqtt
npm ERR! command failed
npm ERR! command sh -c tsc && node index.js writehash

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-04-12T16_11_34_787Z-debug.log

    at ChildProcess.exithandler (child_process.js:308:12)
    at ChildProcess.emit (events.js:314:20)
    at maybeClose (internal/child_process.js:1022:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5)
(node:6160) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:6160) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

lux73, 2022-04-12:
there's a missing `export` before your NODE_OPTIONS

uli-rpi, 2022-04-12:
😬 i will retry