Koenkk/zigbee2mqtt #12136: TS130F (curtain motor) entities in wrong HASS category

SamMousa, 2022-04-10:
### What happened?

I installed the TS130F, it shows like this:
[image]

### What did you expect to happen?

I would expect calibration & motor reversal to be configuration entities.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

 0x26720700

### Adapter

ConBee2

McGiverGim, 2022-04-11:
It seems the covers have different names to publish the same things...

For Tuya, there is:
- motor_reversal 
https://github.com/Koenkk/zigbee-herdsman-converters/blob/893c989ad10dee2dcfbc55adaff2e1bfbe3b15c4/converters/toZigbee.js#L6183
- calibration (switch) and time_calibration (numeric), depending on the Tuya device:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/893c989ad10dee2dcfbc55adaff2e1bfbe3b15c4/converters/toZigbee.js#L6169
https://github.com/Koenkk/zigbee-herdsman-converters/blob/893c989ad10dee2dcfbc55adaff2e1bfbe3b15c4/converters/toZigbee.js#L6194

But to Home Assistant, it is prepared to use:
- invert_cover
https://github.com/Koenkk/zigbee2mqtt/blob/6f1460e47b430f8e1fe22819da56a238f8c92174/lib/extension/homeassistant.ts#L545
- And yes, it has a calibration one but is numeric:
https://github.com/Koenkk/zigbee2mqtt/blob/6f1460e47b430f8e1fe22819da56a238f8c92174/lib/extension/homeassistant.ts#L619

@Koenkk what is the correct solution here? To me it seems the cleaner way is to rename the tuya sensors by the "correct" ones, but is a breaking change. Add them to the homeassistant file produces duplications.

SamMousa, 2022-05-12:
This should be fixed in the next release that contains the merged pr above.

SamMousa, 2022-05-12:
Checked my mqtt discovery message and it looks good in 1.25.1. Hass does not seem to be changing entity category for existing entities though.
 Thanks @McGiverGim!

closed 2022-05-12 by SamMousa